brainstorm-tools/brainstorm3 #10: Grid button in PSD and time series figures

ftadel, 2017-05-17:
In the PSD and time series figures, the Grid button still requires some work:
- When switching between the two types of display modes (butterfly / columns), the selected properties should be maintained, both for the PSD and the time series plot
- Use the fields TsInfo.ShowXGrid and TsInfo.ShowYGrid to record the expected status of this property for the plot - this way, it will be also compatible with the menus in the popup menu Figure > Show XGrid/YGrid
- In "columns" mode: we only want to display the X grid, the Y grid is too dense and makes no sense.

martcous, 2017-05-17:
It's good you reported this, I realized there's a memory leak in the way the PSD figure was coded. Every time it's refreshed (e.g. changing display modes), new buttons are added on top of each other. I fixed it, I'll add it in the PR addressing these issues.

I have some reservations on the last point. Are you sure you want the same button having different behaviors depending on the display mode? The code for that will be messy also. Anyway, if so, I need a use case clarified: If the user checks the grid button in butterly mode, both X and Y Grids will show. What would be the expected behavior when the user switches to column view? The YGrid will be hidden automatically but the button still checked? What if they manually showed XGrid and YGrid seperately through the right click popup menu? That distinction is lost since both that menu and the single show grid button use TsInfo.ShowX/YGrid to store the information.

ftadel, 2017-05-18:
The YGrid makes no sense in the columns view, so it has to be disabled in this mode.
Switching to column mode disables the YGrid, but keeps the XGrid if it was enabled. 
Switching to butterfly enables the YGrid if the XGrid if is enabled.

The button is checked in XGrid or YGrid are enabled. 
Unselecting the button disables both.
Selecting the button enables the XGrid (all cases) and YGrid (only in butterfly mode).

Every change must be reported both
- in the properties of the axes, and 
- in the appdata TsInfo.

Does it make sense?
Have fun with all the corresponding tests :)

martcous, 2017-05-18:
Thanks for the clarifications. See PR #11.

closed 2017-05-19 by martcous