brainstorm-tools/brainstorm3 #20: Displaying averaged timefreq sources in MRI (3D) view returns an error

jeremymoreau, 2017-06-21:
Steps to reproduce:

1. Import a 10s segment of MEG data into Brainstorm database (downsample to 600Hz)
2. Compute head model (MRI volume, overlapping spheres, whole-head 5mm isotropic grid) and noise covariance
3. Compute sources (dSPM)
4. Compute timefreq on sources (Welch's PSD, group in frequency bands)
5. Repeat steps 1-4 for a second 10s segment in another run)
6. Average both timefreq files
7. Under "Intra-subject" node, right click on averaged timefreq result -> "Display on MRI (3D)"

"Display on MRI (MRI Viewer)" works fine

```
Error using cellfun
Non-scalar in Uniform output, at index 1, output 1.
Set 'UniformOutput' to false.

Error in bst_figures>GetChannelsForFigure (line 252)
        iChanZero = find(cellfun(@(c)(~isequal(size(c),[3,1]) | all(abs(c)<1e-5)), chanLoc));

Error in bst_figures>CreateFigure (line 217)
    [selChan,errMsg] = GetChannelsForFigure(iDS, iFig);

Error in bst_figures (line 59)
eval(macro_method);

Error in view_surface_data (line 271)
        [hFig, iFig, isNewFig] = bst_figures('CreateFigure', iDS, FigureId, 'AlwaysCreate');

Error in tree_callbacks>@(h,ev)view_surface_data(MriFile,filenameRelative) (line 1871)
                            gui_component('MenuItem', jPopup, [], 'Display on MRI   (3D)',
                            IconLoader.ICON_ANATOMY, [], @(h,ev)view_surface_data(MriFile,
                            filenameRelative), []);
```

ftadel, 2017-06-23:
Hi Jeremy,

I guess this is not only MEG, the recordings contain a few channels defined as EEG or SEEG. This should not be a problem though. I tried but I cannot reproduce this error.
I don't understand how the expression  "~isequal(size(c),[3,1]) | all(abs(c)<1e-5)" could be non-scalar...

Could you place a break point at bst_figures, line 252, and check the contents of the variable chanLoc?

jeremymoreau, 2017-06-23:
chanLoc is an empty 1x22 cell (chanLoc{1} = Empty matrix: 3-by-0)

Let me know if I can test anything else, thanks!

ftadel, 2017-06-26:
I don't understand why this can cause an error in cellfun...
Can you package a small example? Duplicate the subject, delete all the unnecessary files, export it to a .zip and send it to me.
Thanks

jeremymoreau, 2017-06-26:
ok, shared via box.BIC. The timefreq file in question is in Intra-subject -> Avg. Non-averaged timefreq sources give me no problems, and the standard MRI viewer works fine even for the average, as mentioned above.
Thanks!
Jeremy

ftadel, 2017-06-28:
I changed the way I was doing this test and it seems to work now.
Update is already on /meg/meg1

closed 2017-06-28 by ftadel

jeremymoreau, 2017-06-28:
yep, works for me too. Thanks!