brainstorm-tools/brainstorm3 #24: Page duration should remain the same across files with different sampling frequencies

jeremymoreau, 2017-07-24:
I resample 1-70Hz bandpassed raw EEG/MEG recordings to 300Hz (acquired at 2400Hz) for visualisation and event marking to improve page flip speeds. However, I've noticed that the "Duration" value in the "Page settings" box of the "Record" tab seems to readjust the page duration based on the number of samples. For instance, if I have I have a run sampled at 2400Hz open with page duration set to 10s and open another run resampled at 300Hz, then the page duration switches to 80s. This seems like a bug; the page duration setting should stay constant no matter the sampling frequency (e.g. 10s at 2400Hz or 300Hz).

ftadel, 2017-07-24:
The parameter MaxSamples was just supposed to be something preventing the display of "too much data" causing a crash/freeze of Matlab. It is set here, if someone in interested in improving this: 
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/gui/panel_record.m#L591

Related to the same topic: To match the behavior in the SPM viewer, I made the red cursor at the bottom of the figure resizable. You can set the page duration from there too.
http://neuroimage.usc.edu/brainstorm/Tutorials/ReviewRaw#Epoched_vs._continuous

jeremymoreau, 2017-07-24:
Ok. I find this behaviour a bit strange because the setting is labelled "Duration ... (s)" in the user interface, which I don't intuitively interpret as a number of samples, but I see your point. In my use case (epilepsy), it's important to use standard page lengths, so I was hoping this would work a bit more like the equivalent setting in the NK software (or other clinical EEG viewers). The resizable cursor is nice. Thanks.

ftadel, 2017-08-19:
In commit https://github.com/brainstorm-tools/brainstorm3/commit/94779672d2f936e0d6340387e9d674d868f5786c, I replaced the field MaxSamples (samples) with PageDuration (seconds). When you open files with different sampling rates, it should always use the same duration. 
Test it and close the issue if you're satisfied.

jeremymoreau, 2017-08-21:
Yes, this fixes the issue. Thanks!

closed 2017-08-21 by jeremymoreau