brainstorm-tools/brainstorm3 #46: Tree does not refresh after reloading moved protocol

jeremymoreau, 2017-10-31:
I'm running Brainstorm 24-Oct-2017 in Matlab 2016b on Ubuntu 16.04.3. After I reload a protocol moved to a different location, the file tree in the main Brainstorm window no longer refreshes when switching protocol (e.g. if I try switching from protocol A to protocol B in the dropdown menu, the files in protocol A remain displayed in the file tree below). There are no error messages displayed in the Matlab console.

Steps to reproduce:
In a brainstorm database with more than 2 protocols,

1. Move one of the protocols to a different location on the disk (in my case, this was after moving my data to a new computer)
2. Open Brainstorm and try loading the protocol. The "Load protocol, The following folder has been moved..." dialog box will pop up.
3. Select "Pick folder..." and choose the new protocol location. This loads the protocol as expected.
4. Try switching to a different protocol using the dropdown menu. This does not work.

My current work-around is to close Brainstorm and open it again, after which switching protocols using the dropdown menu works again.

martcous, 2017-10-31:
Well this is a weird bug... jComboBoxProtocols's ContentsChangedCallback just vanishes after a call to protocolComboBoxChanged_Callback(), even though just before the end of the execution it is still present, so I am unable to find what actually clears it. However, I did find that the call to UpdateProtocolsList() on line 971 of gui_brainstorm.m (which is not necessary in this case in my opinion) is what causes this weird behavior.

@ftadel In gui_brainstorm's UpdateProtocolsList(), do we really need to save and restore the callback? I tried without and it still works, and fixes the above bug. Programmatically adding and removing items to jComboBoxProtocols does not trigger the callback. I am also on Ubuntu, this bug could be platform dependent.

ftadel, 2017-10-31:
Does this commit fixes the issue:  https://github.com/brainstorm-tools/brainstorm3/commit/c7c2f7ac16329072a0a80aeebd8f33080bc101ef

The combobox events are triggered in different ways with different versions of Java (and therefore different versions of Matlab) and different operating systems. There are combinations where changing the selected item triggers ContentsChangedCallback... It's been a nightmare to get some of the JList and JComboBox callbacks called once and only once. The solution I found that works most of the time was to remove and set again the Matlab callback function...

martcous, 2017-11-01:
I see, that's annoying, thanks for the info. It's now working on my end. Jeremy?

jeremymoreau, 2017-11-01:
yes, fixed for me as well (Brainstorm 31-Oct-2017). Thanks!

closed 2017-11-01 by ftadel