brainstorm-tools/brainstorm3 #51: Cursor stuck at start of page after opening a second figure/window

jeremymoreau, 2017-12-07:
I am using Brainstorm 06-Dec-2017 on Ubuntu 16.04.3 with Matlab 2016b. My desktop environment is MATE 1.12.1 and I use the mutter 3.18.3 window manager. I have noticed an annoying bug where the red cursor is reset and stays stuck at the start of a page after opening a second window/figure and clicking anywhere one the first window. This does not affect only the latest version of Brainstorm (unfortunately, I do not know in which version this appeared), but seems to be platform dependent (I could not reproduce the problem on another workstation running an older version of Ubuntu MATE with a different window manager).

Steps to reproduce:

1. Open a MEG recording time series view.
2. Move the red cursor by clicking somewhere (e.g. at 5s)
3. Open a second time series (e.g. EOG) or figure (e.g. topography)
4. Click somewhere on the first time series as if to move the red cursor. The red cursor is reset to the start of the page, and won't move.
5. If the second window that was opened is a time series (e.g. EOG), then it is still possible to move the red cursor by clicking on that window, but clicking anywhere on the first window resets the cursor position again. Moving the cursor with arrow keys or page up/down is also still possible.

martcous, 2017-12-08:
I cannot reproduce the issue on my Ubuntu 16 with the same version of Matlab. Could it be specific to certain MEG time series?

jeremymoreau, 2017-12-08:
Yes, as we discussed earlier, this only seems to affect raw time series. Imported time series do not appear to be affected.

jeremymoreau, 2018-03-06:
I've figured out a couple more things.

- I believe this bug is related to the implementation of NVIDIA's TwinView feature on Linux, which treats multiple monitors a single extended x-screen. This is an old feature (c. 2000) and is on by default. The alternative mode (using separate x-screens for separate displays) is not attractive because it prevents drag-dropping windows from one screen to the other.
-  In my setup I have two 2560x1440 monitors (display 0 on the left at (0,0) , display 1 on the right at (2560, 0)). Display 1 is the primary display. I want the Brainstorm main interface window to be on disp0 and the figures to open on disp1.
    - In the "NVIDIA X Server Settings". Setting the position of display 1 to "relative (Right of)" instead of the default "Absolute +2560+0" makes no difference.
    - After opening a time series and then another time series or figure (which triggers the bug), moving the first time series from disp1 to disp0 enables moving the red cursor again. Weirdly, moving the window from disp1 to disp0, and then back from disp0 to disp1 without clicking anything, also re-enables correct function of the red cursor.
    - Moving disp1 to be on the left of disp0 in the Ubuntu "Monitor Preferences" makes no difference.
    - Switching the HDMI cables so that disp1 is on the left of disp0 makes no difference. However, switching the HDMI cables on the graphics card AND switching the order of displays in the "NVIDIA X Server Settings" appears to temporarily solve the issue.
   - Unfortunately closing Brainstorm and Matlab, then opening them again brings back the problem...

This is the output of `get(0,'MonitorPositions')`:
```
ans =
           1           1        2560        1440
       -2559           1        2560        1440
```
Doing the following, I can draw figures on the display of my choice:
```
pos = get(0,'MonitorPositions');
hfig = figure('Position',pos(1,:));  % Right
hfig = figure('Position',pos(2,:));  % Left
```

It appears that which display the Brainstorm main interface window opens on is fairly arbitrary. The only way I've figured out to change it involves some sequence of switching from "Two Screens" to "One Screen" mode, moving the main interface window from one display to another, restarting Matlab and Brainstorm, switching back to "Two screens mode", restarting Brainstorm and Matlab again. I still haven't quite figured out how to do this reliably, and get all kinds of weird problems like both the figures and the main interface window drawing on the same screen (overlapping) or the tiling mode acting up. I think it would be nicer/cleaner if we could define in the Brainstorm preferences on which display to draw the main interface window and the figures (e.g. some dropdown menu that appears if nb_displays > 1 and let's you select which display to draw the main interface window on).

ftadel, 2018-06-07:
Maybe related with this problem?
https://neuroimage.usc.edu/forums/t/mouse-issues-with-timeseries-plots-and-windows-screensize-in-ubuntu/6983/2

closed 2018-11-28 by ftadel