brainstorm-tools/brainstorm3 #52: Add a new default average referential EEG Montage with all left channels on top

jeremymoreau, 2017-12-07:
Creating new bipolar montages in Brainstorm is easy, but average montages are not so obvious. I would like an additional default (automatically generated regardless of number of EEG electrodes) montage with all the left channels on top, followed by midline channels, and right channels (e.g. Fp1, F7, ..., Fz, Cz, ..., Fp2, F8, ...). This has many use cases, but lateralisation of epileptiform activity is one. A spacer between the three groups (left, midline, right) would be nice also).

ftadel, 2017-12-08:
Good idea.
Can you prepare the montage(s) you want and send me corresponding text?
I'll add then as default options.

jeremymoreau, 2017-12-08:
The "text montage editor" isn't available for the average reference montage (the montage editor just shows a read-only identity matrix) unlike for the other bipolar or referential montage. The montage would also differ depending on the number of EEG channels recorded or bad channels set. I think this needs to be automatically generated like the current interleaved average reference montage, just in a different order.

This "Average reference (L -> R)" montage would go something like this:
Left channels
 :
Midline channels
 :
Right channels

The order within each group should be anterior -> posterior (e.g. Fp1, F7, F3, T7, C3, P7, P3, O1; this should just be the index order in the channel file since channels are typically labelled anterior-posterior by convention).

ftadel, 2017-12-12:
Ok, I see... I won't have time to work on this before next year.
Maybe @martcous is intersted in working on that?

martcous, 2018-07-12:
Done in PR https://github.com/brainstorm-tools/brainstorm3/pull/94

closed 2018-07-12 by martcous