brainstorm-tools/brainstorm3 #53: Custom keyboard shortcut definition for EEG/MEG montages

jeremymoreau, 2017-12-08:
Switching montages via keyboard shortcuts is currently possible via shift + A, B, C etc. There are a couple of issues with this setup, however:

1. Deleting a bad EEG channel often results in all the keyboard shortcuts changing (because some montages become unavailable). This also means keyboard shortcuts are frequently different across runs...
2. The only way to change the keyboard shortcut for a montage via the GUI is by deleting multiple montages, and adding them back in a different order. This also makes it annoying to get the same keyboard shortcuts on different computers.
3. No keyboard shortcuts for montages >26 (which means only 8 custom montages on top of the defaults for CTF MEG). 

It would be nice if it were possible to set the keyboard shortcut in the montage editor (even if they all still follow the "shift + letter" pattern).

ftadel, 2017-12-12:
It would possible to set shortcuts the way we configure shortcuts for the event marking. But I won't have time to work on this before next year.
@martcous?

ftadel, 2018-06-07:
PR merged, now we need to update the documentation (text and screen captures):
https://neuroimage.usc.edu/brainstorm/Tutorials/ReviewRaw#Montage_selection
https://neuroimage.usc.edu/brainstorm/Tutorials/MontageEditor

There are probably many other places where some updates are needed, an exhaustive review of the tutorials is needed.

martcous, 2018-06-08:
Change documented here: https://neuroimage.usc.edu/brainstorm/Tutorials/MontageEditor#Assigning_a_keyboard_shortcut

closed 2018-06-08 by martcous