brainstorm-tools/brainstorm3 #63: BstDownload.java errors on first download attempt

martcous, 2018-02-21:
BstDownload.java throws the error "Error: File is empty." the first time you try to download a file with gui_brainstorm's DownloadFile(). It works on subsequent attempts. Sadly, this is hard to reproduce as after deleting everything and restarting brainstorm + Matlab it still works the second time. However, this was reported by multiple users using WaveClus or Plotly. Did this error ever occur for downloading OpenMEEG?

I'm not sure how to fix this... We could let DownloadFile() try to download twice in a row (ugly solution), we could add a sleep delay in BstDownload before measuring the contentLength... Any ideas?

ftadel, 2018-02-22:
No, I don't know how to fix this, I've never had this problem and no one reported it to me.
This function is used quite a lot to download to the Brainstorm updates, and it seems to work.
Maybe the issue is related with the web server on the other side?...
You could try and play with the way the HTTP connection is created.

martcous, 2018-02-22:
Good point, both for WaveClus and Plotly it tries to download a file from GitHub.

ftadel, 2018-04-10:
Please close this issue after testing that it works with the new brainstorm.jar

martcous, 2018-04-12:
The switch to TLSv1.2 did the trick, AWESOME! HTTPS download links will require Java v7+ now, which shouldn't restrict users of older versions since our "important" downloads are on HTTP.

closed 2018-04-12 by martcous