brainstorm-tools/brainstorm3 #68: Add support for basic boolean logic in filter search box

jeremymoreau, 2018-03-10:
Adding support for basic boolean operators in the filter search box would be a nice feature. I know more complex file selection operations can be achieved by stacking "Select Files" processes (and with scripting of course), but I think it would enhance usability to make the filter search box a bit more powerful. Specifically, I would propose adding support for [PubMed-style syntax][1] to the filter box, because I expect it would be familiar to most Brainstorm users.

1. Supported operators are: AND, OR, NOT
2. Search terms are separated by a space.
3. Search terms containing a space can be joined by enclosing them in double quotes.
4. AND is the default operator. It is added between search terms if none are specified.
5. Operations are processed left to right.
6. The order of processing can be changed by nesting search terms in parentheses. The terms inside the parentheses will be processed as a unit first.

Some examples:
- 20180109 AND somatosensory
- resample AND ("left mn" OR noise)
- (stim AND trial) NOT "stimulation 2"

[1]: https://www.nlm.nih.gov/bsd/disted/pubmedtutorial/020_390.html

ftadel, 2018-03-12:
Sure. That's a job for @martcous :)

ftadel, 2018-03-12:
Do not forget to edit the documentation when adding this feature:
http://neuroimage.usc.edu/brainstorm/Tutorials/PipelineEditor#Filter_by_name_or_comment

martcous, 2018-03-13:
Great suggestion, this was a lot of fun to implement! See PR#69.

martcous, 2018-03-20:
The change was merged and pushed to the BIC repo.

closed 2018-03-21 by ftadel