brainstorm-tools/brainstorm3 #76: SEEG Montage process_montage_apply.m

gabri470, 2018-04-24:
Case 1: SEEG subject with 185 monopolar recordings. I used the panel_montage > SEEG (Bipolar 2) auto-created montage file which will have lead to 162 referenced dataset. Then Standardize > Apply Montage process. I ended up to a " Dimension mismatch Error " in process_montage_apply.m at 139.

`                isChanBad = (double(sMontage.Matrix(iMatrixDisp,iMatrixChan) ~= 0) * double(DataMat.ChannelFlag(iChannels)' == -1) > 0);
`

workaround is the transposition of ChannelFlag(iChannels) mat in order to have 162x185 * 185x1 matrix multiplication.

ftadel, 2018-04-26:
This means that in your file, you have a matrix DataMat.ChannelFlag that is of dimensions [1 x Nchannels]? It should always be saved as [Nchannels x 1]. 
Can you tell me how this happened?

I added a fix for your existing file (using "reshape" instead of a transposition):
 https://github.com/brainstorm-tools/brainstorm3/commit/a2b1a71182400dae599d46636f575863ca0b85d7#diff-e5c8e6bc456202d4a14872df359a1a9c
Please let me know if this works.

I'd also like to fix any process that produce files with wrong dimensions of the ChannelFlag field.

gabri470, 2018-04-27:
I will replicate the whole pipeline on a fresh data before applying your patch and try to figure out the offending process.

gabri470, 2018-11-15:
with latest release I processed 8 new subjects and nothing wrong happened so far. hence we can probably consider this issue resolved.