brainstorm-tools/brainstorm3 #81: Moving Average

Edouard2laire, 2018-05-22:
Hi. 

is there a simple way to do a moving average of a signal in Brainstorm ? 

Best. 
Edouard

ftadel, 2018-05-22:
If you have a specific function you'd like to run on any signal, you can do it with the process "Pre-process > Run Matlab command". 

With the existing processes, you can maybe obtain the result you want with a low-pass filter. A moving average is a form of low-pass filter.

For questions relative to the use of the software, please post your messages on the user forum (section Forum on the website): https://neuroimage.usc.edu/forums/
We would like to keep using the github issues as an actual bug tracker.
Thanks

closed 2018-05-22 by ftadel

ftadel, 2018-05-22:
If you have a good reason for doing so, why not. You can read on how to create new processes with this tutorial: https://neuroimage.usc.edu/brainstorm/Tutorials/Scripting
If this is just to smooth your signal in time: use a low-pass filter instead.

For what specific reason would you like to use a moving average instead of the low-pass filter implemented in Brainstorm?
http://www.analog.com/media/en/technical-documentation/dsp-book/dsp_book_Ch15.pdf