brainstorm-tools/brainstorm3 #83: BIDS Importer Improvements

martcous, 2018-06-05:
Submitted by Elizabeth Bock. Issues when importing many BIDS subjects at once.

- is it possible to start import optionally at a specified subject. Example, I have the BIDS dataset imported and then I add more subjects. I would like it to start at subject 221 instead of 1. Currently it starts again at 1 and then skips through each one - which can take a really long time.
- is it possible to skip a subject that has an error and continue to the next. Example, I have a corrupted .nii file that killed the import at subject 137. Then I had to re-start to get the rest of the subjects imported (the previous issue might be the same solution here). Maybe it could just report the error and skip to the next subject?
- for emptyroom subjects (named sub-emptyroom), you can skip the anatomy and registration steps - there are no humans involved here
- there is a message "Warning: Multiple .pos head shape points found..." but actually there is no .pos file in those recordings, so I'm not sure what it means
- can we have all the output messages during the import be sent to a report?

ftadel, 2018-06-05:
> is it possible to start import optionally at a specified subject. Example, I have the BIDS dataset imported and then I add more subjects. I would like it to start at subject 221 instead of 1. Currently it starts again at 1 and then skips through each one - which can take a really long time.

Added in: https://github.com/brainstorm-tools/brainstorm3/commit/c6ce87a32eca166a7cc1c576597def5aa16f0eea

A new option "Names of subjects to import" in the process "Import BIDS dataset" lets you enter a list of subject names, separated with comas.

> is it possible to skip a subject that has an error and continue to the next. Example, I have a corrupted .nii file that killed the import at subject 137. Then I had to re-start to get the rest of [...]

If you use the process instead of the interactive menu (File > Load protocol > ...), it should keep going after errors instead of stopping the process. 

> for emptyroom subjects (named sub-emptyroom), you can skip the anatomy and registration steps - there are no humans involved here

But it works without adding any specific exclusion right? Except for a few warnings, it doesn't harm to process sub-emptyroom just like all the other subjects, does it?

> there is a message "Warning: Multiple .pos head shape points found..." but actually there is no .pos file in those recordings, so I'm not sure what it means

Fixed in the same commit.

> can we have all the output messages during the import be sent to a report?

This is more complicated to do...

martcous, 2018-06-22:
All done!!!

ftadel, 2018-06-25:
Not yet: For all the changes you make to the interface, please adjust all the documentation, tutorial pages and screen captures.

Also, this is something I would have preferred to have in a single pull request than in multiple direct commits, so that we could move around some of the processes before validating everything, to create a new menu "Registration" for instance.

martcous, 2018-06-26:
I went through the Importing tutorials already, can you pinpoint specifics that I missed? I did not make a pull request because I thought we agreed beforehand with the mock-up. What would you like to move around exactly?

martcous, 2018-06-28:
Thanks Francois, those pages were updated.

closed 2018-06-28 by martcous