brainstorm-tools/brainstorm3 #107: New labels

thinault, 2018-08-16:
Labels = {...
1000, 'lh-unknown'; ...
1001, 'lh-bankssts'; ...
1002, 'lh-caudalanteriorcingulate'; ...
1003, 'lh-caudalmiddlefrontal'; ...
1004, 'lh-corpuscallosum'; ...
1005, 'lh-cuneus'; ...
1006, 'lh-entorhinal'; ...
1007, 'lh-fusiform'; ...
1008, 'lh-inferiorparietal'; ...
1009, 'lh-inferiortemporal'; ...
1010, 'lh-isthmuscingulate'; ...
1011, 'lh-lateraloccipital'; ...
1012, 'lh-lateralorbitofrontal'; ...
1013, 'lh-lingual'; ...
1014, 'lh-medialorbitofrontal'; ...
1015, 'lh-middletemporal'; ...
1016, 'lh-parahippocampal'; ...
1017, 'lh-paracentral'; ...
1018, 'lh-parsopercularis'; ...
1019, 'lh-parsorbitalis'; ...
1020, 'lh-parstriangularis'; ...
1021, 'lh-pericalcarine'; ...
1022, 'lh-postcentral'; ...
1023, 'lh-posteriorcingulate'; ...
1024, 'lh-precentral'; ...
1025, 'lh-precuneus'; ...
1026, 'lh-rostralanteriorcingulate'; ...
1027, 'lh-rostralmiddlefrontal'; ...
1028, 'lh-superiorfrontal'; ...
1029, 'lh-superiorparietal'; ...
1030, 'lh-superiortemporal'; ...
1031, 'lh-supramarginal'; ...
1032, 'lh-frontalpole'; ...
1033, 'lh-temporalpole'; ...
1034, 'lh-transversetemporal'; ...
1035, 'lh-insula'; ...
2000, 'rh-unknown'; ...
2001, 'rh-bankssts'; ...
2002, 'rh-caudalanteriorcingulate'; ...
2003, 'rh-caudalmiddlefrontal'; ...
2004, 'rh-corpuscallosum'; ...
2005, 'rh-cuneus'; ...
2006, 'rh-entorhinal'; ...
2007, 'rh-fusiform'; ...
2008, 'rh-inferiorparietal'; ...
2009, 'rh-inferiortemporal'; ...
2010, 'rh-isthmuscingulate'; ...
2011, 'rh-lateraloccipital'; ...
2012, 'rh-lateralorbitofrontal'; ...
2013, 'rh-lingual'; ...
2014, 'rh-medialorbitofrontal'; ...
2015, 'rh-middletemporal'; ...
2016, 'rh-parahippocampal'; ...
2017, 'rh-paracentral'; ...
2018, 'rh-parsopercularis'; ...
2019, 'rh-parsorbitalis'; ...
2020, 'rh-parstriangularis'; ...
2021, 'rh-pericalcarine'; ...
2022, 'rh-postcentral'; ...
2023, 'rh-posteriorcingulate'; ...
2024, 'rh-precentral'; ...
2025, 'rh-precuneus'; ...
2026, 'rh-rostralanteriorcingulate'; ...
2027, 'rh-rostralmiddlefrontal'; ...
2028, 'rh-superiorfrontal'; ...
2029, 'rh-superiorparietal'; ...
2030, 'rh-superiortemporal'; ...
2031, 'rh-supramarginal'; ...
2032, 'rh-frontalpole'; ...
2033, 'rh-temporalpole'; ...
2034, 'rh-transversetemporal'; ...
2035, 'rh-insula'; ...
 };

ftadel, 2018-08-16:
Please add them on your panel_scout.m, test it and make sure it does what you want:
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/gui/panel_scout.m#L5253

Then I encourage you to post the modification yourself on github (open a pull request). It's a good way to keep track of the various contributions and evolutions of the software. If you don't know how to do it and don't want to spend time in learning how github works, send me the file and I'll push it myself.

thinault, 2018-08-20:
I added the labels on panel_scouts but nothing changed. I will let you know when it works.

ftadel, 2018-08-29:
Fixed in https://github.com/brainstorm-tools/brainstorm3/commit/40631a4f45ad262af4484fc69cb610dfa8564af0

closed 2018-08-29 by ftadel