brainstorm-tools/brainstorm3 #114: Inverse methods: Add eLORETA

ftadel, 2018-09-06:
https://neuroimage.usc.edu/forums/t/sloreta-poor-performance/7959/8

jcmosher, 2019-10-28:
**Possible Background References:**

ELoreta is an iterative method that evokes memories of FOCUSS, by Gorodnitsky, George, and Rao, building on a premise by Clark and Ionnides, all from the late 1980s and early 1990s. Irina and Bhaskar published a detailed paper on FOCUSS in

http://citeseerx.ist.psu.edu/viewdoc/download?doi=10.1.1.154.6856&rep=rep1&type=pdf

**Sparse Signal Reconstruction from Limited Data Using FOCUSS:** 
**A Re-weighted Minimum Norm Algorithm**
Irina F. Gorodnitsky and Bhaskar D. Rao

In a subsequent search, I found this very interesting paper by Candes, which reviewed Rao's and others iterative methods in a broader category, ultimately linking to Compressive Sensing:

https://statweb.stanford.edu/~candes/papers/rwl1.pdf

**Enhancing Sparsity by Reweighted ℓ1 Minimization**

Emmanuel J. Cand`es† Michael B. Wakin♯ Stephen P. Boyd§
† Applied and Computational Mathematics, Caltech, Pasadena, CA 91125
♯ Electrical Engineering & Computer Science, University of Michigan, Ann Arbor, MI 48109
§ Department of Electrical Engineering, Stanford University, Stanford, CA 94305

October 2007; Revised July 2008

_It is now well understood that (1) it is possible to reconstruct sparse signals exactly from
what appear to be highly incomplete sets of linear measurements and (2) that this can be
done by constrained ℓ1 minimization. In this paper, we study a novel method for sparse signal
recovery that in many situations outperforms ℓ1 minimization in the sense that substantially
fewer measurements are needed for exact recovery. The algorithm consists of solving a sequence
of weighted ℓ1-minimization problems where the weights used for the next iteration are computed
from the value of the current solution. We present a series of experiments demonstrating the
remarkable performance and broad applicability of this algorithm in the areas of sparse signal
recovery, statistical estimation, error correction and image processing. Interestingly, superior
gains are also achieved when our method is applied to recover signals with assumed near-sparsity
in overcomplete representations—not by reweighting the ℓ1 norm of the coefficient sequence as is
common, but by reweighting the ℓ1 norm of the transformed object. An immediate consequence
is the possibility of highly efficient data acquisition protocols by improving on a technique known
as Compressive Sensing._

ftadel, 2019-10-29:
Does this lead to any conclusion regarding Rey's code or eLORETA implementation in Brainstorm?

ftadel, 2021-09-19:
https://neuroimage.usc.edu/forums/t/loreta-and-source-localization/30525/2

ftadel, 2022-06-08:
No one will apparently work on this any time soon.
Keeping it only in the todo page: https://neuroimage.usc.edu/brainstorm/Next

closed 2022-06-08 by ftadel