brainstorm-tools/brainstorm3 #118: Mixed head model not supported for dipole modelling

ftadel, 2018-09-17:
https://neuroimage.usc.edu/forums/t/zeros-in-imaging-kernel-matrix-after-dipole-modeling/8081/3

jcmosher, 2019-04-24:
I fixed the mixed head model for the sLORETA case, and it was previously re-instated for the other min norm stuff. But I don't recall that we ever agreed that it should work for dipole modeling. To me, I don't even understand why you would use a mixed head model for dipole modeling. We need to discuss.

The code at this point has zero anticipation of mixed head models for LCMV or GLS. It's not a bug, it's a feature, we don't support it?

ftadel, 2019-04-25:
OK, no problem, then this combination of options simply needs to be disabled in the interface.

ftadel, 2019-05-02:
Done in commit: https://github.com/brainstorm-tools/brainstorm3/commit/41a0f980964b5cce4a136920d6cfd11aeb8a2873
Disabled computation of Diplodes+LCMV for mixed head models

closed 2019-05-02 by ftadel