brainstorm-tools/brainstorm3 #119: SEEG: Highlighted electrode contacts not supported with bipolar montages

jeremymoreau, 2018-09-19:
Not a particularly pressing issue, but the feature where the electrode contact is highlighted in red (in the 3D MRI view) when you select a channel only works in average reference. When using a bipolar montage I would expect both channels of the pair to be highlighted when selecting a given bipolar channel. Also, the highlighted contact is sometimes hard to spot because of the default blue-red colour scheme for amplitude. Maybe a different colour for selected contacts would be better (i.e. not red or blue).

ftadel, 2019-03-22:
Finally done in: https://github.com/brainstorm-tools/brainstorm3/commit/91c6e324e707433cdef6297b92d2efa4a44e8906

> When using a bipolar montage I would expect both channels of the pair to be highlighted when selecting a given bipolar channel. 

This works now, but not the reciprocate (if you click on two contacts in the 3D view, it won't select the bipolar channel in the SEEG). It is more complicated to code, but it you think this is a very useful feature, I could still work on it.
 
> the highlighted contact is sometimes hard to spot because of the default blue-red colour scheme for amplitude. Maybe a different colour for selected contacts would be better (i.e. not red or blue).

I proposed the following: red stays the default color for contact selection (because it's what renders the best if you don't have any data displayed on the plain yellow contacts), unless there is data using the RWB colormap, then it uses green.

If you think this could still be improved, let me know. Otherwise, please close this issue.

jeremymoreau, 2019-03-22:
> This works now, but not the reciprocate (if you click on two contacts in the 3D view, it won't select the bipolar channel in the SEEG). It is more complicated to code, but it you think this is a very useful feature, I could still work on it.

Yes, this is great! Thanks! I don't think the opposite (select contact to highlight channel) is as useful since the typical use case is that I would see a spike/event and want to see where that channel localises to.

> I proposed the following: red stays the default color for contact selection (because it's what renders the best if you don't have any data displayed on the plain yellow contacts), unless there is data using the RWB colormap, then it uses green.

I agree, this is a good solution. The green makes the contacts much easier when the RWB colourmap is used. I just noticed an error with the "3D electrodes" SEEG display, though. This only seems to apply to channel file -> Display sensors (head, inner skull, or cortex, but not MRI 3D) and does not impact the displays in Raw file -> SEEG. When I click anywhere, I get this error:
```matlab
***************************************************************************
** Error: Line 79:  select3d (line 79)
** Input argument must be a valid graphics handle
** 
** Call stack:
** >select3d.m at 79
** >figure_3d.m>FigureMouseUpCallback at 664
** >bst_call.m at 28
** >figure_3d.m>@(h,ev)bst_call(@FigureMouseUpCallback,h,ev) at 71
** 
***************************************************************************
```

Sometimes I also get this error after clicking (but not sure how to reproduce):
```matlab
***************************************************************************
** Error: Line 226:  bst_colormaps>GetColormap (line 226)
** Colormap type does not exist.
** 
** Call stack:
** >bst_colormaps.m>GetColormap at 226
** >bst_colormaps.m at 74
** >figure_3d.m>UpdateFigSelectedRows at 2133
** >figure_3d.m at 40
** >bst_figures.m>FireSelectedRowChanged at 1970
** >bst_figures.m>SetSelectedRows at 1925
** >bst_figures.m>ToggleSelectedRow at 1903
** >bst_figures.m at 59
** >figure_3d.m>FigureMouseUpCallback at 770
** >bst_call.m at 28
** >figure_3d.m>@(h,ev)bst_call(@FigureMouseUpCallback,h,ev) at 71
** 
***************************************************************************
```

ftadel, 2019-03-25:
I think this fixes both: https://github.com/brainstorm-tools/brainstorm3/commit/5759cc6530ff9354e32fb30dfb7064fbc55fa2b7
Reopen the issue if it doesn't.

closed 2019-03-25 by ftadel