brainstorm-tools/brainstorm3 #126: Presentation log file

gabri470, 2018-11-09:
Dear all, 

when trying to import Presentation log files, I noticed that event names are read improperly (at least for the file I have). 

here can be my suggested changes. (This works for me)

I will post pull-request

`    % get correct col number
    [~,typeColIdxinHeader] = ismember('Event Type',headerFields);
    [~,codeColIdxinHeader] = ismember('Code',headerFields);
    [~,timeColIdxinHeader] = ismember('Time',headerFields);
    % Split line
    cellLine = str_split(strLine, sprintf(' \t'));
    % If the line contains enough entries: use it
    if (length(cellLine) >= 4) && ~isempty(str2num(cellLine{timeColIdxinHeader})) 
        mrkType{end+1} = strjoin([cellLine(typeColIdxinHeader),cellLine(codeColIdxinHeader)],'_');
        mrkTime(end+1) = str2double(cellLine{timeColIdxinHeader});
    elseif any(~ismember(cellLine,'Trial'))
        break
    end`

martcous, 2018-11-09:
Can you upload the problematic file?

gabri470, 2018-11-09:
[S1-GoNoGo_FB_portcodes_test.log](https://github.com/brainstorm-tools/brainstorm3/files/2566611/S1-GoNoGo_FB_portcodes_test.log)

martcous, 2018-11-09:
At line 2782, there seems to be a new table starting with different columns. Is this normal for a presentation log file?

Line 2782: "Event Type	Code	Type	Response	RT	RT Uncertainty	Time	Uncertainty	Duration	Uncertainty	ReqTime	ReqDur"

martcous, 2018-11-09:
I made the importer more robust, it was assuming the position of the Code and Time columns: https://github.com/brainstorm-tools/brainstorm3/commit/e089de04510dc30332e7285c3488bec0c23f303b

If you split your file in 2 at line 2782, it works properly. See attached. Just import both.
[S1-GoNoGo_FB_portcodes_test1.log](https://github.com/brainstorm-tools/brainstorm3/files/2567308/S1-GoNoGo_FB_portcodes_test1.log)
[S1-GoNoGo_FB_portcodes_test2.log](https://github.com/brainstorm-tools/brainstorm3/files/2567309/S1-GoNoGo_FB_portcodes_test2.log)

gabri470, 2018-11-11:
My suggested edits didn't require to split manually the file. it stops reading the file once it encounters an header-like raw that didn't contained the word 'Trial'. 

I assume that your edit works fine as well. let me know if you think worth for me to send a pull-request.

ftadel, 2018-11-12:
@gabri470 I'll let you fix the code with Martin, but I just have one very generic comment: I recommend avoiding the Matlab syntax "[~," as it is not supported by older versions of Matlab and makes the parser crash. You can easily replace it with a variable name that you don't use (eg. "[tmp," to make it explicit in the code this is not a value you are interested in).

@martcous Can you try adapting your edits to fit better what Gabriele needed?
Thanks

gabri470, 2018-11-12:
Ok I'll replace ~ with tmp and send a pull-request

martcous, 2018-11-12:
Since you both seem convinced that redefining the columns mid-file is normal, I adapted my changes here: https://github.com/brainstorm-tools/brainstorm3/commit/0e329c29e1923efcfa7983b8985fa40aeae70b1c

Does that work for you, Gabriele?

gabri470, 2018-11-14:
Dear @martcous, 

your fix it still doesn't work in my case unless I manually split the file. I have 750 go and 250 nogo events marked in my file. Your fix read 1500 Go tag and 500 noGo from file. 

This happens because your code continues reading the second block of data which do not holds information about time of the events as far as I understood.

All the best,

martcous, 2018-11-14:
Hi Gabriele,

I am confused as to why you are saying it should only have 1000 total events, when the file you uploaded has 2000 (1500 Go and 500 NoGo). Like I mentioned earlier, your file seems to have two separate blocks, with a header redefinition midway through. Are you saying we should always ignore the second block? Do you understand why your file has two blocks? Other examples do not redefine the header later in the file.

gabri470, 2018-11-14:
Hi @martcous ,

the trials are only 1000 (100% sure about this) In fact if you look at second block, Time column for each event contains the same information between the two blocks.

martcous, 2018-11-14:
I see, so the second block is a perfect duplicate of the first one. This does not however answer our question of why there is one or whether we should always ignore it.

gabri470, 2018-11-14:
Indeed it doesn't but that is the only file type I have. Is there anyone else that can provide a different file ? also documentation in manufacturer website isn't helping much to understand.

gabri470, 2018-11-15:
Sure I will asap.

martcous, 2018-11-21:
@gabri470 bump

closed 2018-11-28 by ftadel

gabri470, 2018-12-03:
Sorry for the very late reply and not reporting about the above issue more quickly. It works now. Thanks for the excellent job!