brainstorm-tools/brainstorm3 #132: STA 2D-display shows wrong time selection

mpompolas, 2018-11-16:
This should just be an issue of the 2D layout visualization.

In order to compute the STA, a time window needs to be selected around each spike e.g. [-150,150] ms.

After the computation, the LFP signals are [-150,150]. 

However, when the STA is displayed on the 2D layout, the time window (bottom left values on the 2D-layout window) is shown as [-300, 300] ms.

martcous, 2018-11-19:
@ftadel The following line doubles the chosen time window:
https://github.com/brainstorm-tools/brainstorm3/blob/103b95937cecb3371250c1daf6e9aeb2fbbd398e/toolbox/gui/figure_topo.m#L685
Is that intended?

martcous, 2018-11-19:
Screen capture of bug: ([-150, 150]ms vs [-300, 300]ms)
[image]

ftadel, 2018-11-21:
I've been wanting to take care of this for a long time...
I changed the logic of this time window: before it was representing the duration before and after the current time point (regardless of the actual time points represented in the figure), now it represents the actual time window that is represented.
Commit: https://github.com/brainstorm-tools/brainstorm3/commit/27a5c2810cb2e3396c57f9f0a8def65195a401a4

I changed a lot of things quickly, so it might be a bit buggy.
Test it on your data, and you think the new behavior is acceptable, please close the issue.

martcous, 2018-11-21:
Thanks Francois, it's working fine!

closed 2018-11-21 by martcous