brainstorm-tools/brainstorm3 #134: Band-pass filter transition bands are too large?

ftadel, 2018-11-28:
Sylvain:

> We’re having issues with the filters produced in Brainstorm.
Please see attached screenshot of the filter properties GUI. When requesting a 0.5-100Hz bandpass, the high cutoff is actually at 107Hz. 
> It was verified when applied on data as well.   

[image]

ftadel, 2018-11-28:
Hossein:

> I looked into the code and noticed why it happens. 
> As you know, to find the filter coefficients we need four frequency parameters. 
> 
> fcuts = [f_highstop, f_highpass, f_lowpass, f_lowstop];
> 
> f_highpass and  f_lowpass are set by users and two others, which determine the transition bands, are estimated in our code based on the type of the filter and input. These are the codes for those two variables: 
> 
> if (HighPass <= 5)   % Relax the transition band if HighPass<5 Hz
>     f_highstop = .5 * f_highpass;
> else
>     f_highstop = .85 * f_highpass;
> end
> 
> And 
> 
> if f_highpass==0    % If this is a low-pass filter
>     f_lowstop  = 1.05 * f_lowpass;
> else
>     f_lowstop  = 1.15 * f_lowpass;
> end
> 
> I guess both of these were considered to relax the transitions bands in order to avoid high filter orders and reduce the actual edge effect. However, It seems very loose for f_lowstop as we have seen here, especially when we design a bandpass filter the coefficient is 1.15. 
> 
> If we simply replace 1.15 with 1.05, we will have the following specifications, which looks good: 
> 
> image.png
> 
> As we see, the edge effect has increased a bit while the filter order is constant. Personally, I think we can have smaller coefficients even for bandpass filters but please consider that there might be other situations that very tight transition bands make some problems (like lower frequencies) 
> 
> As we discussed previously, the highpass filter (.5HZ>) causes the high filter order but with near zero edge effect. In contrast, the lowpass filter is responsible for almost all the edge effect with a small contribution to the filter order. 
> 
> So, please let me know how we can address this problem from your point of view. We can do a couple of things: 
> 
> 1. Simply change 1.15 to something smaller. 
> 2. Design the coefficient (or High transition) based on the actual value of  f_lowpass. 
> 3. Ask the user to decide, like another parameter in the process function. 
> 
> I can change the function, based on what you and others suggest, very promptly. (with Martin and Francois help :))

ftadel, 2018-11-28:
Sylvain:

> Thank you, Hossein: very appreciated:
> 
> When designing a low-pass filter, for instance a bandpass of 0-100 Hz as below, the transition band indicated is100-105Hz and the attenuation actually starts immediately after 100Hz - this is fine.  
> [image]
> 
> Now with a 0.05-100Hz bandpass, the high transition printed as 100-115Hz but from the frequency response, it is much sharper than with the lowpass (good) but actually starts at 107.5Hz (bad/inconsistent with indication of 100-115Hz). The transition is so sharp that it could start at 100Hz with great efficiency.
> [image]
> 
> is there a way to reconcile the high-transition numbers shown with the actual frequency response plotted?

HosseinShahabi, 2018-12-31:
Due to the mentioned problem, we have substantially updated the bandpass filtering paradigm. 

**Briefly, we use cascade filtering (Kaiser for highpass and Equiripple for lowpass). Additionally, we have introduced a new parameter to control the upper transition band by the user. These enhancements will completely resolve the discussed issues.** 

The previous filtering mechanism is still available through the process function. Filter descriptions are also updated. 

**Here is the new process_bandpass:** 
[image]

**And the newly developed filter looks like:**
[image]

Comparing with the old implementation, 
[image]

we can observe a substantial improvement.

Two functions are updated/created at this time "process_bandpass" and "bst_bandpass_hfilter_cascade". However, the filtering functions are utilized in several other functions, so all of those should be updated. I have searched other files using "process_bandpass" and noticed some of them even use our oldest version of filtering (fft-filter) which should be replaced by new filters. 

The only concern with the newly developed approach is that the Octave functions are not available for Equiripple filters. So, when the user doesn't have the Signal Processing Toolbox, the code produces Errors/Warnings and uses the former (Oct 2016-18) method. 

@sbaillet @richardmleahy Please take a look and let me know if there is any problem. 

@martcous @ftadel Please tell me how to upload the codes for review. Is it a commit or pull request? 

Thanks

ftadel, 2019-01-02:
> Here is the new process_bandpass

Maybe make this simpler with an option "radio_linelabel": "Filter implementation: 2014 / 2016 / 2019"

The various versions should be documented online:
https://neuroimage.usc.edu/brainstorm/Tutorials/ArtifactsFilter#Filter_specifications:_Low-pass.2C_high-pass.2C_band-pass

> Two functions are updated/created at this time "process_bandpass" and "bst_bandpass_hfilter_cascade". However, the filtering functions are utilized in several other functions, so all of those should [...]

Please suggest modifications for all of the calls to process_bandpass.

> Please tell me how to upload the codes for review. Is it a commit or pull request?

Please open a PR so we can review the code/comment/fix before merging.

HosseinShahabi, 2019-01-03:
> Maybe make this simpler with an option "radio_linelabel": "Filter implementation: 2014 / 2016 / 2019"

I did. 

> The various versions should be documented online:
> https://neuroimage.usc.edu/brainstorm/Tutorials/ArtifactsFilter#Filter_specifications:_Low-pass.2C_high-pass.2C_band-pass

Sure. I will update when you finalize the code.

> Please suggest modifications for all of the calls to process_bandpass.

Should I open PRs for all of them? 

> Please open a PR so we can review the code/comment/fix before merging.

https://github.com/brainstorm-tools/brainstorm3/pull/154

ftadel, 2019-02-25:
Done with PR: https://github.com/brainstorm-tools/brainstorm3/pull/177

closed 2019-02-25 by ftadel