brainstorm-tools/brainstorm3 #140: Fix the dSPM averaging

ftadel, 2018-11-29:
Scaling of dSPM values for averaged files (ERP/ERF) by the number of trials was added in April 2015:
https://neuroimage.usc.edu/brainstorm/News#April_2015

It was then reverted it in this commit:
https://github.com/brainstorm-tools/brainstorm3/commit/4b32f0fc1e3272e27ca45c0d979368324588ff05
It is now documented in this tutorial:
https://neuroimage.usc.edu/brainstorm/Tutorials/SourceEstimation#Averaging_normalized_values

But now Dimitrios suggested to reimplement it again, maybe under a different format. Some important decisions have to be made here, as it impacts dramatically the output results.

ftadel, 2018-11-29:
Last messages exchanged on this topic.

Dimitrios: 
> Attached slides: [Frontiers dataset - Dimitrios Pantazis.pptx](https://github.com/brainstorm-tools/brainstorm3/files/2628852/Frontiers.dataset.-.Dimitrios.Pantazis.pptx)
>
> A summary of some findings:
> -The |Faces| - |Scrambled| contrast works well for sources
> -Applying z-score after low-pass filtering leads to noisy maps. It should be applied before.
> -I think dSPM maps produce better results that min-norm (more sensitive). (Results should be the same if we apply min-norm + z-score, though I did not test)
> -We may need a better colormap than the red-blue-white
> -I reproduced the MNE paper results reasonably well but with FDR thresholds
> -Cluster size inference failed to produce good clusters (unlike the MNE paper). I used the fieldtrip cluster-size test implemented in brainstorm. It took over 10 hours and I don’t know why the results don’t match the MNE paper ones.
> -Critically, the |Faces – Scrambled| contrast is good for sensors, but does not appear to be good for source maps. In particular, the maps appear ok but the Chi-squared statistic fails miserably and I could not find a way to make it work (everything is significant). I think we seriously need to consider removing it from our recommendations. Which is unfortunate because I suggested it before. It was the only option I could think of to perform statistics on a single condition case (when there is nothing to permute). This means we do not have a statistical procedure for a single condition, but the Chi-squared does not work anyway. In a single condition case, researchers should use condition vs. baseline (thus making it a 2-condition test) and use |condition| - |baseline| for sources.

Sylvain:
> > -The |Faces| - |Scrambled| contrast works well for sources
> 
> Qualitatively, the maps you produced with dSPM or MNE 2-condition comparisons are consistent with one another: there is no major unexpected difference between them, with the peaks of activity similar regions. The only differences are in the threshold cut. This is good.  
> 
> > -Applying z-score after low-pass filtering leads to noisy maps. It should be applied before.
> 
> I think this is because LPF reduces the signal variance tremendously, hence an over estimation of the z-scores. The zscore map you show on slide#2 would probably require a higher cutoff. 
> 
> > -I think dSPM maps produce better results that min-norm (more sensitive). (Results should be the same if we apply min-norm + z-score, though I did not test)
> 
> Would be good to test. I would say dSPM is more specific rather (less spurious activity). Is that what you meant? 
> 
> > -We may need a better colormap than the red-blue-white
> 
> Agreed. I like the jin one you have tried.
> 
> > -I reproduced the MNE paper results reasonably well but with FDR thresholds
> 
> Agreed.
> 
> > -Cluster size inference failed to produce good clusters (unlike the MNE paper). I used the fieldtrip cluster-size test implemented in brainstorm. It took over 10 hours and I don’t know why the results don’t match the MNE paper ones.
> 
> This is something we need to understand. 
> 
> > -Critically, the |Faces – Scrambled| contrast is good for sensors, but does not appear to be good for source maps. In particular, the maps appear ok but the Chi-squared statistic fails miserably and I could not find a way to make it work (everything is significant). I think we seriously need to consider removing it from our recommendations. Which is unfortunate because I suggested it before. It was the only option I could think of to perform statistics on a single condition case (when there is nothing to permute). This means we do not have a statistical procedure for a single condition, but the Chi-squared does not work anyway. In a single condition case, researchers should use condition vs. baseline (thus making it a 2-condition test) and use |condition| - |baseline| for sources.
> 
> I don’t understand the logic of applying chi-squared here because it is a 2-condition  study design on the first place. What do you see if you apply chi2 stats on abs(Faces) only for instance?  
> Another thing I don’t understand is why you z-scored the  abs(dSPM(Faces-Scrambled))) before applying chi2. I thought you said that zscoring abs maps is not recommended/wrong.  

Dimitrios:

> > “Would be good to test. I would say dSPM is more specific rather (less spurious activity). Is that what you meant?”
> 
> For the min-norm maps, there are 3 cases:
> 1) no z-score is applied to the min-norm maps. In that case I found less detected activity than for dSPM maps (thus less sensitive; this is what I meant in the above sentence).
> 2) z-score is applied before low-pass filtering. In that case I expect results similar to dSPM
> 3) z-score is applied after low-pass filtering, in which case we saw overestimated z-values and non-specific significance maps.
> 
> > “I don’t understand the logic of applying chi-squared here because it is a 2-condition  study design on the first place. What do you see if you apply chi2 stats on abs(Faces) only for instance?  “
> 
> Also, regarding the chi-squared test logic: In this study we have paired measurements, i.e. every participant was measured for the faces and scrambled conditions. Thus we can construct the difference: faces – scrambled and this becomes equivalent to a one-sample test across participants. Therefore, in that sense the test becomes equivalent to a one condition experiment [... cut ...]
> https://www.sciencedirect.com/science/article/pii/S1053811916303470
> (but I don’t agree entirely with this paper and their prevalence inference arguments)
> 
> At least for decoding we have proper statistical procedures, whereas the Chi-squared test seems very approximate.
> 
> My feeling is that we should be encouraging contrasts like Faces – Scrambled in the sensor domain, and |Faces| - |Scrambled| in the source domain. This may seem awkward at first, but it is mostly common practice. And take away the chi-squared test (sorry Francois).

jcmosher, 2018-12-12:
Updating the conversation from an email Francois sent.

From: Francois Tadel, Mr <francois.tadel@mcgill.ca> 
Sent: Tuesday, November 6, 2018 8:28 AM
To: Dimitrios Pantazis <pantazis@mit.edu>; John Mosher <john@emosher.net>; Sylvain Baillet, Dr <sylvain.baillet@mcgill.ca>; Richard Leahy <leahy@ee.usc.edu>
Cc: Martin Pierre Cousineau, M <martin.p.cousineau@mcgill.ca>
Subject: Re: averaging in brainstorm

Hi Dimitrios,

Thanks for the summary.

nAvg is the number of trials averaged, nothing else, it would be very confusing to start saving another type of data in it. This "n" prefix indicates an integer, we should not put any floating point value in a variable named this way. If you need another value that cannot be recomputed directly from the existing nAvg, then you should save it in a new variable.
How would you like this new variable to be called?

Reverting the last modifications (that removed this dSPM scaling factor) and adding this new variable in all the functions handling recordings and source maps requires a substantial amount of coding, and intense testing to make sure we don't break anything. 
I'd like to get the explicit approval of all of you (Sylvain, John, Richard) before proceeding.

Sylvain just told me we were supposed to submit the Brainstorm article by Friday. This requires us to upload the current version of Brainstorm on a permanent storage (I think I'll go for Zenodo), and the link to this version will be the one published together with the article, for the strict reproducibility of the results presented in the article in the future. We won't be able to update the code after. 
Do we want me to publish the current version (that has no scaling at all for the dSPM averages), or do you think you have time to do this massive update with me first (including code reviewing and independent testing on your own data) before Friday? 
(knowing I won't be able to work much on Friday - I'll need to wrap up the distribution before 6pm my time on Thursday)

François

jcmosher, 2018-12-12:
If we set aside the non-integer question for a moment, let's look at Francois' definition: "nAvg is the number of trials averaged, nothing else"

The problem I had encountered originally is the following: I averaged 20 files together to get the "A" data set "AvgA". Properly, the field for the number of averages, AvgA.nAvg, is set to 20. I similarly do another 20 files into another data set, and properly, AvgB.nAvg is also set to 20. No problem.

But then when I average these two files, A and B, I find that incorrectly the number of averages is set to two, Avg_AandB.nAvg = 2, because Brainstorm thought I only averaged two files. It should be properly 40 (20 + 20), if I am to still correctly apply measures that require the true number of averages, such as noise whitening, or dSPM. I believe it was this basic error that started Dimitrios down the path of trying to remove the nAvg from the dSPM calculation.

Now if set A has 30 averages and set B has 20, then the "incorrect way" to average these two files is (A + B) /2, i.e. weights of 1/2 applied to each file.  The correct way is a weighted average, weight 1 is (30/50) for A and weight 2 (20/50) for set B, with the nAvg of the result set to 50. So we have the conflation of needing to do **weighted** averages by default, and also setting the nAvg right for the weighted average of the two files.

Matti's elegant formulas shown in the PDF page earlier in this comment stream are for general and arbitrary weights, which could lead to Leff (his variable for the effective number of averages) being non-integers, and quite appropriately so. His equations are quite elegant for arbitrary weights, which could have utility in more sophisticated averaging schemes, but I think there was an earlier discussion that the user should be guided to the correctly weighted averaging discussed here.

If we stick to "properly" weighted averages, where the weights are determined by the underlying number of trials, e.g. w_1 = 30/50 and w_2 = 20/50, **then the new effective average will remain an integer, as Matti shows, and which is one of Francois' concerns that nAvg remain an integer.**

So we would only need a new non-integer Leff, if we decide to use other weights; however, the default weighted average should be just fine. When averaging multiple data files

1) set nAvg_new = sum(nAvg of the individual files).
2) set the weight for each "ith" file as w_i = nAvg_i / nAvg_new
3) sum over all files, w_i * data_i, to get data_new, the properly weighted average of the individual data sets.

Then, as Dimitrios notes, we use sqrt(nAvg_new) in our calculations of whitening, z-stats, dSPM. The critical assumption, which so far remains valid in our workflows, is that the noise covariance was calculated from the unaveraged data (nAvg == 1), so that averaging the data reduces the variance of the noise by the number of averages, and therefore the z-stats are increased by the square root of the number of averages. We could get cute with nAvg not equal 1 in the noise covariance matrix, but I don't recommend that, if only for simplicity.

ftadel, 2018-12-18:
>But then when I average these two files, A and B, I find that incorrectly the number of averages is set to two, Avg_AandB.nAvg = 2, because Brainstorm thought I only averaged two files. It should be properly 40 (20 + 20), 

You simply have to select the option "Weighted average" in the averager options to get nAvg = nAvgA+nAvgB.

> I think there was an earlier discussion that the user should be guided to the correctly weighted averaging discussed here.

Weighted average for averaging run-level averages into a subject-level average.
Regular average for averaging subject-level averages into a group-level average.
https://neuroimage.usc.edu/brainstorm/Tutorials/Workflows#Constrained_cortical_sources

ftadel, 2019-03-25:
Reminder: You haven't made any choice of implementation yet.
This is a critical decision that affects deeply all the dSPM measures computed with Brainstorm.
It should be fixed ASAP.

This is causing some confusion. See for instance this post:
https://neuroimage.usc.edu/forums/t/dspm-normalization-and-workflow/10498

dimitriospantazis, 2019-05-20:
John and Francois,

As you point out above, there are two types of averaging in brainstorm, weighted average and regular average. Both are needed in different circumstances, such as subject-level versus group-level averages. Subject-level analysis may require regular average (if we want all conditions to contribute equally to the average) or weighted average (if we want all trials to contribute equally). Group-level analysis typically requires regular average so that all subjects contribute equally for random-effects inference. Importantly, the degrees of freedom are not only for averages. Users may subtract averages, e.g. in a typical mismatch negativity paradigm where we compute standard minus deviant responses.

My point is that we need to properly compute the effective degrees of freedom in *all* cases that we linearly combine trials (or source maps), as in y(t) = sum_i ( w_i*x_i(t) ). My post on Nov 29 (email Nov 6) addresses this issue. There should be dSPM maps for regular and weighted averages, not just for one case.

I propose we create a new variable to track the effective number of trials. We can call this variable ‘eff_nAvg’ to differentiate from the existing ‘nAvg’. I suggest we keep the ‘nAvg’ as it is for possible compatibility issues and to avoid breaking any code, but introduce the ‘eff_nAvg’ to all trial and source files. After this variable exists, it will become feasible to properly normalize the dSPM maps in all circumstances. I don’t see any issue if the ‘eff_nAvg’ is non-integer, it should not be confusing.

Best,
Dimitrios

ftadel, 2019-05-28:
@jcmosher @dimitriospantazis 
I tried to address your requests in this pull request:
https://github.com/brainstorm-tools/brainstorm3/pull/212

There are many pending questions. Please try to review it soon, as it will make it difficult for me to update these functions for other purposes before I merge this into the main distribution.

@dimitriospantazis The "Case B" in your first message is not taken care of anywhere, it feels to me this doesn't match what you wrote in the online tutorial, that a when this function "Apply dSPM scaling" is applied, the file should not be used for any other purpose after. Is this correct?

ftadel, 2019-07-25:
Done: https://github.com/brainstorm-tools/brainstorm3/pull/212

closed 2019-07-25 by ftadel