brainstorm-tools/brainstorm3 #144: Filter specifications: Notch and and band-stop

ftadel, 2018-11-29:
Hossein, Richard:
- Add a button "View filter response" to the notch and band-stop processes 
- Create "transient" events in the filtered files, as it was done for the band-pass filter
- Update the section "Filter specifications" of the tutorial: http://neuroimage.usc.edu/brainstorm/Tutorials/ArtifactsFilter#Filters_specifications_.5BTODO.5D

Francois: 
- Update section "Apply a notch filter" to add the transients in screen capture and text (after the documentation of the notch filter)

HosseinShahabi, 2018-12-18:
I want to draw your attention to the current band-stop filter in Brainstorm.

This filter is an 8 order Butterworth filter which mostly likes a wideband notch filter than a bandstop (1-bandpass). I've attached the magnitude and impulse response of the filter (designed for a bandstop [.2 .4]). 
[image]
[image]

The edge effect based on 99% energy is around 20 samples for this case. Please note that the edge effect will be doubled since we use "filtfilt". Here is my question: 
Do we want to keep this configuration or change it?

HosseinShahabi, 2019-01-03:
Dear @ftadel 

I edited the process_bandstop by adding necessary codes and filter response panel. 
I will PR the code (Single function in this case). Once you approved the code, I'll edit the Tutorial. 
[image]

The process_notch needs further work since we are modifying the algorithm.

HosseinShahabi, 2019-01-09:
Both are set. 
https://github.com/brainstorm-tools/brainstorm3/pull/158

ftadel, 2019-01-15:
Have you modified the code for the notch and bandstop filters?
If so, it would be good to discuss it with @sbaillet @richardmleahy @jcmosher and @dimitriospantazis before pushing them to the Brainstorm distribution.
To be added to the agenda of a biweekly meeting?

sbaillet, 2019-01-15:
that was indeed the plan.

Best wishes,

Sylvain.
-
Sylvain Baillet, PhD
Professor, Montreal Neurological Institute
McGill University
> more<https://goo.gl/591JdQ>

ftadel, 2020-11-09:
Fixed with the PR 177

closed 2020-11-09 by ftadel