brainstorm-tools/brainstorm3 #145: Multi-electrode array support for Invasive Neurophysiology

mpompolas, 2018-11-29:
Linear probes are perfectly supported. Users select the tip of the probe and the skull entry and Brainstorm takes care of the position assignment.

[image]

-------------------------------------------------------------------------------------------------------------------
However, when multi-electrode ARRAYS (let's say a 10x10 array) are used, a new option should be introduced. Manufacturers give a diagram for each array they sell, that contains the electrode positioning on the array (the electrodes' indexing doesn't follow the traditional numbering 1,2,3,4... but is messed up based on how the wiring is implemented in the array).

So for this to work in Brainstorm:

Users should be able to:
1. Select the size of their array (10x10, 8x8 etc...)
2. Select a position on the cortex where the implant will be attached
3. Manually add the indices of the electrodes on the array.

I already have code that does all these things, but it needs to be implemented in the main iEEG tab.
Check: https://github.com/mpompolas/e-Phys/blob/master/TUTORIAL_channel_manipulation_for_arrays.m

The idea behind this algorithm, is that it takes 3 points on the CORTICAL surface for each array and based on these 3 points it create a plane. Then the algorithm takes the 2D array positioning and converts it to an orthonormal base that aligns it on the plane that was computed.

ftadel, 2018-12-20:
What you describe looks like what was already available with the ECOG (defining the full grid based on 3 points + spacing, or 4 points)?
(except for the random naming order)

Are you expecting anything from me or Martin, or was it just your personal to-do list?

mpompolas, 2018-12-20:
Even better since it would be easier to implement.

I wish I had time to implement all the features that I have in mind on the toolbox but realistically I can't dedicate more effort in this. I have to finish this life-sucking PhD.

Can either of you add support for the arrays please?

ftadel, 2018-12-21:
I might work a bit on ECOG next year, but not right now. 
@martcous interested in diving into intracranial display?

We need a proper tutorial instructions + test dataset for this, otherwise I'd have no chance to understand what is going on in your function.

martcous, 2018-12-21:
I can work on it next month but I will need guidance. Want to meet in the new year to get me up to speed on this aspect of the software?

sbaillet, 2018-12-21:
Good idea: we could discuss this at our next MEG core meeting Tues Jan 08 @ 3pm - would that work for your @mpompolas ?

martcous, 2018-12-21:
I meant a meeting with @ftadel to discuss the technical aspects this would entail.

ftadel, 2018-12-21:
I'm not sure, as don't know how what is available for ECOG relates with what Konstantinos does in his script. The ECOG part in this iEEG tab still has some bugs I think, for a lack of example datasets.

A good starting point is the SEEG tutorial:
https://neuroimage.usc.edu/brainstorm/Tutorials/Epileptogenicity

mpompolas, 2018-12-21:
@sbaillet @martcous  I can meet you at 3:30 on Tuesday the 8th since I have a lab meeting right before. 
Let me know if you want my input in this.

sbaillet, 2018-12-22:
@martcous: we'll let you decide if you want to meet.

ftadel, 2020-11-09:
No activity...

closed 2020-11-09 by ftadel