brainstorm-tools/brainstorm3 #165: Bugs with the colormap option "Use stat threshold"

ftadel, 2019-01-28:
New option implemented in: https://github.com/brainstorm-tools/brainstorm3/pull/162

Things to be fixed or added:
- 2D/3D topographies: apply different threshold on the values
- Time-frequency maps
- Time-frequency files displayed as 2D/3D topographies
- All types of files displayed with a menu "View as image"

thomas-vincent, 2019-02-16:
I can have a look at these in March.

For the interpolated map that you showed:
[image]

I guess the behavior is normal since there is some smoothing. I'll try to propose a better representation.

ftadel, 2020-11-09:
@thomas-vincent 
Please re-open the issue if this is something you still want to work on.

closed 2020-11-09 by ftadel