brainstorm-tools/brainstorm3 #166: Kilosort download fail?

mpompolas, 2019-01-29:
Very weird behaviour on a clean install of Kilosort.

The installation of Kilosort depends on 4 GitHub zips.
There are checks for each one if the file to be downloaded is present.
However, although the urls are correct, the code fails to pull the files.

But if you try to download each file again and again, at some point it works, without any changes on the code!

In my understanding this is not a Brainstorm issue, but mostly Matlab-GitHub communication problem.
Do you guys have any insights on this?

martcous, 2019-01-29:
This is probably related to the problem other users have been having with the updater... It's becoming quite an annoying and unpredictable bug. What is your Matlab, Java and OS versions?

In the meantime, it is possible to download each GitHub zip yourself and put them under .brainstorm/kilosort_tmp/ for the installation to work properly.

mpompolas, 2019-01-29:
Yeah, once it's download it, it's done.
Already did it.

Matlab R2016b
Windows 7
Java Version 8 Update 161 (build 1.9.0_161-b12)

jschmiedt, 2019-04-17:
I'm having the same problem on Windows 10, Matlab R2018b with both Kilosort and UltraMegaSort:

```
***************************************************************************
** Error: [process_spikesorting_kilosort]  Electrophysiology > Unsupervised Spike Sorting > KiloSort
** Line 564:  process_spikesorting_kilosort>downloadAndInstallKiloSort (line 564)
** Impossible to download KiloSort.
** Error: File is empty.
** 
** Call stack:
** >process_spikesorting_kilosort.m>downloadAndInstallKiloSort at 564
** >process_spikesorting_kilosort.m>Run at 123
** >process_spikesorting_kilosort.m at 33
** >bst_process.m>Run at 229
** >bst_process.m at 36
** >panel_process1.m>RunProcess at 151
** >panel_process1.m at 26
** >gui_brainstorm.m>CreateWindow/ProcessRun_Callback at 759
** >bst_call.m at 28
** >gui_brainstorm.m>@(h,ev)bst_call(@ProcessRun_Callback) at 292
** 
** 
** File: monkeyB/@rawytu288c-01/data_0raw_ytu288c-01.mat
** 
*********
```
```
***************************************************************************
** Error: [process_spikesorting_ultramegasort2000]  Electrophysiology > Unsupervised Spike Sorting > UltraMegaSort2000
** Line 272:  process_spikesorting_ultramegasort2000>downloadAndInstallUltraMegaSort2000 (line 272)
** Impossible to download UltraMegaSort2000:Error: File is empty.
** 
** Call stack:
** >process_spikesorting_ultramegasort2000.m>downloadAndInstallUltraMegaSort2000 at 272
** >process_spikesorting_ultramegasort2000.m>Run at 115
** >process_spikesorting_ultramegasort2000.m at 33
** >bst_process.m>Run at 229
** >bst_process.m at 36
** >panel_process1.m>RunProcess at 151
** >panel_process1.m at 26
** >gui_brainstorm.m>CreateWindow/ProcessRun_Callback at 759
** >bst_call.m at 28
** >gui_brainstorm.m>@(h,ev)bst_call(@ProcessRun_Callback) at 292
** 
** 
** File: monkeyB/@rawytu288c-01/data_0raw_ytu288c-01.mat
** 
***************************************************************************
```

mpompolas, 2019-04-17:
Yeah it's the same issue. Matlab tries to download the files from GitHub and it fails.
You can try to do it a few times and it finally works!!!

I'll add a section to try multiple times until it reaches a timeout.

mpompolas, 2019-04-25:
Added repeated attempts for downloads. Although not the cleanest solution, it works.
Please update Brainstorm and try again @jschmiedt 
(Simply use the menu "Update > Update Brainstorm" to get the fixes)

afalaki, 2019-06-03:
Recently, I also tried to run Kilosoft spike sorter, but encounter a problem, which is probably different than above mentioned ones, while simply following the tutorials. Please note the error at the end. I highly appreciate your suggestions.

** Error: [process_spikesorting_kilosort]  Electrophysiology > Unsupervised Spike Sorting > KiloSort
** Line 258: Index in position 2 exceeds array bounds (must not exceed 1).
** 
** Call stack:
** >process_spikesorting_kilosort.m>Run at 258
** >process_spikesorting_kilosort.m at 33
** >bst_process.m>Run at 229
** >bst_process.m at 36
** >panel_process1.m>RunProcess at 151
** >panel_process1.m at 26
** >gui_brainstorm.m>CreateWindow/ProcessRun_Callback at 759
** >bst_call.m at 28
** >gui_brainstorm.m>@(h,ev)bst_call(@ProcessRun_Callback) at 292
** 
** 
** File: Floyd/@rawytu288c-01/data_0raw_ytu288c-01.mat

ftadel, 2019-06-04:
@martcous @mpompolas ?

ftadel, 2020-04-27:
?

ftadel, 2020-11-09:
Is this still an issue?
Will this move forward at some point?

mpompolas, 2020-11-10:
Just downloaded Kilosort and run it on a few protocols.
Multiple attempts to download Kilosort and dependencies from GitHub are still happening, but ultimately everything is downloaded.

I spike-sorted 3 datasets and had no issues.

I think what @afalaki experienced is dataset specific.

ftadel, 2020-11-10:
> Just downloaded Kilosort and run it on a few protocols.
> Multiple attempts to download Kilosort and dependencies from GitHub are still happening, but ultimately everything is downloaded.

Thanks for checking!

This download from github issue is annoying in many parts of the software...
I added a switch in gui_brainstorm('DownloadFile'): when downloading from github, it will use websave or urlwrite instead:
https://github.com/brainstorm-tools/brainstorm3/commit/7cd0f62f684f7aab1dc73e341cda68727b79af13

Please re-open the issue if this is not a satisfying solution.

closed 2020-11-10 by ftadel