brainstorm-tools/brainstorm3 #169: PAC: Using old Bandpass

ftadel, 2019-01-30:


ftadel, 2019-01-30:
bst_pac.m uses bst_bandpass_fft.m, which is not used anywhere anymore... And it uses a signal mirroring, which has proven not to help much reducing the duration of the edge effects.
@SoheilaSamiee @sbaillet If this is really what you want to do, please simply close this issue.

SoheilaSamiee, 2019-01-30:
Hi Francois,

bst_pac.m is the function from Canolty's paper, which got implemented in brainstorm more than 4 years ago. Probably the reason that it still uses bst_bandpass_fft.m is that no one changed it. As far as I know there should not be a particular reason behind using the old version of bandpass filter in this process.

ftadel, 2020-04-27:
Would you still like to update the filters in your PAC functions?

sbaillet, 2020-04-27:
Please hang on as we are in the process of producing faster versions of PAC functions. For now, I am in favor of keeping the current versions as such, for legacy purposes.

Thanks,

Sylvain.

ftadel, 2020-11-09:
@sbaillet @SoheilaSamiee 

Can you please close this issue if you think this is not an issue?
Thanks

ftadel, 2021-09-19:
Any update on this front?

ftadel, 2022-04-18:
@sbaillet @SoheilaSamiee 
Any update?
If you are satisfied with the filters used in the current function, we will simply close this issue.

ftadel, 2022-06-08:
@sbaillet @SoheilaSamiee
Can you please let me know whether you want to keep this issue open or not?
Thanks

SoheilaSamiee, 2022-06-17:
Hi Francois (@ftadel )

Sorry for my late reply, this is a long time that I am not working with brainstorm. I think Sylvain (@sbaillet ) would be the best person to answer this question.

Thanks a lot for your continuous support,
Soheila