brainstorm-tools/brainstorm3 #172: File selection in Process1: added time tags selection doesn't differentiate between positive and negative values

mpompolas, 2019-02-06:
Came across this weird selection from the filter selection. 
I added some tags (-350, 0) and (0,350) ms on my files.

When I try to make the selection with the filter based on the "**Search Comments**" field and add **(0, 350)**, it selects BOTH which is problematic:
[image]

If I add the negative time selection in the filter: **(-350, 0)** it works fine:
[image]

martcous, 2019-02-06:
The Comment Filter works with boolean logic, which includes reserved characters such as parentheses, and/&/&& and or/|/||. If you want to include a reserved character in your search without making use of the boolean logic operators, add strings to your query. In this case, "(0, 350)" would work properly and only select your first file.

closed 2019-02-06 by martcous

mpompolas, 2019-02-07:
This is neat. Didn't even know there was a boolean filter.
I assume it was on the latest updates.

Either way, thanks