brainstorm-tools/brainstorm3 #181: NWB initialization error

mpompolas, 2019-03-09:
**CHECK THIS ON MY PULL REQUEST / AFTER YOU MERGE**

_Installation of NWB SDK fails._

in_fopen_nwb automatically downloads and extracts the files needed, and then it needs to call the initialization function for the SDK ("generateCore")

However the function fails to initialize the SDK. 

To my understanding this is a java issue since I can install the downloaded files once I close Brainstorm.

@ftadel can you please take a look at it?

Thanks

mpompolas, 2019-03-13:
Warning: Objects of org/brainstorm/file/BstFileSelector class exist - not clearing
java 
> In javaclasspath>local_javapath (line 190)
  In javaclasspath (line 119)
  In javaaddpath (line 71)
  In generateExtension (line 26)
  In generateCore (line 22)
  In in_fopen_nwb>downloadAndInstallNWB (line 401)
  In in_fopen_nwb (line 60)
  In in_fopen (line 144)
  In import_raw (line 126)
  In bst_call (line 28)
  In tree_callbacks>@(h,ev)bst_call(@import_raw,[],[],iSubject) (line 613)

ftadel, 2019-03-13:
This is due to known issues of the management of the dynamic java class path in Matlab.
Adding a new .jar to the java class path (needed by generateCore) first clears all the existing java objects, which is not working if some have already been instanciated (eg. when Brainstorm was already started). The warnings you reported are due to clear the java objects. If the "clear java" calls does not work, the new .jar is not added. And the functions that need them (the rest of the NWB installer) crash.

There are technical solution we could implement in Brainstorm but they are quite cumbersome would probably not transfer well to the compiled distribution. For future reference if we ever need this:
https://stackoverflow.com/questions/3580752/java-dynamically-loading-a-class/3581598#3581598
https://stackoverflow.com/questions/4434209/matlab-finds-persistence-xml-for-eclipselink-only-within-the-first-few-seconds/4436200#4436200

The best solution would be to add all the needed .jar files need by the NWB library at the Brainstorm startup, something similar to what @martcous has done recently with the mffmatlabio library:
https://github.com/brainstorm-tools/brainstorm3/blob/master/brainstorm.m#L104
https://github.com/brainstorm-tools/brainstorm3/blob/5abee5220fb9dfa4cdb67e4dea84b1e2eff8df0e/toolbox/io/in_fopen_mff.m#L104

The installer should do the following: 
- download and unzip the library
- ask the user to restart Matlab
- when Brainstorm is restarted: finish the installation by calling generateCore.

Not trivial, but doable.

mpompolas, 2019-03-22:
Done

ftadel, 2019-03-25:
Works after: https://github.com/brainstorm-tools/brainstorm3/commit/280817699704698d659468d6174bcd9f81832c12

closed 2019-03-25 by ftadel