brainstorm-tools/brainstorm3 #182: Multiple-spheres forward model, and recommendation for MEG

Moo-Marc, 2019-03-13:
A long time ago, I did a study of various sphere fitting methods for MEG forward modelling.  I remembered that this approximation adds on the order of 10% error to the forward solution, so I switched to using Nolte's method when using Fieldtrip some years ago (it's called "single shell" in FT).  I just looked back at my [paper](https://iopscience.iop.org/article/10.1088/0031-9155/56/17/010/meta) and saw this comment which I had forgotten:
> "Other variations of these methods were tested but produced similar or worse results and are not shown here. For example, the NC method variation implemented in Brainstorm (with α = β = 2) had 27% mean field error and 7 mm mean localization bias as opposed to 12% and 3.5 mm, respectively, for the version with α = β = 1."

So I think it's worth considering changing the sphere fitting method used in Brainstorm.  It might also be a good idea to rethink the recommended method for MEG.  We can discuss at our next video meeting.

ftadel, 2019-03-13:
This is something you can discuss with @sbaillet and @jcmosher

ftadel, 2020-04-27:
@sbaillet @Moo-Marc @juangpc @jcmosher 
Can you please decide if you want to keep this issue open?

Moo-Marc, 2020-04-27:
Hi François.  Yes please.  At the time I had started trying to partly reproduce my paper results within Brainstorm.  I had also identified some strange things in the sphere fitting code.  Given the recent work on forward modeling, I'd like to get back to this soonish.

I also still believe spheres give on the order of 10% error at best and that we might want to rethink our recommendation of using them by default for MEG.

tmedani, 2020-04-27:
Hello, 
It could be nice to compare it with Duneuro results.
Also, it can be a good opportunity to check the difference in the results by considering the outer layers for the MEG using the FEM (accuracy and computation time).   

I can help when it needed and if it worth it :)

ftadel, 2021-03-13:
Any update on this front?

Moo-Marc, 2021-03-13:
Unfortunately not.  Not enough hours in the day!

jcmosher, 2021-03-13:
But, I think the new FEM head model gives us a perspective so that we can
compare multiple ground truths now: sphere, overlapping sphere, Nolte's
shells, OpenMEEG BEM, and DuneNeuro FEM. And we have new leadfield
visualization tools now that make it easier to understand the similarities
and differences between methods.

tmedani, 2021-03-13:
it was on my to-do list, but now it's going down and down in this list.
and I agree with John with all the methods within Brainstorm it's easier to do it now than in some months/years before.

A review of the state of art about all these forward computations will be interesting :)
@jcmosher?

ftadel, 2021-03-14:
I'm trying to clean up the list of issues, and would like to keep only **issues** or **work in progress** in here. 
If someone plans to work on this in the next few months, we can leave this issue open.
Otherwise, I will move it to the infinite list of interesting things we might do some day:
https://neuroimage.usc.edu/brainstorm/Next

Please let me know.

Moo-Marc, 2021-03-16:
I would vote to keep it open for now. If nothing else, there were a few small things I wanted to fix in the current sphere fitting code.  And I agree it would be great to do at least a simple analysis to better quantify levels of errors in the various forward models, and consequently make better recommendations.