brainstorm-tools/brainstorm3 #185: Brainstorm-Simbio implementation

juangpc, 2019-03-14:
Additions, modifications or alterations both in Simbio or/and Brainstorm toolboxes to make these two work together.

juangpc, 2019-03-14:
Hi, 
After a very initial assessment of the situation, I would like to share a couple of conclusions which might not be totally accurate and to which maybe someone else could add corrections and comments.

- Simbio is divided into IPM (Inverse Problem Method) and NeuroFem sub-packages, both written in C++. IPM contains algorithms for BEM methods, apart from several other sub-applications. NeuroFem is based on Cauchy project developments. Cauchy is therefore the kernel which supports  Simbio's FEM capability. Cauchy was developed in Fortran. ([source](https://www.mrt.uni-jena.de/simbio/index.php/Main_Page))

- Simbio is directly available for Linux (i32 and amd64) platform (examples for several versions of ubuntu, red hat, debain or gentoo can be found). But, at least in Simbio's page, nothing is found in relation to windows platforms. ([source](https://www.mrt.uni-jena.de/simbio/index.php/Compiling_SimBio))

- There should be some Fortran to C translators also in windows platform. Another option would be to try to separate back Simbio's C++ code and Fortran code. The reason is that Matlab's api is compatible with fortran therefore separate calls to simbio and cauchy routines should be achievable. 

- There is another option mentioned in a deep brain stimulation toolbox (Lead-DBS). Apparently it is possible to make simbio work in windows but Visual C++ Redistributables and Visual Fortran Redistributables are needed). ([source](http://www.lead-dbs.org/helpsupport/knowledge-base/atlasesresources/fem-based-vta-model/))

**Questions:** 

1. Can IPM and NeuroFem be compiled separately?
2. How hard is Simbio's dependency on Cauchy?
3. Does it make sense to add compatibility between Simbio to Brainstorm only in Linux platform (and perhaps mac os)?
4. In case only NeuroFem depends on Fortran code. Does it make sense to make Simbio and Brainstorm work together for BEM methods but not for FEM?
5. Separating Simbio (NeuroFem) from Cauchy could solve the dependency issue but that might complate things too much. Ideas?
6. At some point a move towards implement DUNEURO-Brainstorm compatibility should be done. Therefore, it might make sense to go for DUNEURO directly, given that some of the previous problems can be reduced that way. 

Thanks,

jcmosher, 2019-03-15:
Hi Carsten,

Can you see this comment by my new Brainstorm post-doc.

https://github.com/brainstorm-tools/brainstorm3/issues/185

My question to you is should we spend much effort trying to get SIMBIO implemented, given the problems of compiling it for Mac and Windows? We want Brainstorm to be multi platform, and I didn't realize that it was only really setup for linux.

I'm guessing that our effort would be better spent going straight to duneuro, since you have modern Galerking error controls methods in there, and that I recall that you were not too excited about us pursuing simbio.

Juan Garcia-Prieto is my new brainstorm post-doc whose primary task is to bring your head modeling into Brainstorm, while learning other aspects of e-phys. So you'll be hearing a lot more from him.

You can either respond directly to me, or if you'd like, edit directly into the Brainstorm GITHUB site, whatever you prefer. If you email me, I'll transcribe into github, where we're trying to keep all of our corporate knowledge contained.

jcmosher, 2019-03-15:
Reply to above from Carsten, March 15, 2019:
Dear John,

I am a little bit confused about Juan "whose primary task is to bring your head modeling into Brainstorm", 
since I thought that Takfarinas should do this integration work in LA!?

Following the latest DUNEuro implementations and results I think we can now claim that with regard to the FEM-aspects, 
DUNEuro is a clesar superset of NeuroFEM for both EEG and MEG and also brain stimulation (it offers CG-FEM, like
NeuroFEM, but many more modern options like DG-FEM and especially unfitted FEM) and a more modern code, 
so I would strongly recommend to directly put all of our NIH-efforts into DUNEuro. On the MEG-side, 
we could now also replicate all NeuroFEM-results, as presented last week by Andre Dachwitz and as shown by 
Maria Carla Piastra (she successfully defended her thesis on Monday). Also see the ARXIV-DUNEuro attachment, 
we are about to submit it to a journal for publication. Backside: Together with you guys, also I have
to enter this new code now and have to leave my good old SimBio...

Johannes implemented NeuroFEM into Fieldtrip and that version would run on multi-platform,
but he focused on EEG. The MEG-part is not included and would be still quite some work to decouple
it from IPM. Don't consider doing this, I would say, too much work that doesn't pay off, since
DUNEuro offers more options and is more future-code.

I am not an operating system expert, and here we mainly work with Linux, but Sophie Schrader (my EEG/MEG calibratzion
PhD student working with DUNEuro) told me that she once got DUNEuro running under Windows with some "hand-work". 
As far as I remember  the discussion with Christian, he claimed that DUNEuro on Windows is very close to stably work. 
I will ask again in the next meeting how much work could that still be to get it stably done and if it would be helpful if we 
had some support from our NIH-colleagues for this aspect.

BR
   Carsten

jcmosher, 2019-03-15:
Thanks Carsten. Juan is the Houston counterpart for me to Takfa in Los Angeles for Richard. So Juan will work with Takfa to implement and test all aspects of forward modeling. 

Multi platform is very important for us, so compiled code carries extra complexity. OpenMEEG, compiled, does work in Brainstorm, so I’m sure we will make a similar path for DUNEuro, once Takfa and Juan work with Francois to understand the best path forward, and work with your experts.

jcmosher, 2019-03-15:
From Carsten, March 15, 2019:
wunderful, John, we are then really well equipped with two PostDocs. I am convionced we will get it running, and DUNEuro is
the way I will go and you can believe me that it was especially difficult for me to accept, because SimBio-NeuroFEM was years-long
"my baby" and brought me on this position :-), but I have to accept also that there is now something better with DUNEuro.
So I will use our project to get into the BrainStorm-DUNEuro world.

Have a nice week-end
   Carsten

tmedani, 2019-08-22:
Dear colleagues;

From our previous discussions and works, I would like to summarize what we know and what we can do for the integration of Duneuro (DN) into Brainstorm (not with SimBio).

The computation of the forward model with DN (and also with other tools) requires : 
**1.  The Head Model**: with Deuneuro there are two main options:
    a - Fitted method: these methods use a mesh, it can be a tetrahedral or hexahedral mesh
    b - Unfitted method: these methods are based on the level set approach
   
For this point and at this time, we will focus on the 'Fitted method' therefor we need a mesh. 
For that, since Duneuro does not offer any options for head model generation, we need to implement our proper functions for the head volume mesh generation and integrate into bst. 
With @ajoshiusc and @juangpc we are working in a different direction to have the best pipeline for the FEM head model. 

As a first step for the integration, we can think about options in which the user can either :  
    a- Use a FEM head model template that we can distribute within bst as a default FEM head model 
         or the user can download (more layers 4 or 5: scalp, skull, CSF, gm,wm). 
    b- Add options that can be used to generate a FEM head model from the surfaces. For that we 
         need to call iso2mesh and add a panel with the mesh options resolution of the mesh, similar 
         with the once used to generate the BEM surface.

For this step, the output from brainstorm should be a mesh file (*.msh) that Dueneuro uses as an input.
The other options related to the head model choice (fitted, unfitted) are fixed on an initialization file with (*.ini)
        
**2. The Source Model:** Duneuro offers four (or five) source model, this is due to the different 
    implementation of the dipole in the FEM. These models are : 
    Partial integration, Subtraction, Venant, Whitney ... 
    For each source model, there is a set of parameters that we have to fixe. We can fixe some of them and give access to advanced users to change some of the others.    

All the parameters related to the source model are fixed on an initialization file with (*.ini). We will set this file from Matlab (brainstorm) according to the user choice.

For this step, the output from brainstorm should be a text file (*.txt) that Duneuro uses as an input. We will use a similar approach as now, the dipoles will be defined on each cortex vertex.
(or in the middle of the grey matter layer if we distinguish between gm/wm )
This file contains the positions of the dipoles (x,y,z) and the orientation (xd, yd, zd) (or vector components) on the three directions (similar file as OpenMeeg)
With this format : 
x0 y0 z0 1 0 0
x0 y0 z0 0 1 0
x0 y0 z0 0 0 1  
x1 y1 z1 1 0 0
x1 y1 z1 0 1 0
x1 y1 z1 0 0 1
...

**3. Electrode Model:** As in the OpenMEEG, the electrodes' positions are defined by their cartesian coordinates (x,y,z) also in a text file. 
For best control of the electrodes' position, I suggest projecting the electrodes to the nearest surface mesh node.

For this step, the output from brainstorm should be a text file (*.txt) that Duneuro uses as an input.
This file contains the positions of the electrodes (x,y,z), or their projection.

**4. Conductivity Model:**  The conductivity value for each layer of the head is defined also in a file (*.con). The conductivity can be either isotropic or anisotropic. 
The number of the line on this file corresponds to the number of layers. In the case of anisotropic tissues, a second column is added to the file as the value on the tangential direction. 
(I have to confirm this with Carten team/or check to code)

** The initialization file:** More other options are specified in this file related to the choice of the solver, the accuracy... but for better documentation, we need more clarification
I put some explanations in this repo: https://github.com/svdecomposer/brainstorm-duneuro/blob/testing/apps/the%20mini%20file%20and%20definition

Duneuro works fine on a Linux machine and an easy Matlab binding is included (via MEX files).

The main problems that we face are : 
The Matlab binding is only available for Linux and not possible for windows. 
The cross-compilation from Linux to generate the windows MEX files was just terrible.

We were able to generate binary file for Windows (EXE) for a specific application (EEG for now). 
so since we are not able to generate the Mex file for multiple platforms, the best option for the integration at this time is to use the binary files. 

We got some errors on the obtained solution and we think this is due to the positions of the electrodes. It seems that DN modifies the positions of the electrodes!!
When this point fixed, we can work on the implementation of the different applications (MEEG, ECoG, sEEG...)

Attached files & links : 
For point 1 I attached the Matlab script and a basic function to write the *.msh file
these codes can be used for volume mesh generation from surface files.

For the other points (2, 3, 4), we put examples in this repo: https://github.com/svdecomposer/brainstorm-duneuro/tree/testing/apps/eeg_test1

I think I said most of the details, @juangpc could you validate?
and of course, we will keep discussing this.

@ftadel 
For our discussion on the bst meeting: here two Matlab files that I use to generate volume mesh from the surface.
1- an old function that I have and use to write msh file, 
not optimized but it works, maybe there better version : 
[fc_ecriture_fichier_msh.txt](https://github.com/brainstorm-tools/brainstorm3/files/3527914/fc_ecriture_fichier_msh.txt)
2- Matlab script pour le maillage, j'ai mis une petite description au debut.  
[generateVolumeMeshFromSurfaceMesh.txt](https://github.com/brainstorm-tools/brainstorm3/files/3527824/generateVolumeMeshFromSurfaceMesh.txt)

Thanks for your contributions & Best regards 
Takfarinas

ftadel, 2019-08-27:
@tmedani 
Thanks for the functions. I will add a new file type in the anatomy for storing FEM head models.

There are many options for representing a volumetric head model, for example:
- N groups of tetrahedrons for N types of tissues + N conductivity values 
- 1 group of tetrahedrons + 1 "texture" that associates one conductivity value to each node or each tetrahedron
(1 group of tetrahedrons = 1 list of [Nv x 3] 3D positions of the nodes + 1 list of [Nf x 4] indices of these nodes, similar to a "surface" file in Brainstorm) 

Could you please send me an example of Matlab structure you would like to be saved in the Brainstorm database? (the file that would be converted to a .msh file before calling duneuro)

Thanks
Francois

tmedani, 2019-08-29:
@ftadel 
Here the link to a folder with some scripts, functions and an example of the data file.
https://drive.google.com/open?id=13TBxxrk6NlDRH9qDrp1K0bFFvXWVaUsk&authuser=medani@usc.edu&usp=drive_fs
The main function is 'bst_fem_lf_process.m' and it uses the other functions&scripts.
It should work without errors, but before you should change the path to the surface mesh, and add iso2mesh to the Matlab path (section 0 of the main script).
These function could be the first approach for integration.

Kind regards 
Takfarinas

ftadel, 2019-08-31:
> By the way: I'd like to start a new page on the bst tutorial dedicated for this part,
however, I don't know how to create a new page. Could you, please, show me how to do?

To create a new page, just type the name of the new page in the URL (eg. https://neuroimage.usc.edu/brainstorm/MyNewPage), and then click on the link to create the page.

ftadel, 2019-08-31:
The structure of the files I thinking about adding to the database is the following (names of the variables kept in coherence with the conventions used in the other files of the database):

`    case 'femmat'                        % FEM head model (tetrahedral or hexahedral)
        template = struct(...
            'Comment',         '', ...
            'Vertices',        [], ...   % [Nvert x 3] double: List of position of the nodes with their three cartesian coordinates
            'Faces',           [], ...   % [Nnode x 4] integer for tetrahedral meshes; or [Nnode x 8] integer for hexahedral meshes
            'Tissue',          [], ...   % [1 x Nnode] integer: tissue classification for each node
            'TissueLabel',     [], ...   % [1 x Ntissue] cell array: label of the tissues in this head model
            'History',         []);`

> cond: [0.3300 0.1400 1.7900 0.0100 0.4300] : Nb_tissu x 1 or Nb_tissu x 2 in the case of anisotropy. the conductivity of each tissue-type (in order of tissuelabel)

To remain coherent with OpenMEEG, the conductivities will be computation options, not anatomical data saved in the database.

> After the FEM computation, new data should be saved :
> stiff: [nb_Node x nb_Node] : the stiffness matrix
> lf: [nb_Node x3] : the lead field matrix for the whole volume
> or
> transfer: [nb_Electrod-1 x nb_Node]
> lf: [nb_Electrod x3] : the lead field matrix only on the electrodes

You need to save this data in the Brainstorm database?  Where and for which purpose?
Aren't the forward models saved in output supposed to be compatible with the others existing? If not, how are we supposed to use them with the inverse functions we currently have?
https://neuroimage.usc.edu/brainstorm/Tutorials/HeadModel#On_the_hard_drive

We have to be clear about the separation between the data representation in Brainstorm (has to be compatible with the current structures and conventions) and the structures  in input and output of duneuro. The two are different, and we need conversion steps for both ways.

tmedani, 2019-09-01:
> To create a new page, just type the name of the new page in the URL (eg. https://neuroimage.usc.edu/brainstorm/MyNewPage), and then click on the link to create the page.

Super thank you

tmedani, 2019-09-01:
> The structure of the files I thinking about adding to the database is the following (names of the variables kept in coherence with the conventions used in the other files of the database):
>
> ```
> case [...]

That should be fine for the geometry definition.  
some suggestions : 
1-   'Faces',           [], ...   % [Nnode x 4] integer for tetrahedral meshes; or [Nnode x 8] integer for hexahedral meshes
I think Faces refers to surfaces,  it's better to use the word 'Elem' for volume
Elem it uses 4 nodes for tetra or 8 for hexa, and Face only 3 nodes.
2- 'Tissue',          [], ...   % [1 x Nnode] integer ==> should be [1 x Nelem], in FEM the element should be belonging in the same tissue but nodes could be shared. 

> > cond: [0.3300 0.1400 1.7900 0.0100 0.4300] : Nb_tissu x 1 or Nb_tissu x 2 in the case of anisotropy. the conductivity of each tissue-type (in order of tissuelabel)
> 
> To remain coherent with OpenMEEG, the conductivities will be computation options, not anatomical data saved in the database.

yes, makes sense, and the users can change the value from the GUI. 

> You need to save this data in the Brainstorm database? Where and for which purpose?

> Aren't the forward models saved in output supposed to be compatible with the others existing? If not, how are we supposed to use them with the inverse functions we currently have?
> https://neuroimage.usc.edu/brainstorm/Tutorials/HeadModel#On_the_hard_drive
> 
I agree the most important is the LF and it's the only output used for the inverse problem.

> We have to be clear about the separation between the data representation in Brainstorm (has to be compatible with the current structures and conventions) and the structures in input and output of duneuro. The two are different, and we need conversion steps for both ways.

I understand,  since the FEM computation is too long, I think that saving the stiffness matrix could be useful or maybe not... we have to discuss that.

tmedani, 2019-09-04:
Dear colleagues, 

Following the meeting discussion about the volume mesh generation.
I wrote some Matlab functions that can be used to generate tetra mesh from a T1 MRI.
This contribution is mainly based on the "roast" and "iso2mesh" toolbox. 

All the codes and the documentation are here : 
https://github.com/svdecomposer/brainstorm-duneuro/tree/testing/matlab_codes/Mesh_generation/BasedOnRoast_Takfa 

There is an example in the main function "bst_mri2tetra.m" that you can easily run and check the output mesh.

Advantages : 
- Fully automated
- It works fine for clean MRI data, especially for the distributed model: MNI152_T1_1mm.nii 
that you can find in the roast toolbox : \roast\roastV2.7.1\example\ MNI152_T1_1mm.nii 
- It takes less than 20 minutes to get the mesh (tetra)
- The main extracted tissue are : {'WHITE','GRAY','CSF','BONE','SKIN'}

However, I noticed some drawback, I tested this pipeline for another example, the results are not as good as the MNI152. There are some holes in the surfaces and some isolated isles in the mesh. 
I assume that my example has some noise during the acquisition. 
For this kind of problem, some manual correction should be done to clean the mesh.

@sbaillet, @jcmosher, @richardmleahy, @ajoshiusc, @juangpc   

Thanks and Kind Regards 
Takfarinas

ftadel, 2019-09-05:
I added a new file type "fem" to the database:
 https://github.com/brainstorm-tools/brainstorm3/commit/f305ffae750179a0a1c7a8eccd0c099089b18074

[image]

I followed your recommendations for the file contents.
If you have such a structure available in your Matlab environment, you can save it in the Brainstorm database by right-clicking on the subject folder > File > Import from Matlab > Select the variable name.

```
    case 'femmat'                        % FEM head model (tetrahedral or hexahedral mesh)
        template = struct(...
            'Comment',         '', ...
            'Vertices',        [], ...   % [Nvert x 3] double: List of position of the nodes with their three cartesian coordinates
            'Elements',        [], ...   % [Nnode x 4] integers for tetrahedral meshes; or [Nnode x 8] integers for hexahedral meshes (1-based indices in the Vertices matrix)
            'Tissue',          [], ...   % [1 x Nelem] integer: tissue classification for each node
            'TissueLabels',    [], ...   % [1 x Ntissue] cell array: label of the tissues in this head model
            'History',         []);
```

So far, there are no actions attached to this file type. The only menus you get are the "File" submenu (copy, view contents, view history...)
Do you think you will have anything to add in there? 
Are you expecting to be able to display the FEM head model in Brainstorm? How?

Now I need to add the connectors to call the functions to generate this FEM head model.
Can you describe exactly what should be implemented? With Roast and/or iso2mesh?
(eg. the user right-clicks on the MRI > Generate FEM head model, description of the options window, function to call, etc....)

> I understand, since the FEM computation is too long, I think that saving the stiffness matrix could be useful or maybe not... we have to discuss that.

What would that change in terms of workflow?

ftadel, 2019-09-05:
PS: I'll create a new process "process_generate_fem" (with "process_generate_bem" and "process_segment_cat12" as models) which downloads automatically iso2mesh (https://sourceforge.net/projects/iso2mesh/files/iso2mesh/1.9.0-1%20%28Iso2Mesh%202018%29/) and ROAST (https://www.parralab.org/roast/roast-3.0.zip) to $HOME/.brainstorm/, similarly to OpenMEEG

tmedani, 2019-09-05:
> PS: I'll create a new process "process_generate_fem" (with "process_generate_bem" and "process_segment_cat12" as models) which downloads automatically iso2mesh [...]

This is super, let say that is the first approach that we can integrate into bst for volume mesh generation. 
With @ajoshiusc and @juangpc, we are working on another approach based on brainsuite pipeline.

> PS: I'll create a new process "process_generate_fem" , 
roast folder includes iso2mesh, therefore download "roast" should be sufficient.

tmedani, 2019-09-05:
> I added a new file type "fem" to the database:
> [f305ffa](https://github.com/brainstorm-tools/brainstorm3/commit/f305ffae750179a0a1c7a8eccd0c099089b18074)

Super, thanks

> [image]
>
> I followed your recommendations for the file contents.
> If you have such a structure available in your Matlab environment, you can save it in the Brainstorm database by right-clicking on the [...]

It's almost what the FEM need from the anatomy (geometry) side.
  'Elements',        [], ...   % [Nnode x 4]  ==> shoud be [Nelements x 4] or [Nelements x 8]

> So far, there are no actions attached to this file type. The only menus you get are the "File" submenu (copy, view contents, view history...)
> Do you think you will have anything to add in there?

We can also discuss these during the meeting?
we can maybe add an option to re-mesh (less or high)
We can also display it, this is important from the user side.
==>  we can also add function to display each tissue 
==>  also some funny visualization options similar to the Resect [X,Y,Z] already available in bst

> Are you expecting to be able to display the FEM head model in Brainstorm? 
Yes, as described above

It needs some work, I can work on that.
> How?

There is some build in MatLab function and also iso2mesh 

> Now I need to add the connectors to call the functions to generate this FEM head model.
> Can you describe exactly what should be implemented? With Roast and/or iso2mesh?
> (eg. the user right-clicks on the MRI > Generate FEM head model, description of the options window, function to call, etc....)

If we have the surfaces of the tissue we can generate the volume by calling iso2mesh, 
we can use the functions that I shared.

From the MRI, we can use the roast pipeline that I described in the previous post, but for this, the user should be aware of the quality of the mesh.  

The other option is to distribute a FEM template that the user can use with their electrode position and conductivity values.  

> > I understand, since the FEM computation is too long, I think that saving the stiffness matrix could be useful or maybe not... we have to discuss that.
 
> What would that change in terms of workflow?

good question, I think from the brainstorm user side, the forward computation will be done only one time (fixed value of conductivity and same electrode locations), therefore saving the stifness matrix is not useful in this case.

ftadel, 2019-09-06:
> we can maybe add an option to re-mesh (less or high)

Please send me the exact function calls you'd like to use for this.

> We can also display it, this is important from the user side.
> ==> we can also add function to display each tissue
> ==> also some funny visualization options similar to the Resect [X,Y,Z] already available in bst
> It needs some work, I can work on that.

Send me examples of screen captures and corresponding function calls, and I'll work it out in the Brainstorm GUI.

> If we have the surfaces of the tissue we can generate the volume by calling iso2mesh,
we can use the functions that I shared.
> From the MRI, we can use the roast pipeline that I described in the previous post, but for this, the user should be aware of the quality of the mesh.

So two options. One that is available when there is at least an MRI, and one that is available when there are multiple layers in the anatomy folder. 
In the second case, which surfaces are required and which ones are optional?

Please describe precisely what would be the options available for these two solutions (and if possible a precise workflow, eg. "user clicks on this file, gets these options offered, then when this is selected, this function is called).
And post example of calls to your functions that do what you want (input = only these options and Brainstorm structures for surfaces/MRI/FEM head model). 

> The other option is to distribute a FEM template that the user can use with their electrode position and conductivity values.

Send me the "FEM head model" file you'd like to distribute, and I will add it to the ICBM152 template (either the small version distributed in the brainstorm package or the large one that can be downloaded separately, depending on the size of the file).

ftadel, 2019-09-07:
It is now possible to display FEM meshes with Brainstorm, using the resection sliders:
https://github.com/brainstorm-tools/brainstorm3/commit/b7076807620fe6ecd2a1da283940dba6fb67c8eb

[image]

[image]

tmedani, 2019-09-07:
> It is now possible to display FEM meshes with Brainstorm, using the resection sliders:
> [b707680](https://github.com/brainstorm-tools/brainstorm3/commit/b7076807620fe6ecd2a1da283940dba6fb67c8eb)
> 
> [image]
> 
> [image]

Hi Francois, 
This is super.  I was just going to send the function for visualization adapted for the femtemplate,
with a panel of option for the visualization. But it seems to be too late... 
Anyway, here is what I did if it can add sone improvement for the actual version

From the bst femtemplate "template" as an input,
we can display the mesh and the user can choose to change many options specified in the plotMeshOption structure : 
% CutingPlanEquation: '1*x+0*y+0*z+0>0' : Eqution of the cuttin plan 
% displayedge: 1
%   edgecolor: 'k'
%   linestyle: '--'
% displaynode: 1
%   nodestyle: '.'
%   nodecolor: 'k'
%   markersize: 10
% displayfacecolor: 1
%   facecolor: 'g'
% transparency: 1
%   facealpha: 0.6000
The main function is "bst_plotFemMesh.m"
The setteing of the option is done from the script displayMesh2.m

Thanks 

[bst_plotFemMesh.txt](https://github.com/brainstorm-tools/brainstorm3/files/3586859/bst_plotFemMesh.txt)
[DefineTissuColors..txt](https://github.com/brainstorm-tools/brainstorm3/files/3586860/DefineTissuColors.txt)
[displayMesh..txt](https://github.com/brainstorm-tools/brainstorm3/files/3586861/displayMesh
[displayMesh2.txt](https://github.com/brainstorm-tools/brainstorm3/files/3586866/displayMesh2.txt)
.txt)

sbaillet, 2019-09-07:
Great!

[image]<https://user-images.githubusercontent.com/6920058/64477862-aa153580-d1a0-11e9-93b0-d0b9742040cb.png>

[image]<https://user-images.githubusercontent.com/6920058/64477866-ad102600-d1a0-11e9-816c-0b2a113db874.png>

ftadel, 2019-09-10:
I added the default colors you use in your code:
https://github.com/brainstorm-tools/brainstorm3/commit/e135918871aa844ee187db92c7bee36990d8f69e#diff-39e93b6df431e133c7a52dacc9743c0e

Except for that, I'm not sure how this improves a lot lot what is already posted. I couldn't find any configuration panel in there. Let me know what is missing, or what you'd like to be different.
If you want to play with the rendering of the surfaces, it's here:
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/gui/view_surface_fem.m#L183

Please send me code examples for calling the generation of the meshes, so I can work on the integration with Brainstorm. Thanks

ftadel, 2019-09-11:
First option done: computation from BEM surfaces.
https://github.com/brainstorm-tools/brainstorm3/commit/3be35785b34ebd2991a8b95c39873fd53897f5a0#diff-7c3a114ee98415cadd8af55c3c3b262e

I moved all your code for the mesh generation to there. Please check that it does what you want, then move to this implementation instead of keeping on developing your own branch.
I had to make a separate automatic downloader for iso2mesh when using the BEM surfaces, because it requires a function of iso2mesh which is not bundled with ROAST (tetgen).

[image]

[image]

tmedani, 2019-09-12:
Super Francois. 
I will check the process and then move here. 

Some important point here : 
at : toolbox/process/functions/process_generate_fem.m
% TODO: THIS PART SHOULD BE AUTOMATED at line 205 
I tested many scenarios and checked the results, 
the output of the surf2mesh is sometimes unstable, the tissu labels are randomly distributed.   
We have to be careful of this, the main reason is when the seed points are not correctly defined,
and also when the mesh parameter : keepratio is <0.8.
Two possible solutions : 
- 1 : I'm working on it, and I'm waiting for a possible solution from Qianqian Fang.
- 2 : We can add an option that the user can check and set the tissue (set as WM, GM .... ), 
similar to what bst do for the surfaces. **<==** may be only if the first solution is not working

Another important point, for now, this scripts works for three surfaces,
but we need to adapt it for N surfaces and add errors messages if the surfaces are intersecting (or too close).  

Next 
**Head model:**
- Work on the ICBM152 FEM mesh template, I need to check the quality and try to run some computation on it. After that, we can add it to the bst templates. 
- Check the new version of roast, it seems to be better ... 
- Check/test the CAT process and extract the other layers, could be interesting to see the output. 
**Interface bst/dueneuro :**
- write & check all the conversion functions and process that can call duneuro from bst.
- implement the first test of leadfield computation from bst using DN :)
 
A+

tmedani, 2019-09-16:
Hi, 
I worked on the head model that we can use as a template with 5 layers. 
The one I selected It's also based on the ICBM152 head model available here: https://simnibs.github.io/simnibs/build/html/dataset.html

This model is cleaner than NY head model that we can get with the roast toolbox. 
[image]

The model has 5 layers: wm, gm, scf, skull, and scalp. 
I did some scaling and rotation to the original model in order to fit as possible the to bst ICBM  template. 

The adapted files for brainstorm (the surfaces and the volume) model are here: https://drive.google.com/open?id=1gKuMe5vB1KUmYxgCLXqQ2pRtasItWzUh&authuser=medani@usc.edu&usp=drive_fs

I wrote also a function that can generate a source space from the GM surface with depth as a parameter, it based on deflating the cortex surface.  
The dipole position is challenging since the thickness of the GM is not fixe and the interface with the WM is too close in some area. 
Even with 1mm depth, the source space merge with the WM in some areas. 
The second problem that we have to discuss is keeping or not the  'spinal cord and the cerebellum in the model (the source space is also related to these surfaces).

I will run some FEM, asap, on this model and check the results.

tmedani, 2019-09-18:
Hi, 
Following the discussion of this morning about the mesh VS MRI,

Here just a view for the MRI around the cutting plan of the mesh. 
The segmentation and the mesh look really good, and the big gyrus is normal in this subject.

I will do some advanced comparison on that asap. 
[image]

 Best

ftadel, 2019-09-18:
It looks really good indeed.
So I guess the "CSF" layer is actually CSF+meninges.

tmedani, 2019-09-18:
> It looks really good indeed.
> So I guess the "CSF" layer is actually CSF+meninges.

Yes, until now I did not found model including these layers, there are other layers but they are merged either into the skull and/or CSF. 

I think also that is difficult to distinguish on the MRI these layers, they are too thin compared to the voxel resolution.    

The skull also has at least two different tissue, in Carsten's group they were able to distinguish between two tissues for the skull (skull compacta and skull spongiosa)[ ref](https://www.ncbi.nlm.nih.gov/pubmed/24971512) .

Here a real picture of these layers, I found it on facebook :) 
[image]
[image]

sbaillet, 2019-09-18:
Great!

Would be interesting to probe Carsten’s mind on dura layers. Can we assume they are like CSF in terms of conductivity?

Sylvain.

[tissus]<https://user-images.githubusercontent.com/37831900/65181973-dee97e00-da25-11e9-85dd-75dbd99b342c.png>
[tissus_labels]<https://user-images.githubusercontent.com/37831900/65181976-e14bd800-da25-11e9-8f19-7185497460d5.png>

tmedani, 2019-09-23:
Hi,
This is the answer of Carsten for Sylvain’s question: 

in
https://link.springer.com/article/10.1007/s10548-017-0568-9
we had also modeled the dura, following the Ramon-papers (1,2, see below).

The segmentation is the difficult step:
"The segmentation of the dura mater, including the sagittal sinus space filled with venous blood that is surrounded by dura mater, was 
performed using Seg3D and involved manual segmentation as well as some basic image processing steps such as smoothing and 
thresholding."

I am not aware that currently a software exists that does it automatically :-(.

Further literature, where this has been deeper investigated:

(1)
Ramon C, Garguilo P, Fridgeirsson EA, Haueisen J (2014) Changes
in scalp potentials and spatial smoothing effects of inclusion of
dura layer in human head models for EEG simulations. Front
Neuroeng 7:32. doi:10.3389/fneng.2014.00032

(2)
Ramon C (2012) Effect of dura layer on scalp EEG simulations. Int J
Bioelectromagn 14:27–28.

Best regards
     Carsten

ftadel, 2019-09-23:
And no indication about how important it would be to model it (regardless of how feasible it is technically)?

> Can we assume they are like CSF in terms of conductivity?

No. They report a conductivity for the dura mater (0.1) that is closer to the skull spongiosa (0.01) than to the CSF (1.79).

tmedani, 2019-09-23:
> And no indication about how important it would be to model it (regardless of how feasible it is technically)?
> 
> > Can we assume they are like CSF in terms of conductivity?
> 
> No. They report a conductivity for the dura mater (0.1) that is closer to the skull spongiosa (0.01) than to the CSF (1.79).

From the Ramon C (2012): It says that the dura has a significant effect on the EEG amplitude (simulated EEG) 
[image]
In the other papers, the authors recommend to include the dura for the forward computation.  

But it seems that the segmentation of the dura layer is difficult (thickness o.8 to 0.9mm less than 1mm).  And no automatic segmentation can do this for now. 

Also, I think since the effect of the dura is on the amplitude and not on the topography, the effect is less important.

tmedani, 2019-10-05:
Hi all, 
I would like to add some comments related to this discussion, in particular for the head model generation.
As discussed in the previous meeting, the process of mesh generation is an important part of the integration within bst to use the Dueneuro computation.
At this time we found and tested two main tools, Roast and Simnibs.

**The Roast** process is good, but I think that is not the best for brainstorm for now,
I tried it in different MRI and I noticed many problems (mesh not regular, many holes, ...) 
Roast is based on Matlab, it uses SPM segmentation, + homemade functions for checking the segmentation and then call ISO2MESH for the mesh (based on tetgen). This process is multiplatform and the process duration is about 20 to 30 min.

**SimNibs toolbox** has two main processes: 'headreco' and 'mri2mesh'.
The results of these functions are much better than expected and better than Roast. 
The headreco is multiplatform, it based on SPM and CAT, then it uses GMESH for mesh generation (based on netgen).  The duration of this process is about 2 to 3 hours  
The mri2mesh uses FSL and Freesurfer for the segmentation and then call GMESH for the mesh, it lasts around 20 hours.
These two process use also another toolbox called MESHFIX that check and correct the mesh. 

SimNibs is a mixture of Matlab and Python, calling other binary tools.

@ftadel, for now, I propose to postpone the integration of the Roast, and we will focus on SimNibs. 
With @ajoshiusc and @juangpc we will work to follow the process of headreco and use only Matlab. 

In the meantime, I build some functions that can generate a volume mesh from the BST ICBM Model. 
This process will add a new layer between the outer and the scalp. 
We can set/change the thickness of the CSF, skull and scalp and also the quality of the mesh.

This could be a basic template model for the FEM computation (and we can use it to compare the results with the OpenMeeg).

(on the legend : GM and not WM)
[image]

[image]
with the source space : 
[image]

ftadel, 2019-10-07:
@tmedani Thanks for the report.
Do you need me to do something at this stage, or should I wait until you find some stable solutions?

tmedani, 2019-10-08:
@ftadel  thanks :)

I will send you (or push) the new functions that can generate the FEM head model from the surface files asap. 

I corrected the automatic labeling of the volume tissues (the warning ⚠ as a todo in the previous file)
And I wrote a function that can generate a new layer based either on the template ICBM model or any other bst anatomy (scalp, inner and outer surface). 

These surfaces will be the input of this function.
The inner surface will be kept as it is in order to enclose the source space. 
The CSF layer will be generated by inflating the inner by X mm (thickness X mm)
The Skull will be generated by inflating the inner by  Y mm (thickness (Y-X) mm)
The scalp will be the same and will have the rest of the thickness.  
(The space between the inner and scalp will be divided into 3 layers)

All these four surfaces will be merged together and will be meshed. 

The user can choose the thickness of the CSF and the skull (and/or scalp) and the quality of the mesh. 

Questions: any suggestion for the default values for the thickness (scalp, skull, scf)?
@ftadel  How do you fixe this in the BEM model?

With this process, the user can keep the same head model and can use either the BEM and/or the FEM for the same data :)

Do you agree? 

This should be the intermediate step and we will work on the SimNibs process for better models.

Best regards 
Takfarinas

ftadel, 2019-10-09:
> Questions: any suggestion for the default values for the thickness (scalp, skull, scf)?
> @ftadel How do you fixe this in the BEM model?

So far, we've been using the defaults recommended by the OpenMEEG team: 
https://neuroimage.usc.edu/brainstorm/Tutorials/TutBem#BEM_surfaces
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/anatomy/tess_bem.m

Users can modify the thickness from the interface, but I don't think anybody does it. If we don't know how to set this value, who would?
I don't think the answer to this question is a scalar anyway, because it is not constant: some areas have a much thicker skull, others more CSF... 

Maybe @agramfort or @mclerc have interesting suggestions?

mclerc, 2019-10-10:
Hi François, 
Skull thickness and conductivity are intimately linked : 
you can think of the conductivity as an "effective" conductivity meaning that 
a thinner skull with lower conductivity behaves similarly to a thicker skull with higher conductivity. 
The skull is in fact not homogeneous since the skull contains compacta and spongiosa, with higher conductivity. 
So BEM models are only semi-realistic, and allowing users to change conductivity values and tissue thicknesses 
can be worthwhile, if they have some ground-truth information they could use to guide their choices. 

And of course, adjusting the thicknesses to match a specific age group is very important, 
see eg https://www.pedeheadmod.net 

Maureen 

-----

| [ https://neuroimage.usc.edu/brainstorm/Tutorials/TutBem#BEM_surfaces |
| https://neuroimage.usc.edu/brainstorm/Tutorials/TutBem#BEM_surfaces ]
| [
| https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/anatomy/tess_bem.m
| |
| https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/anatomy/tess_bem.m
| ]

| https://github.com/brainstorm-tools/brainstorm3/issues/185?email_source=notifications&email_token=AAKTX3H6AEHPHZGGMGECOHTQNWESHA5CNFSM4G6USKNKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEAW7OWI#issuecomment-539883353

ftadel, 2019-10-11:
Hi Maureen,

Thanks for the link, we'll look into this.
(@tmedani: these could be interesting test cases for your current meshing experiments - it would be great to distribute not only one MNI FEM template, but also one per age range)

We do give this option to modify the skull thickness and conductivities when calling OpenMEEG from Brainstorm. But as we don't give any direction on how to adjust these values, I guess in practice 99% of our users are going to keep the default parameters without trying to understand what they mean. Making a method correct and having it correctly used are different objectives :)
Would you have any reference for adjusting these parameters to age groups or special populations we could cite in the BEM tutorial page (https://neuroimage.usc.edu/brainstorm/Tutorials/TutBem) ? 

Thanks
François

tmedani, 2019-11-13:
Hi, 
Thanks, @mclerc,  for the explanation and the link. 

>>(@tmedani: these could be interesting test cases for your current meshing experiments - it would be great to distribute not only one MNI FEM template, but also one per age range)

@ftadel, I will also generate FEM models for these examples. Or maybe better to implement the functions that users can use to generate the fem mesh (similar to the previous posts).

Also, I think I will divide this thread into small issues, there are many different topics here, and it difficult to follow. 
Here the main topics : 
1- FEM and Duneuro computation & bst integration (we keep this one)
2- Volume Head model generation (from MRI, from Surface via SimNibs, iso2mesh, Roast)
3- Conductivity mapping from DTI (BDP/SimNibs pipeline or other tools )
4- Volume mesh visualization with bst 
5- Define BIDS format for the FEM mesh
6- ...

@ftadel  do you agree?

ftadel, 2019-11-14:
Sure, feel free to reorganize the discussion in multiple issues, at lease for points 1 and 2.
The only annoying point is that with github issues, I don't think we can move a comment to a different issue, unlike on the Discourse forum. If you create new issues, make sure you copy the relevant information available in this thread to the new issues.

Let's leave the BIDS discussion on the side for now, and consider these are not data we would share, but rather recompute from the anatomical scans.  

More anatomy examples: 
https://iopscience.iop.org/article/10.1088/2057-1976/ab4c76

juangpc, 2019-12-11:
Hi, 

while working with simnibs, I have a questions/doubt:

In the first step, the T1 and the T2 are coregistered. This means that the actual mri is slightly altered in the first step of the simnibs pipeline. After this step, the volume segmentation operates on the 'slightly-altered' mri, which means that the reconstructed volumes slightly miss-match the original T1. 
    
   Solution 1. use the flag "-d no-conform" when calling headreco. This decreases the overall quality . Don't know how much. @tmedani ?
   Solution 2. make the process change the original T1 already added to bst database and use the "conformed" one. This solution will need to translate the already localized ac/pc/ih nas/lpa/rpa points into the new conformed mri. I think we can do that. However this, in turn, alters the vox2scs conversion. I'm now trying to find a solution to this percise problem.
   Solution 3. Make the user understand/acknowledge this in some way and ask him to perform this step before adding the t1/t2 to the whole process. 

What do you guys think?

ftadel, 2019-12-12:
> T1 and the T2 are coregistered

Do you have more details on this step? 
Does it register the T1 on the T2? or vice versa? does it actually reslice the volume and save the registered one in a new file?

Is this simply a rigid affine registration?  (I'm not sure why you'd want to use anything else to register two MRI sequences of the same head)
If so, we could probably find a way to get the transformation (4x4 rotation+translation) and apply the inverse of it to the vertices of the meshes before saving them to the database, so that everything is saved in the original coordinates system defined in Brainstorm.

juangpc, 2019-12-12:
Hi, 

> > T1 and the T2 are coregistered
> 
> Do you have more details on this step?
> Does it register the T1 on the T2? or vice versa? does it actually reslice the volume and save the registered one in a new file?

Yes. It does reslice. However, to avoid this, in the simnibs call you have to add a flag: 
```
headreco all -d no-conform ernie4 ernie_T1.nii.gz ernie_T2.nii.gz
```
This ```-d no-conform``` makes spm12 do not reorient and resample the original T1. I'm testing it right now. (ernieT1 and T2 are example mris from withing simnibs, freesurfer's bernie counterpart).

I don't know how much avoiding this step affects the total accuracy. But the problem is that if a user performs this task after adding the subject to brainstorm, then there will be a mismatch between both MRIs (see below).

Retarding the registration. After installing simnibs if you run  ```headreco all --help``` you'll get the following message: 
 ... **SPM12 is called to coregister the T2 to the T1** ...

There is also a non-linear warping involved in the registering, not only affine. 

Once you run simnibs if you read the triangles and tetrahedra information from the output file, in order to transform the coordinate system you can run: 
```matlab

%read scalp vertices info (region=5)
ernie = mesh_load_gmsh4('ernie.msh');
regions=unique(bernie.triangle_regions);
tris=bernie.triangles(bernie.triangle_regions==regions(5),:);
simnibsV=bernie.nodes(unique(tris(:)),:);
simnibsV=simnibsV(1:3:end,:); %only 1/3 is enough for this demo

T1 = diag([sMri.Voxsize(:) ./ 1000; 1]);
T2 = [sMri.SCS.R, sMri.SCS.T./1000; 0 0 0 1];
T = T2 * T1 * inv(sMri.InitTransf{1,2});

simnibsVSCS = [simnibsV ones(size(simnibsV,1),1)];
simnibsVSCS = (T * simnibsVSCS')';
simnibsVSCS(:,4)=[];
```
And see this...
[image]

and this...
[image]

The _slight_ difference is visible.

More info:

If you download simnibs, you'll find it installs (in windows) in ```C:\Users\username\AppData\Local\SimNIBS.```
In that folder you'll find headreco python script in 
```miniconda3\envs\simnibs_env\Lib\site-packages\simnibs\msh\headreco.py```
If you go through the python script you'll see how the actual matlab commands that is executed by simnibs is:

```
matlab -nosplash -nodesktop -wait 
-logfile "C:\projects\simnibs\hcp\m2m_hcp_example\segment\spmcat.log" 
-r "addpath('userPath\appdata\local\simnibs\miniconda3\envs\simnibs_env\lib\site-packages\simnibs\msh',
            'userPath\appdata\local\simnibs\miniconda3\envs\simnibs_env\lib\site-packages\simnibs\resources\spm12');
try,
  segment_SPM('userPath\appdata\local\simnibs\miniconda3\envs\simnibs_env\lib\site-packages\simnibs\resources\templates',
              'C:\projects\simnibs\hcp\m2m_hcp_example\segment\spm\T1fs_conform.nii','C:\projects\simnibs\hcp\m2m_hcp_example\segment\spm\T2.nii',0.01,3,0); 
  segment_CAT('C:\projects\simnibs\hcp\m2m_hcp_example\segment\cat\T1fs_conform.nii','c:\users\jgarciaprieto\appdata\local\simnibs\miniconda3\envs\simnibs_env\lib\site-packages\simnibs\resources\templates',0);
catch ME,
  rethrow(ME);
end,
exit;
```
both segment_SPM and segment_CAT can be found in ```miniconda3\envs\simnibs_env\Lib\site-packages\simnibs\msh```
and mesh_save_gmsh4.m can be found in ```miniconda3\envs\simnibs_env\Lib\site-packages\simnibs\matlab```

The transformation T applied is basically to _unapply_ the original mri vox2world transformation. Then apply to simnibs info the vox2scs transformation. The transformation vox2world is decided by the user when importing the mri. So if the user decides not to apply it we'll need not _unaply_ it. correct @ftadel?

I'll try to show the results when using -no-conform option, at the moment they are still running. The headreco script takes 2.5h in my (quite a powerful) computer. So soon.

tmedani, 2019-12-12:
Hi Juan, 
This is super and thanks you for this report. 

> ```matlab
> %read scalp vertices info (region=5)
> ernie = mesh_load_gmsh4('ernie.msh');
> regions=unique(bernie.triangle_regions);
> tris=bernie.triangles(bernie.triangle_regions==regions(5),:);
> simnibsV=bernie.nodes(unique(tris(:)),:);
> simnibsV=simnibsV(1:3:end,:); %only 1/3 is enough for this demo
>
 
> T1 = diag([sMri.Voxsize(:) ./ 1000; 1]);
> T2 = [sMri.SCS.R, sMri.SCS.T./1000; 0 0 0 1];
> T = T2 * T1 * inv(sMri.InitTransf{1,2});
> simnibsVSCS = [simnibsV ones(size(simnibsV,1),1)];
> simnibsVSCS = (T * simnibsVSCS')';
> simnibsVSCS(:,4)=[];
> ```

Finally you got it closer ... bravo
> T = T2 * T1 * inv(sMri.InitTransf{1,2}); 

@juangpc : what are the values stored sMri.InitTransf ?

There is a slight difference, I think it's mainly due to the segmentation and/or reslicing difference between bst and simnibs ?

Also, could you, please, check how is the difference between the brain surfaces between bst model and the output of simnibs? 
I think this is also important. 

Takfarinas

juangpc, 2019-12-12:
When running previous script with simnibs (no-conform) option, you get this, which is much better, in my opinion.

[image]

ftadel, 2019-12-13:
Using the head surfaces for validating the results is good for getting an overview of the registratin, but is not necessarily very accurate. We're probably doing very different morphological operations to reconstruct these surfaces. Alignment of volumes could be checked by overlaying the volumes together (eg. with Brainstorm or MRIcron). 

My guess is that if their pipeline uses a non-linear registration T1+T2, we should try to keep using it.
There are maybe other options to investigate, like reusing the transformation fields generated by SPM: can you find any y_...nii files in the output folder?
If we can access the non-linear registration information in Brainstorm after running headreco, we could invert it and project back the meshes in the original brainstorm T1 space.
Does it make sense?

juangpc, 2019-12-13:
> can you find any y_...nii files in the output folder?

Yes, there is actually a .mat file with a Twarp inside, among other stuff.

>If we can access the non-linear registration information in Brainstorm after running headreco, we could invert it and project back the meshes in the original brainstorm T1 space.
Does it make sense?

It does to me. I'll try to have it done.

juangpc, 2019-12-16:
Hi @ftadel, I've located [this file](https://we.tl/t-lkdtiVymDH). I think it is the y_...nii file you were referring to, but simnibs team change the name to BiasField_... .

Does anybody have experience with this kind of conversion? (or inverse conversion, better said, of the deformation field?)

ftadel, 2019-12-17:
No, this is not the file I am talking about. 
What you shared here is really a bias correction.

The volume describing non-linear registrations done with SPM is a .nii file with three volumes in it, one for each coordinates X Y Z, indicating how each (i,j,k) coordinates of one volume corresponds to the registered coordinates of the other volume.

Example of what the three volumes look like when you open them in MRIcron:

[image]

[image]

[image]

juangpc, 2019-12-18:
Hi, various things: 

- regarding simnibs calling a python script that calls matlab (actually spm12) etc... we can either: 
   - Leave simnibs completely as it is, favoring an easy integration of simnibs' future developments. This will require the users to execute a .exe installation file which takes care of the python stuff.
   - Or, since the itinerary would be Matlab(brainstorm) making a system call which calls a python script which calls matlab to run an spm script. We can also directly call those matlab scripts which are called by simnibs. They are pretty well itomized in simnibs.

To add more info. Simnibs repo is a little over a GB today. 

Even more info. The location of python scripts and the actuall matlab call has been posted previously. Simnibs dflt folder is ```C:\Users\pepito\AppData\Local\SimNIBS``` Matlabs scripts reside inside there in ```miniconda3\envs\simnibs_env\Lib\site-packages\simnibs\msh``` They are called ```segment_SPM.m``` and ```segment_CAT.m```.

In line # 72 of ```segment_CAT.m``` we need to change a configuration parameter.

```
matlabbatch{1}.spm.tools.cat.estwrite.output.warps = [1 0]; % [forward inverse]
``` 
for 
```
matlabbatch{1}.spm.tools.cat.estwrite.output.warps = [1 1]; % [forward inverse]
```
I'm not sure how to do that through other simnibs configuration methods. 
The file can also be found here: 
https://github.com/simnibs/simnibs/blob/6a52c56ba93d8f11e6c7764721bcd5aa9b12709e/simnibs/msh/segment_CAT.m#L72

This way we indicate spm to compute the inverse deformation field (in addition to the actual deformation field) which we can then apply to the generated surfaces/volumes (actually to the nodes) so that they end up better aligned with the initial mri and other brainstorm images/surfaces.
For "ernie" the example subject in simnibs, here are the [deformation field file](https://www.dropbox.com/s/hemhp366azjumvd/y_T1fs_conform.nii?dl=0), and the [inverse deformation file (180mb!)](https://www.dropbox.com/s/91b5p2hc8ayac2t/iy_T1fs_conform.nii?dl=0). The original t1 is [here](https://www.dropbox.com/s/upzldnwscyhfexz/ernie_T1.nii?dl=0), and the "conformed" or modified one is [here](https://www.dropbox.com/s/yhh5jc3vhlxhbkq/ernie2b_T1fs_conform.nii.gz?dl=0). These are all files that can be found in the (quite good) simnibs documentation and example files.

I'd like to share my understanding of this. My intention is to:
 1. multiply the matrix of nodes (of each triangle or tetrahedra) which should be in mm, by the inv(vox2ras) of the conformed nii image. This will give me the actual voxel each node belongs to by rounding the decimals. 
2. Look for the vector to deform through the inverse deformation field. 
3. Apply to each node the deforming vector. The resulting surfaces/meshes (I guess) should be in the original space. 

Is this correct?

tmedani, 2019-12-19:
Hi Juan, 

Thank you for these detailed reports. 

I'm not sure, that the integration of simnibs as it is, is possible!. 
Another option is to isolate the headreco (and or mri2msh) from the whole package.  

Or we can ask the user to generate the head model as an external step from brainstorm, by orienting them to simnibs, after bst will load the mesh (not recommended) 

@ajoshiusc is working on a "brainsuite" version of "headreco", we can also have his feedbacks? 

For the rest of your comments and questions, I'm lost, I think you have already solved the problem?
or I missed something? 
==> let discuss tomorrow :0
Cheers 
Tak

ftadel, 2019-12-19:
@juangpc  
To convert the conformed coordinates Pc(Xc,Yc,Zx) of a node of the output mesh back into the original space, isn't it the forward deformation field you need?

I have this piece of code that uses the inverse transformation to MNI space to convert subject SEEG coordinates into MNI coordinates.
https://github.com/ftadel/IntrAnat/blob/master/locateElectrodes.py#L133
Reading this piece of code again, I wonder why there is this loop instead of a simple call to interp3 or griddata that would do a table lookup on all the points at once. (the loop would take forever on tens of thousands of points of the meshes).

So to convert normalized coordinates back to native subject space, the same logic could be applied but using the forward transformation?

I'm a bit lost, I'm not so familiar with these SPM deformations matrices... I would need to think about this for longer, and do it on a real example. Let me know if you need additional help with this, and we'll work from where you are, maybe based on a prototype that is fully working but without the conversion back to native space.

As a side note, it would be nice to add a support for these deformation matrices in Brainstorm, in order to allow non-linear MNI normalization, instead of the cheap affine transformation we use now. That's on my todo list...

@tmedani 
> I'm not sure, that the integration of simnibs as it is, is possible!.

What are the blocking points?

> Or we can ask the user to generate the head model as an external step from brainstorm, by orienting them to simnibs, after bst will load the mesh

This would be a nightmare, especially with these transformations of coordinates... not an option.

tmedani, 2019-12-20:
After discussed with @juangpc., it's more clear for me now.

It will be nice to see how much is the effect on the mesh with and without this transformation on the input MRI.
Also, I'm not sure that the inverse transformation on the mesh is a good idea. I think since it's not a linear transformation, there is a high risk to end up with self-intersecting elements/edges on the transformed mesh... but it good to check
let see! 

@ftadel 
> > I'm not sure, that the integration of simnibs as it is, is possible!.
> 
> What are the blocking points?

I'm not sure if we can call headreco in the fly without the intervention of the user. 
It needs to have SimNibs installed with all its dependencies, and it asks for a user account(need to re-check),  and the total size (with the full miniconda) is ~ 4 Go.  

> > Or we can ask the user to generate the head model as an external step from brainstorm, by orienting them to simnibs, after bst will load the mesh
> 
> This would be a nightmare, especially with these transformations of coordinates... not an option.

I agree

juangpc, 2019-12-20:
Hi, 
In relation to the accuracy of the simnibs to scs transformation.. The transformation can be checked with the scalp surface here. Important conclusions I've found:

 - The most important thing is to realize that simnibs output is all in voxel coordinates. Thus, given a mri, we need to transform from vox to scs to be able to work on brainstorm's ctf/scs coordinates.
 - Simnibs realigns and reslices the input T1, then non-linearly wraps T2 to be coregistered to T1. So T1 is basically not modified apart from the problems the reslicing might cause (I don't think that many). The output mris are saved by simnibs with the word "conform" added to the mris name.
 - We know from simnibs documentation that the output meshes are aligned to the conformed mris, so. One solution would be to ask bst users to give up previous anatomical models when running simnibs. I don't think this is very elegant but it would have saved me (and you guys) all these lengthy posts. If this solution was adopted we would only need to import to bst the conformed T1 and everything would match perfectly. But if someone already has freesurfer or other anatomical data  (inlcuding forward and source models) they would need to throw them. 
I've learned how this is the actual behaviour of brainstorm when importing a freesurfer folder. So we would only need to replicate that same behaviour. Still I don't think it is a good solution. It is always better to be able to navigate between coordinate systems, given transformations between them. 
 - Another solution is to call simnibs with the option *no-conform* which fixes the input mri so that it is not modified. The problem is that we don't know how much the no-reslicing might affect the segmentation procedures. So we have opted to keep digging.
 - The following result confirms that we are able to transport data from conformed t1 (only realigned and resliced) to bst scs's coordinates. I think this is better, but there is always some risk of me having done something moronic. So I'll let y'all the decision. :)
 - Apart from this confirmation with the scalp surfaces, a last step will be to check the actual brain surfaces (wm and gm) but for that, I need to compute freesurfer on ernie and then simnibs on the output of freesurfer (Freesurfer also modifies the input T1 file, go figure...).

So, this is the registration between simnibs scalp and the original T1.

[image]

**Left is simnibs based scalp and right is bst based scalp (right-click in mri --> generate head surface).
I think (hope) all this is a step forward.**

To reproduce this image:
1. download install and run simnibs on example data subject ernie. No need to do *no-conform option* simnibs options can be all left  to the default values: 
```
headreco all ernie ernie_T1.nii.gz ernie_T2.nii.gz
```
2. In bst, new subject, and import mri. Import ernie_T1.nii.gz **no need to import conformed t1**.
This mri will be typically called **ernie_T1** in brainstorm. Apply transformation and define fiducials. This will in turn define the world2scs transformation (see later T1 and T2). 
3. Right-click on mri and generate head surface. Repeat this, we need to do this two times. One head surface will be bst one the second one will be "hacked".
4. Right-click on the second head mask--> File --> go to this directory (matlab). In there run the following:
```
%this fcn is part of simnibs. it can be found there.
ernie = mesh_load_gmsh4('path_to_simnibs_result_folder/ernie.msh');
regions=unique(ernie.triangle_regions);
tris=ernie.triangles(ernie.triangle_regions==regions(5),:); %region 5 is scalp
hmS=ernie.nodes(unique(tris(:)),:); %head mask Simnibs
clear ernie regions tris
```
@ftadel this next part is part of cs_convert.m
```
%build transformation matrix (from simnibs coord to bst scs coordinates).
sMri=load('subjectimage_ernie_T1.mat'); %this is the mri imported to bst.
T1 = diag([sMri.Voxsize(:) ./ 1000; 1]); 
T2 = [sMri.SCS.R, sMri.SCS.T./1000; 0 0 0 1];
T = T2 * T1 * inv(sMri.InitTransf{1,2});
clear T1 T2 sMri

%apply transformation hmS_scs stands for *head mask Simnibs in scs coordinates*
hmS_scs = [hmS ones(size(hmS,1),1)];
hmS_scs = (T * hmS_scs')';
hmS_scs (:,4)=[];
```
and finally this little hack
```
%substitute the coordinates of bst's headmask with simnibs'
load('tess_head_mask_02.mat','Vertices') %Second headmask
idx=randperm(size(hmS_scs,1)); %match the sizes of the head masks
Vertices = hmS_scs(idx(1:size(Vertices,1)),:);
save('tess_head_mask_02.mat','Vertices','-append')
```

**important:** the faces have all been screwed up so we need to right-click on this 2nd head mask and remesh. Then the new surface will be all set. Vertices' positions will be those given by the transformed simnibs coordinates. 
Finally, righ-click on the head masks and MRI registration --> Check MRI/surface registration for the two head masks in this subject.

juangpc, 2019-12-20:
Hi again, 
Since Francois, gave me the idea of using mricron, if you overlay the different results from simnibs on the original T1(not the conformed one). You can see this:
[image]
Green is the mask for the scalp ("skin" it is called in simnibs' context). And is exactly the same info I transformed to brainstorm scs and plotted in my previous post. So given that the scalp is correctly aligned to scs and that the gm/wm are correctly aligned to each other, I think we can consider the previous transformation to be a good one, for all the tissues.

ftadel, 2019-12-23:
> Simnibs realigns and reslices the input T1, then non-linearly wraps T2 to be coregistered to T1. So T1 is basically not modified apart from the problems the reslicing might cause (I don't think that many). The output mris are saved by simnibs with the word "conform" added to the mris name

Then there is no non-linear transformation of the T1 in the end? Even when NOT using the no-conform option?
And can we know (or recompute) the transformation that is applied between brainstorm's T1 and simnib's? 
If so, there is nothing preventing us from just inverting this transformation and save the convert the simnibs surfaces in brainstorm coordinates next to the "original" T1 (the one saved in the brainstorm database).
Am I missing something?

> One solution would be to ask bst users to give up previous anatomical models when running simnibs. I don't think this is very elegant but it would have saved me (and you guys) all these lengthy [...]

No, this doesn't work. One imports the FreeSurfer/CAT/BrainSuite output for the cortex surfaces, yes, but also for the anatomical atlases and the registration with the FSAverage atlas, which allows the coregistration between subjects in the context of group studies. Simnibs will not replace that, so we can't simply get rid of it.

Plus, it doesn't fit the current overall workflow in Brainstorm, which is:
1) import the FreeSurfer/CAT/BrainSuite anatomy
2) set anatomical fiducials NAS/LPA/RPA
3) import the recordings
4) coregister sensors with MRI
5) compute the head model

This suggestion of simply starting over from simnibs modified T1 at step #5 would get rid of the steps #2 and #4... which means that we need to reimport the sensor positions for each head model we compute. It doesn't work.

> and finally this little hack
> %substitute the coordinates of bst's headmask with simnibs'
> load('tess_head_mask_02.mat','Vertices') %Second headmask
> idx=randperm(size(hmS_scs,1)); %match the sizes of the head masks

Why randomly shuffling the vertices?

-- 

I will download simnibs tomorrow, but won't have time to work much on it before Christmas. I will test all this after January 6th.

juangpc, 2020-01-02:
Hi, back in business. (HNY!!!)
>Then there is no non-linear transformation of the T1 in the end? Even when NOT using the no-conform option?

That is correct. There is no non-linear transformation between input T1 and "conformed" one.  If no-conform option is used there is no transformation at all. If no option is used, by default, the transformation is a re-orientation to LAS (and a resampling, which won't affect us).

>And can we know (or recompute) the transformation that is applied between brainstorm's T1 and simnib's?

Yes. The actual transformation is NOT saved, so we can recompute it and invert it. In hmutils.py (line 319) there is a function called reorient2LAS. In it, the input T1 is reoriented if needed. If you want I can try to figure out the actual Matlab code for it and invert it. 

note. A (quite ugly) workaround would be to compute the mni tranformation of the input T1 because simnibs does provide a subject-space to mni 4x4 transformation. I leave up to you which solution to implement.

This is the actual code...

```python
def reorient2LAS(v):
    """Reorients axes so as to bring the volume in to LAS (left-anterior-
    superior) space, i.e. radialogical convention. This reorientation is not
    robust, however, in that it relies on finding the primary direction of each
    axis and assigning it to the corresponding dimension. As such, if an axis
    is more than 45 degrees off, this will fail to correctly match an array
    axis with its corresponding dimension.
    
    PARAMETERS
    ----------
    v : nibabel image object
        The image volume.
        
    RETURNS
    -------  
    vout : nibabel image object
        reoriented nibabel image object in LAS (radiological) convention.
    """    
    qform_inv = np.linalg.inv(v.get_qform())[:3,:3]
    qform_inv_sign = np.sign(qform_inv)
    
    # Get sorting to LAS. This should work except for a few "pathological"
    # cases (e.g. two 45 deg axes)
    _, LASdim = np.where((np.abs(qform_inv) == np.max(np.abs(qform_inv),axis=0)).T)
    LASsign = qform_inv_sign[LASdim,range(3)]
    LASsign[0] *= -1

    # Apply to qform
    p = np.zeros_like(v.get_qform())
    p[-1,-1] = 1
    p[LASdim,range(3)] = 1
    
    dims = np.array(v.shape)
    dims = dims[LASdim] # dimensions of the reoriented image 

    # flip axes to LAS   
    flip = np.eye(4)
    for i in range(3):
        if LASsign[i] < 0:
            flip[i,i] = -1
            flip[i,-1] = dims[i]-1
    
    # Apply to data
    data = np.transpose(v.get_data(),LASdim) # permute axes
    if LASsign[0]<0: data = data[::-1,...]   # and flip...
    if LASsign[1]<0: data = data[:,::-1,:]
    if LASsign[2]<0: data = data[...,::-1]
    
    # Make reoriented image
    vout = nib.Nifti1Image(data,affine=(v.get_qform().dot(p)).dot(flip))
    vout.set_qform((v.get_qform().dot(p)).dot(flip))
    vout.header.set_xyzt_units(*v.header.get_xyzt_units())
    
    return vout
```
We already have this code/functionality in brainstorm. It is in anatomy/cs_nii2bst.m called when a user applies the available transformation matrix in a mri, while importing it into bst. However, I would need to understand it a bit better to be able to apply it here.

>Why randomly shuffling the vertices?

Nevermind, that was just for testing purposes.

juangpc, 2020-01-03:
Hi again, 
I think I'll have it. Allow me to recap.
Simnibs called without options will (i) resample and (ii) reorient the initial T1. This will modify the image with a linear transformation which is not saved, but can be reproduced if needed. 
 - The resampling might cause some slight errors but far less relevant that the precision we will be dealing with meeg, so we can discard them. 
 - The reorient process will only switch axis (if needed) and performs a very similar operation to the one already implemented in brainstorm in anatomy/cs_nii2bst.m. So, my opinion is that if the user has already imported an mri to brainstorm when simnibs is invoked, no orientation should be needed. I've tested with 3 different mris and I've encountered no problems. Maybe with "strange" mris, out there, some problems might occur, but we can always attend those cases later (again, imho).

I've done some more testing. Simnibs has a testing subject called "ernie". I've segmented ernie's mri with freesurfer and imported the folder to brainstorm. Then I've combined the information from simnibs and brainstorm, considering ernie's gray matter and this is the result.
Blue = bst(freesurfer imported into bst), red = simnibs.

[image]

Both meshes seem to overlap pretty well.

[image]

Here is a slice. Sorry for the moving center... 

[image]

I went into bst's code a bit to understand what was going on. I don't understand why we apply a "reference change to input images" from 000 to .5 .5 .5 See io/in_mri.m in line 239. (@ftadel?) But, in any case... if we add this same reference change to the volumetric info from simnibs, everything matches with reasonable precision. 

Here is the code to transform the 3d info from simnibs meshes into brainstorm.
Simnibs info is in voxel's space. Since we have already imported the mri which simnibs uses, we can assume it has already been referenced and therefore scs's transformation matrices have already been defined.  Thus, sMri.SCS fields will be populated. With this transformation matrix we transform from voxel space to subject's space (brainstorm's ctf) the meshes created with simnibs.
 
```matlab
T1 = diag([sMri.Voxsize(:) ./ 1000; 1]);
T2 = [sMri.SCS.R, sMri.SCS.T./1000; 0 0 0 1];
T = T2 * T1 * inv(sMri.InitTransf{1,2}) * [1 0 0 -.5; 0 1 0 -.5; 0 0 1 -.5; 0 0 0 1];
```

tmedani, 2020-01-05:
Hi all & Happ New year, 

@Juan, this is great progress on the head model generation. 
I think we are too close... 

I did more testing with roast and fieldtrip, and I found that the simnibs process in the best at this time.
Here is the script to call all these tools from brainstorm gui (process_generate_fem.m) :
https://www.dropbox.com/transfer/AAAAAGXXQShx-2QmRBMPhvckuaYe2WIp0x-iK5Sv7TKeNSqJjeDIeN0
We can add your modification in this script for SimNibs 

Here are some points about the next steps : 
- Generation of the source space, (dipole positions): the best option is to generate a new surface in the middel of the GM layer. 
I have already a script to inflate or deflate a surface, but I'm not sure if it's the best option since the GM thickness is not regular. 

-  Add T2 MRI to the segmentation process, as it's mentioned in the SimNibs papers, the T2 improves the segmentation of the CSF and Skull. 
So, if it's available, we can add an option for the user to link the T2 image.
@ftadel @juangpc  ==> **do we need to ask again for the fiducial points?**  

- Also, @juangpc  is there any option within simnibs to separate the cerebellum from the cortex? 

- Add the Hexahedaral mesh model (FEM computation and BST visualization)

- Improve the BST display function for the FEM model (Hexa and Tetra) ==> view_surface_fem.m
(we can use iso2mesh functions for hexa, but before we need to convert the hexa to tetra)

juangpc, 2020-01-06:
Hi Tak,

>do we need to ask again for the fiducial points?

I don't think so. The registration is t2 --> t1. Perhaps we could do some testing but, I don't think there will be a problem here.

>  is there any option within simnibs to separate the cerebellum from the cortex?

Been looking around and it seems that gray matter includes cerebellum. If someone wants gm not to include the cerebellum, I can think of. 
 - run the process normally. 
 - in the templates/ folder there is a template mask 1/0 for the cerebellum. it is a nifti file. We could use it to mask away all the triangles whose nodes appear lying in the masked volume. Why would you want to take the cerebellum out? Wouldn't it add a bit more accuracy to the model?

[66 more replies]

closed 2021-09-23 by ftadel