brainstorm-tools/brainstorm3 #196: Updating Tutorial "Introduction dataset"

juangpc, 2019-04-25:
Hi,

I'm trying to learn as much as possible Brainstorm as I can and figured I should better go through every single tutorial as a normal user. A bit of a pain but that way I intend to review code and structures etc...

So, I came across Introduction dataset: https://neuroimage.usc.edu/brainstorm/DatasetIntroduction
There is a "[TODO]" sign in the tutorial. Exactly here: 
https://neuroimage.usc.edu/brainstorm/DatasetIntroduction#line-40

I'm trying to figure out:  
 - Can I help get rid of it? What does it mean? I mean, what is wrong with the file description?
 I reviewed the history and it was added by @Francois in rev11 with some other '[TODO]'s which were later edited. See:
https://neuroimage.usc.edu/brainstorm/DatasetIntroduction?action=diff&rev1=10&rev2=11
 - There is another line where FieldTrip is mentioned which I don't really understand. Exactly here: 
https://neuroimage.usc.edu/brainstorm/DatasetIntroduction#line-47
 - Should I ask these type of questions here in a normal GitHub issue? Some other issues explicitly relate to the tutorials, and I figured that this is more a development issue more than something to discuss in a forum with BST users.

Thanks.

juangpc, 2019-04-25:
In relation to FieldTrip line, with a fast G search,  it appears to come from this other OpenNeuro file:
https://openneuro.org/datasets/ds000246/versions/00001/file-display/README

ftadel, 2019-04-25:
Thank you for this detailed proofreading!

This tutorial was initially supposed to be produced in close collaboration with the FieldTrip team. We designed the specifications of the experiment with them, but unfortunately after the acquisition we never heard back from them... We've been waiting for their validation for a few years until I finally decided to change the title of this auditory dataset and removed the mentions to FieldTrip.

The TODO block was maybe dating from a time when we were not sure if we would distribute downsampled recordings. I don't think there is anything to do left in this introduction, so I simply removed it. I also removed the other line mentioning FieldTrip preferring regular .ds files than extra AUX files. These are two types of files that can be saved simultaneously by the CTF MEG system, Brainstorm can deal with both, FieldTrip can't, and in general no one uses these AUX files. We don't care about this info here.

(it doesn't matter if the line stays in the OpenNeuro dataset, I'm not going to create a new version of the the dataset just to remove this line - which might be of interest for someone, some day)

In general, I think issues are OK for discussing issues in the documentation. For technical details or typos (maybe like this one), a simple email could be preferable in the first place, to avoid flooding all the github followers with info of not much interest to others. I guess the fewest messages we send, the highest the chance the important ones get addressed :)

Thanks for your comments, this is all very useful!

closed 2019-04-25 by ftadel