brainstorm-tools/brainstorm3 #208: Minimum iEEG tab values should be lower

mpompolas, 2019-05-17:
The minimum values that can be set within the iEEG window for electrode metadata (electrode diameter, contact diameter etc.) seems to be 0.1 mm.

I got contacted by some mice people that want to use Neuropixels:
http://www.open-ephys.org/neuropixels

The selling point of these electrodes is how insanely dense they are (1000 channels within 10mm length and 70 micron width).

Can we lower these minimum values?

ftadel, 2019-05-22:
Done in https://github.com/brainstorm-tools/brainstorm3/commit/93cbe223161746b2fadc14e46062014d590725bb#diff-5d03d3ef5441acf005318a4a8bd366b4

You can go down to 0.01, and therefore enter your 70microns length.
If this is OK with you, please close this issue.

mpompolas, 2019-05-22:
Yes, each column is 0.016 mm apart according to the attached figure, so we're good.

[image]

mpompolas, 2019-05-30:
extension of this problem when trying to assign:
Set tip
Set skull 
entries on a mouse.

[image]

ftadel, 2019-06-03:
I lowered the distance check to 2mm.
Is that OK?

mpompolas, 2019-06-06:
It appears to be ok,
Thanks

closed 2019-06-06 by mpompolas