brainstorm-tools/brainstorm3 #219: Integration with SPM12/CAT12 segmentation

ftadel, 2019-06-21:
CAT is an extension of SPM12 which include surface-based morphometry tools (SBM) which we could use in Brainstorm for computing individual cortical surface from a MRI.
http://www.neuro.uni-jena.de/cat/

The output we can expect using from CAT: 
- Central cortical surfaces for the left and right hemispheres (surface half-way between the white and the pial FreeSurfer surfaces)
- Registered spheres for inter-subject registration
- All the atlases available in FreeSurfer

Main advantages for Brainstorm users: It runs much faster than FreeSurfer (~1hr per subject) and is platform-independent (Windows, Linux and MacOS).

ftadel, 2019-06-21:
Current issues:

## 1) How to generate atlases in native space?

The surfaces are generated in native space (surf/*h.central.subjid.gii) and registered to the fsaverage space through the registered spheres (surf/*h.sphere.reg.subjid.gii). 
The atlases are available in the CAT12 installation folder (spm12/toolbox/cat12/atlases_surfaces/*.annot) but only on the fsaverage surfaces, which are not matching point-to-point with the individual surfaces generated by CAT.

Response from Christian:

> The transformation of the annotation atlas files from fsaverage space to native space will be not very straightforward. I have not yet prepared that because you need to transform and interpolate each region separately to avoid interpolation issues with the binary atlas information. 

## 2) What is the "native space" of the surfaces generated by CAT?

When processing a subject MRI (subjid.nii) with CAT, the surfaces are generated in native space (surf/*h.central.subjid.gii). But these .gii surfaces or the other CAT folders do not seem to include any transformation explaining how the vertices of the surfaces relate to the voxels of the original MRI.
If we simply apply the inverse of the 4x4 sform transformation from the original .nii (https://nifti.nimh.nih.gov/nifti-1/documentation/nifti1fields/nifti1fields_pages/qsform.html) we don't obtain valid voxel coordinates.

Reponse from Christian:

> This should be comparable to the Freesurfer coordinate system of the fsaverage template after resampling (=applying spherical registration from sphere.reg). The native surface are of course in native space... 
>
> The surface can be transformed to fsaverage space using the registered spheres in surf ([lr]h.sphere.reg.*) using the resampling&smooth function. Then you can decide to merge hemispheres and select either 32k HCP or 164k fsaverage space.
The volumes need to be normalized by saving the deformation field (forward deformation). These fields are not saved by default, because you usually only need them for applying spatial registration to other (co-registered to the T1) modalities such as DTI/*fMRI.
>
>In SPM there is also a function to apply 3D deformation fields from volume registration to surfaces, which is not the recommended way, because you don't make use of the spherical registration. However, I don't know whether you need the spherical registered surfaces, which I usually use for surface analysis or rather the native surfaces simply in MNI space. Then the SPM function would be fine for that purpose.

tmedani, 2019-09-10:
Hi Francois, 
Thanks for this information shared in this thread. 

I tested this pipeline from brainstorm and it's super easy to use and fast to generate the surfaces from the MRI. 

I did not have a look at the code right now, but I think this could be another approach that we can test for the FEM head model.

Since the CAT process is based on SPM, I think somewhere within the code other tissus (WM, CSF, GM) could be extracted but not included in the whole process.  

Do you agree? 
From your point of view, should I have a deep look at that and check or not? 

Thanks 
Takfarinas

ftadel, 2019-09-11:
> Since the CAT process is based on SPM, I think somewhere within the code other tissus (WM, CSF, GM) could be extracted but not included in the whole process.

Indeed: There are 6 volumes in the TPM.nii, so there are at least 5 tissues you should be able to extract from the SPM12 or CAT12 processing pipeline. Yes, I guess you could probably use these masks for the volume meshing. 
I'm not sure this is something that should be done from SPM or CAT, but the code is probably readily available in one or the other. It's probably worth investing time in exploring this track.

I'm closing this issue now, because the initial goal was matched with this commit:
https://github.com/brainstorm-tools/brainstorm3/commit/44cc28527988621d00ed87b4a09fe6461e7456d8

Open another issue if you want to discuss this further, or continue on the other FEM issue.

closed 2019-09-11 by ftadel

tmedani, 2019-09-12:
Thanks for the information.
I will have a look and if we need, I will open a new issue related for volume mesh generation.
A+