brainstorm-tools/brainstorm3 #221: Raster plot visualization needs some adjustment

mpompolas, 2019-06-21:
The raster plot shows the spiking activity of each neuron.  There is a problem though with the visualization of this activity. 
Since the raster plots are not binned (no temporal smoothing), the x-axis (time dimension) is a much larger vector than the y-axis (number of trials). This creates a rendering problem when the figure is scaled, as shown on the next two figures:

[image]

[image]

Maybe a visualization with dots would be appropriate?

ftadel, 2019-06-25:
Probably. 
Dots can be created with the function line. Example:
```
line(X, Y, Z, 
   'Color', [1 0 0], ...
   'LineStyle', 'none', ...
   'Marker', '.', ...
   'MarkerSize', 1, ...
   'Parent', hAxes);
```

mpompolas, 2019-06-25:
@martcous can you please take a look at this?
If I'm not mistaken, figure_timefreq is the function that needs to be modified to show dots instead of weird lines. 

Here is an example raster file: 
https://www.dropbox.com/s/6mewjk8stfdm9z5/rasterplot_problematic.mat?dl=0

Thanks

ftadel, 2019-06-26:
If you really need to display dots instead of an image, you guys should maybe think about writing a new figure function? The time-freq plane is not useful at all in this case

martcous, 2019-06-26:
The time-frequency is useful in the sense that it gives us the dropdown interface to choose which electrode to display in the Display tab:
[image]

I will need to be pointed in the right direction if you want me to write an entirely new figure function. To me it sounds like a lot of work rather than a simple if/else condition in the time-freq display function.

ftadel, 2019-06-26:
Up to you, if you already have a solution in mind that would satisfy Konstantinos, go for it!
It's not like if we were developing a super clean structure around these time-frequency objects anyway :)
(eg. connectivity matrices and PAC results squeezed-in not-so-gracefully)

mpompolas, 2019-06-26:
Just did a very quick search. This tool might do the trick. Not sure how easy it is to implement within the interactive windows, but it would be a start:

https://www.mathworks.com/matlabcentral/fileexchange/45671-flexible-and-fast-spike-raster-plotting

martcous, 2019-07-03:
Fixed in https://github.com/brainstorm-tools/brainstorm3/commit/65c283cb246e8e70c26b0576985ea78e55dbbf0f

[image]

mpompolas, 2019-07-04:
Just checked the rasterplot. It is a bit sparse on my display. It only shows patterns when I make my window very small.
Can you please make the dots a bit bigger @martcous ?
Thanks

martcous, 2019-07-05:
Sure, I increased the dot size. I also properly implemented the "All Channels" display so that you can view a low resolution snapshot of all channels at once. It's definitely not perfect since these are small dots so a lot are lost with downsizing, but it already gives you an idea of which channels have more than others.

[image]

mpompolas, 2019-07-05:
this is actually neat.
I wanted to be able to display several neurons simultaneuously but I thought it would be an overkill to implement.

It's great you did it. The only comment I would make, is to present the 2d rasterplots from all neurons, on the same column (instead of a 3x4 subplot that you have, make it a 12x1). This way it should show the temporal selectivity of the neurons more appropriately.

This would be a problem if we have too many neurons though. Is it possible to select a few neurons (like a montage) instead of all of them?

Something like the way fieldtrip is set:

[image]

martcous, 2019-07-05:
Maybe in version 3.0! ;)

ftadel, 2020-11-09:
@mpompolas @martcous 
Do you want to keep this issue open?

mpompolas, 2020-11-10:
This is not something anyone is working on at this point, so I would say we close it and revisit it if ever needed.

closed 2020-11-10 by mpompolas