brainstorm-tools/brainstorm3 #223: Line display for cardiac events creates full page vertical lines

jeremymoreau, 2019-06-28:
I've been using the new "Lines" display option for events (which is great for spikes btw!), but it creates distracting full page lines for cardiac events for some reason. In the "Dots" view these non-channel events show up as dots at the top of the page, which is the expected behaviour. See screenshots below.

[image]
[image]

ftadel, 2019-07-01:
No, I actually intended it this way: in the "lines" mode, all the events are displayed as lines. This is useful as well for people who do not use channel events, or can be used to highlight some pars of an ERP for instance:
https://neuroimage.usc.edu/brainstorm/Tutorials/EventMarkers#Display_modes

Indeed, the cardiac events can be very annoying in this mode.
This is why I also added an option to mute some event groups (easy toggle with shortcut key H):
https://neuroimage.usc.edu/brainstorm/Tutorials/EventMarkers#Hide_event_groups

ftadel, 2019-07-01:
(I closed it, but please reopen if you think I really need to fix something here)

closed 2019-07-01 by ftadel

jeremymoreau, 2019-07-02:
No, I agree with you. I can see this being useful for some people. I wasn't aware of the hide event groups option, that's awesome! Thanks!