brainstorm-tools/brainstorm3 #225: Leff ratio in dipole scanning and whitened recordings

ftadel, 2019-07-25:
@jcmosher suggested we change the scaling factor applied in two functions: 

**Dipole scanning**

https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/process/functions/process_dipole_scanning.m#L119
`Factor = sqrt(DataMatP.Leff / sResultP.Leff);`
Replaced with:
`Factor = sqrt(sResultP.Leff);`

Problem, when testing with the introduction tutorials, the goodness of fit of the dipoles drops to very small values, example for standard condition at 90ms (Leff=193):
[image]

Current code (left) vs. John's suggestions (right):
[image]  
[image]

**Whitened recordings**

https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/tree/tree_callbacks.m#L3280
`Factor = sqrt(DataMat.Leff / ResultsMat.Leff);`
Replaced with:
`Factor = sqrt(sResultP.Leff);`

This looks more reasonable. Below, figures obtained by right-clicking on the MN source link for the standard condition (Leff=193) > Model evaluation > Save whitened data (left=before, right=after modification):
[image]   
[image]

@jcmosher @dimitriospantazis please validate whether this is what you want to obtain.

jcmosher, 2019-08-07:
https://neuroimage.usc.edu/brainstorm/Tutorials/TutDipScan

ftadel, 2019-08-12:
@jcmosher  ??

ftadel, 2020-04-27:
@jcmosher @dimitriospantazis @sbaillet 
If you have time now, can you please have a look at this?

ftadel, 2020-11-09:
@jcmosher @dimitriospantazis @sbaillet
Do you just want to ignore this question until someone spots a possible bug?

ftadel, 2021-09-19:
This issue is still pending....

ftadel, 2022-06-08:
@jcmosher @dimitriospantazis @sbaillet