brainstorm-tools/brainstorm3 #229: Forum questions waiting for an answer

ftadel, 2019-09-16:
Sept 2019:
* https://neuroimage.usc.edu/forums/t/diagonal-loading-value/13907
* https://neuroimage.usc.edu/forums/t/computing-connectivity-based-on-source-space/13678 (Connectivity for unconstrained sources)

August 2019:
* https://neuroimage.usc.edu/forums/t/compute-eeg-sources-with-sloreta/13425 (sLORETA)
* https://neuroimage.usc.edu/forums/t/calculating-cross-talk-functions/13485 (Min norm)
* https://neuroimage.usc.edu/forums/t/help-for-connectivity-pipeline/12558/7 (Connectivity for * unconstrained sources)
* https://neuroimage.usc.edu/forums/t/longitudinal-analysis/13047 (Statistics)

July 2019:
* https://neuroimage.usc.edu/forums/t/rank-deficiency-and-source-reconstruction/12758/9 (Min norm)

May 2019:
* https://neuroimage.usc.edu/forums/t/pvalues-for-spectral-granger-analysis/11045 (Granger spectral)

February 2019:
* https://neuroimage.usc.edu/forums/t/display-dipoles-across-subjects-on-default-anatomy/9779/10  (Dipole scanning)

October 2018:
* https://neuroimage.usc.edu/forums/t/source-estimation-options/8489 (LCMV)
* https://neuroimage.usc.edu/forums/t/error-in-simulate-recordings-from-scouts/8087/9 (SNR)

August 2018:
* https://neuroimage.usc.edu/forums/t/question-about-the-workflow-of-functional-connectivity-between-rois/7845 (Connectivity for unconstrained sources)

ftadel, 2019-09-18:
https://neuroimage.usc.edu/forums/t/source-modelling-and-subcortical-structures/10269/15

tmedani, 2019-09-18:
Hi @ftadel 
I'm not getting notifications from the forum, 
is there any option to activate for that? 
Thanks

ftadel, 2019-09-19:
> I'm not getting notifications from the forum,
> is there any option to activate for that?

This is something that can be configured in the user preferences on the forum:
[image]