brainstorm-tools/brainstorm3 #235: Stat tab: Renaming parameter "p-value threshold"?

ftadel, 2019-10-11:
Richard - Oct 8, 2019:

Hi Dimitrios
in the dialog box for our statistical test, we offer an FDR correction option and a box where users can enter a p-value.  Sorry if this is obvious, but shouldn't this be the q-value (FDR)- ratio of false to total positives? Or have I missed something where this can be translated to a p-value (and if so, what the heck does the  p-value represent in terms of a null hypothesis)?

thanks for any light you can shed!

ftadel, 2019-10-11:
Dimitrios - Oct 8, 2019:

Yes, you are right! The FDR correction option should rename the 'p-value threshold' to 'q-value threshold'. It escaped me all these years though I have used it as well.

I cc Francois who can revise Brainstorm to dynamically rename the field (p-value, p-value, q-value) as the user toggles between 'Uncorrected', 'Bonferroni', and 'FDR'.

ftadel, 2019-10-11:
Francois - Oct 9, 2019:

Hi Dimtrios,

So you just wanted to have the label changing in the Stat tab:
- Uncorrected and Bonferroni: "p-value threshold"
- FDR: "q-value threshold"

It's already online: https://github.com/brainstorm-tools/brainstorm3/commit/6e2b32513ebe59ba14c8c6b9247da9aebfa103dc
If it's not what you wanted, please let me know ASAP.

Dimitrios: Can you please add a line to explain this in the Statistics tutorial (section "FDR correction")?
https://neuroimage.usc.edu/brainstorm/Tutorials/Statistics#Correction_for_multiple_comparisons
Otherwise it looks a bit weird that it changes by itself...

Thanks
Francois

ftadel, 2019-10-11:
Dimitrios - Oct 9, 2019

Hi Richard and Francois,

[Francois, if Richard agrees please make the minor edits in the tutorial because I tried to login but somehow I failed to get edit access]

 

I am sorry for this, but on second thought I think we should rename the field into “Significance level α” consistently for all cases (uncorrected, Bonferroni, FDR). This is already consistent with the brainstorm tutorial which uses the variable α everywhere, and is universal in literature, e.g. here:

https://en.wikipedia.org/wiki/Q-value_(statistics)

Interpretation

The q-value can be interpreted as the false discovery rate (FDR): the proportion of false positives among all positive results. Given a set of test statistics and their associated q-values, rejecting the null hypothesis for all tests whose q-value is less than or equal to some threshold α ensures that the expected value of the false discovery rate is α.[5]

We need only make minor revisions in the tutorial:

 

“If we select a critical value of 0.05 (p<0.05)” -> “If we select a critical value of 0.05 (α<0.05)”

 

The false discovery rate (FDR) is another way of representing the rate of type I errors in null hypothesis testing when conducting multiple comparisons. It is designed to control the expected proportion of false positives among all positive results at a given significance level α, while the Bonferroni correction controls the probability to have at least one false positive.
 

Best,
Dimitrios

ftadel, 2019-10-11:
Hi Dimitrios,

For what I understand from this wikipedia page, "α" = "q-value threshold".

One thing is to be mathematically correct, another is to have other people understand what you're doing. The former is useless without the latter :)
I would tend to think that "q-value threshold" and "p-value threshold" and not "too incorrect" but "much easier to understand" than a flat "α". We compute cortical maps of p-values, what's so wrong about thresholding them interactively using a "p-value threshold"...

Could tell me in a clear way what I'm supposed to modify in the tutorial, at least for the q-value thing? Or let me know what's wrong about the wiki that prevents you from editing it, so we can try to fix this.

Thanks

dimitriospantazis, 2019-10-11:
Hi Francois,

My proposition was not 'α' but 'Significance level α'. Or we can call it 'Error rate α' though I prefer the former. This is universal in the statistical literature, used in our tutorials, and also in mne and fieldtrip:
https://mne.tools/dev/generated/mne.stats.fdr_correction.html
http://www.fieldtriptoolbox.org/reference/ft_statistics_montecarlo/

Best,
Dimitrios

martcous, 2019-10-11:
I agree with François that "q/p-value threshold" is much easier to understand. If you still choose to change it to "Significance level α", I think it would be important to add a tooltip information and maybe a question mark button next to the field that gives you a quick explanation of what it is and how it relates to q/p-values.

dimitriospantazis, 2019-10-11:
I understand your concerns. Unless there is feedback from others perhaps we can defer this to one of our biweekly meetings?

Cheers,
Dimitrios

richardmleahy, 2019-10-11:
good idea!

Richard M. Leahy, Chair
Department of Electrical and Computer Engineering (Systems)
University of Southern California

sbaillet, 2019-10-11:
folks: I’ll be travelling next week: sorry but I will have to miss our meeting.

Martin would like to present some new search features he is adding to BST: he could present at our second to next meeting.

Best wishes,

Sylvain.
-
Sylvain Baillet, PhD
Professor, Montreal Neurological Institute
McGill University
https://twitter.com/sylvain_baillet
- > more<https://www.mcgill.ca/neuro/sylvain-baillet-phd>

ftadel, 2019-10-29:
Done in: 
https://github.com/brainstorm-tools/brainstorm3/commit/495b328715dd5a04b339dbd025d5c8897d455543
https://github.com/brainstorm-tools/brainstorm3/commit/676ba217b75ec38e3b8e8ba86820e38bc10da31f

The interface now looks like this:
[image]

The process to apply a given threshold looks like this:
[image]

The output file names include the tag "alpha=" instead of "p<":
[image]

I modified all the tutorials that were using statistics:
- I fixed all the screen captures
- I replaced the text "p<" with "alpha="

closed 2019-10-29 by ftadel