brainstorm-tools/brainstorm3 #238: MEG sensors are displayed smaller than they are

juangpc, 2019-11-14:
Hi all, 

A few days ago, @jcmosher made me realize that the display of the coils was not showing the actual dimensions of the sensors correctly. When you display the channels for a triux system, this is what you see:
[image]

However, when reviewing Elekta's manuals this is what it should look like. 
[image]

At the moment, the distance between points defining a sensor ("patch") in bst is 12.9 mm. In Elekta's manuals, such distance is 28 mm. So, apart from the visual discrepancies, I guess something needs to be done. It is important to mention that Elekta's manual seems to represent the "chips" or the washer, not the actual pickup coils, but still, the coils are 25.80 mm on each side (not 12.9 mm).

****tldr** :)
We need to either modify the way channel information is stored in the database, or modify the way sensors are displayed. (And maybe we need to end up doing both things).**

--- 
What its actually going on in the code is as follows. First, the channel dimensions information is only read from the FIFF file when imported to bst. So: toolbox/io/in_fopen.m --> in_fopen_fif.m --> in_channel_fif.m

Inside in_channel_fif.m [line numbers correspond to [d85f372](https://github.com/brainstorm-tools/brainstorm3/commit/d85f37200d9c10556dce82311293782b5ad730c4)]:
Here is where the information related to each channel is confronted with a "database" of known vendors/models which is a mne file called "coil_def.dat". 
Comments on lines 150 and 153 are quite misleading because the transformation does not take place at that point in the code. At that point, only the transformation matrix ("coil_trans") is modified. The actual transformation from the sensor coordinate system into the CTF coordinates is done in line number 186.

**In line 185 Accuracy is set to 1. [[This is important for another related issue]].**
And the transformation from the sensor coordinate system takes place in line 186 through a mne function called mne_add_coil_defs.

info.chs array has "loc" and "coil_trans" but that is not used (probably legacy info?)
What it is actually used is info.chs(i).coil_def.coildefs. There is another field called FV but that is not used.

"Channel" array struct is built with all previous information and with coordinates already transformed to ctf coordinates, (info.chs(i).coil_def.coildefs) and stored.
Once the "Channel" struct is in the actual bst database, everything else is discarded.

Problem 1. 
The info stored in Channel struct array is actually a set of 4 integration points. _Not the dimensions of the sensors._ FV stores the actual dimensions of the sensors, and it is also transformed to ctf's coordinate sys. The following picture shows everything:
 - green dotted line= physical gradiometer.
 - red dotted line = physical gradiometer.
 - blue dotted line = physical magnetometer.
 - small circles = integration points. **this is stored in Channel struct array (inside bst database)**. And those 4 points are needed in order to compute the forward model. So that info is needed, but it is not the actual dimension of the pickup coils.
 
[image]

If we want to "display the channel" we don't have the actual dimensions of each channel anymore. 
Actually, when displaying the sensors, everything happens inside "toolbox/gui/figure_3d.m" --> inside "PlotCoils" function. In this function, since we only have 4 points, a set of transformations are made in order to "re-compute" back gradiometer patches. All based in those 4 points, which are only integration points in terms of the e/m model of the sensor, and not(!) vertices of the actual coils. In fact, that actual spatial information was already computed correctly transformed, and then discarded as part of the "FV" struct.

I really think the sensors should be displayed as they are and not smaller. In case we agree, here are some ideas.  I would like to first say that I'm, eager to learn from everybody most probably better solutions, so no problem not going further with these ideas, but still, since I've gone through the code, I'd like to propose the following.

Proposed solution 1. 
I think (following @jcmosher) that the information regarding channel in the database should be updated in order to include this FV field which is now discarded. This will generate back-compatibility issues, at least regarding the coil display functions. 

Make "PlotCoils" function (in toolbox/gui/figure_3d.m) check for the presence of FV field in the channel struct array. if it is not there this simple snippet should allow enlarging the sensor sizes to the actual real sizes. I've tested it and the output is 25.80mm as it should be:

```matlab
chLoc = Channels(i).Loc(:,[1 2 4 3])' .* 1.00;
>>>> line 4013 in figure_3d.m
zeros(size(chLoc));
aux(1,:)=0.5*(2*chLoc(1,:)-chLoc(2,:)-chLoc(4,:));
aux(2,:)=0.5*(2*chLoc(2,:)-chLoc(1,:)-chLoc(3,:));
aux(3,:)=0.5*(2*chLoc(3,:)-chLoc(2,:)-chLoc(4,:));
aux(4,:)=0.5*(2*chLoc(4,:)-chLoc(1,:)-chLoc(3,:));
chLoc=chLoc+aux;
<<<<<<
```
This produces the following result, which resembles the true sizes of the coils. And is completely compatible with today's bst.
[image]

However, if we do update way channel info is stored in the Channel struct array, and therefore the FV field is present in Channel, then FV can be used to directly draw the patches correctly with this code:

```matlab
patch('Vertices', info.chs(i).coil_def.FV.vertices, 'FaceColor', [1 1 0],'Faces',info.chs(i).coil_def.FV.faces);
```
Related issues:

1. Related to this issue, there is this file named "coil_def.dat".  This file is part of the mne suite. As compared to an updated mne version, it seems like the file in the mne project has had some updates regarding Compumedics specification since the last time it has been ported to bs repo. Should we update the info in bst's version of the file coil_def.dat to include new templates for Compumedics?

2. Accuracy. 
Regarding the accuracy with which sensors are modeled, as part of the forward modeling process, 
@jcmosher can confirm how now mne uses accuracy = 2 while modeling the sensors. Apparently it used to be too computationally expensive to increase the accuracy, but not so much anymore. Thus, 
if we modify the Accuracy value to 2, in line 185 of in_channel_fif.m, then the result of the actual transformation function model will no longer be 4 points, it will be 8. 

- This change will make inevitable to update the logic by which channels are displayed, which was the origin of all this discussion. 
- The forward model will increase its computational load and its accuracy.
- We need to make sure the lead field computations are ok with this change. (having 8 points to specify each sensor, as opposed to having 4). 

What shall we do?

Moo-Marc, 2019-11-14:
Also, CTF coils are circles. :)
I had a look at this integration point accuracy parameter a while back.  Here were the results for a CTF system:
[image]
So going from 4 to 7 points per coil had negligible improvement, especially compared to the roughly 10-20% relative error for spherical models.

juangpc, 2019-11-14:
While discussing with @tmedani we came to this summary of the problem:
 - We are using the computational model of a sensor to display them.
 - We should separate the computational model from the display of the physical sensor.
 - Maybe, for advanced users, we could also display the number of integration points used while computing the model of the sensor.

sbaillet, 2019-11-14:
I like your suggestions, Juan.

Best wishes,

Sylvain.
-
Sylvain Baillet, PhD
Professor, Montreal Neurological Institute
McGill University
- > more<https://www.mcgill.ca/neuro/sylvain-baillet-phd>

ftadel, 2019-11-15:
Thanks for this summary.
Everything you describe was intended to be this way, because it was decided that it was not a priority to spend more time on it and that moment, as displaying the accurate sensor geometry was not "needed". But this was 10 years ago, the graphic capacities of Brainstorm are now much higher and we could expect this sensor representation to be accurate.

I am not in favor of saving any additional display information in the channel structure. Modifying base structures is always painful, it creates all sorts of hidden bugs that we keep carrying for a while. 
We can reconstruct dynamically all the information we need for correct sensor display, therefore it is even better not to store any redundant information in the database. Making it dynamic would also apply to all the existing databases. 

So I think the way to go is to modify only PlotCoils. I've already pushed your suggestion for Elekta magnetometers:
https://github.com/brainstorm-tools/brainstorm3/commit/f8a6b751ae6ed17ae6b13df765648e61d1d59773#diff-6c618e95639f3e0152ac5c2d6edebfc7

[image] 

Feel free to suggestion modifications for the gradiometers as well if you think this is necessary.
If you need to know exactly which sensor was used from the coil_def.dat, you can get its name in the .Comment field of the channel.

> Also, CTF coils are circles. :)

Same logic, modify the PlotCoil function.
Anybody interested in working to make the CTF coil display more realistic? @Moo-Marc ?

> Should we update the info in bst's version of the file coil_def.dat to include new templates for Compumedics?

Yes, please do. Take it from MNE-Python rather than MNE-Matlab, if the two are different.

> Accuracy. Regarding the accuracy with which sensors are modeled, as part of the forward modeling process @jcmosher can confirm how now mne uses accuracy = 2 while modeling the sensors. Apparently it [...]

This discussion comes up every few years, and invariably, the conclusion is that there is no difference between 4 points and 8 points per coil. FieldTrip and SPM use one point only (center of the coil), and John tested previously using one of the phantom datasets (CTF or Elekta) that it was producing almost exactly the same results anyway. So why bother filling up the memory for nothing?... For the spherical models, it wouldn't make a huge difference to double the number of points where when compute the leadfield. But for BEM and FEM models, it would make it completely unmanageable.
I vote for keeping it with Accuracy=1.

ftadel, 2019-11-15:
PS: In my screen capture (produced with your suggested code), the coils are smaller than in your screen capture. Are you sure this is the code you used to produce this image? Or this is another Elekta machine with different sensor sizes (the one I've been using is the Wakeman/Henson dataset used for the group analysis tutorial)

Moo-Marc, 2019-11-15:
> Anybody interested in working to make the CTF coil display more realistic? @Moo-Marc ?

Unfortunately, I realistically won't have time to work on this anytime soon...

juangpc, 2019-11-19:
Thanks for your comments @ftadel  and @Moo-Marc. And sorry for making the group revisit old discussions. I personally don't agree. I prefer the simulations and models to be as precise as possible, even if it is a minimal effect, who knows, etc... but I'm good with anything you are good with. 

Yes, @ftadel to your question, indeed they are different sensors.
- GroupTutorial is using a T3 sensor: "Vectorview magnetometer T3 size = 21.00  mm".
- I was showing a T2 sensor: "Vectorview magnetometer T2 size = 25.80  mm"
If you check the chLoc matrix after applying the patch you'll see how the distance between sensor vertices are those previously mentioned.

The problem with the code as it is right now is the gradiometers display. If you plot with the ALL option, you will see this, which is inaccurate (gradiometers are actually bigger (like .1mm) than the magnetomters:
[image]

But if you multiply the integration points for the gradiometers by 2, as with the magnetometers, you get this, which is also inaccurate (sensors even bump within each other).
[image]

My take is that the problem here is that we are dealing with two types of info, e/m model of the sensors and physical model of the sensors. Each gradiometer has its own set of 4 integration points and they are different from the integration points for the magnetometer.
[image]

Anyway. if you accept the code modifications, in the pull request I've created, the problem is solved with a slight modification: magnetometers are scaled 2x and gradiometers by 1.556x (dimensions are actually correct to .1mm)

[image]

Hope it helps.

ftadel, 2019-11-22:
@juangpc Please close this issue if you think all your problems are solved.

juangpc, 2019-11-22:
@ftadel ok ok. Sorry for making you turn your attention into this. And thanks for the pull.

closed 2019-11-22 by juangpc