brainstorm-tools/brainstorm3 #253: Color map for single channel NIRS time course

Edouard2laire, 2020-01-28:
Hi there,

I was plotting the time course of a single channel NIRS by selecting it from the plot of all channels and hit enter. The image was plotted well but the colormap could be better or consistent with the all-channel plot (eg HbO in red, HbR in blue and HbT in green). It is the case when we plot all of the channels but not when we only show one channel. 

[image]

Do you think it is easy to fix? 
Thank you :)

ftadel, 2020-04-27:
Sorry, I missed this post... I'll have a look at it soon.

ftadel, 2020-04-28:
Would this simple fix work for you ?
https://github.com/brainstorm-tools/brainstorm3/commit/c7108fc9e22d7f5464e7f985390e53747ee13a0c

If it does, please close the issue.

Edouard2laire, 2020-04-28:
yes, that's great ! 
Thank you

closed 2020-04-28 by Edouard2laire