brainstorm-tools/brainstorm3 #258: Update The Neuro logo on spash screen

Moo-Marc, 2020-02-07:
Low priority, but available here:
https://mcgill.ca/neuro/visual-identity-guide-and-templates

sbaillet, 2020-02-07:
👍

https://mcgill.ca/neuro/visual-identity-guide-and-templates

ftadel, 2020-02-10:
I also removed the Cleveland Clinic logo and added the UTHealth one:
[image]

closed 2020-02-10 by ftadel

juangpc, 2020-02-10:
Thanks, on behalf of UThealth!

Since this logo-talking has already started... one question. MD's I work with, find kind of odd that the binary version of BST is a batch script file with no logo. I usually create a desktop link for them, but still, is this intentional? or if I find the time and push a logo-ed script will it be included? Just asking. Nothing really important. Thanks.

ftadel, 2020-02-11:
I'm not sure what you mean with "a logo-ed batch script".
We could add a link (.lnk file, like you desktop link, which can reference an icon) on top of the existing .jar and calling scripts for Windows (.bat) and Linux/MacOS (.command). But I think that it would be even more confusing to have both the .lnk and .bat files.

The proper Windows approach would be rather to have an installer that creates the desktop and start menu links. But then it makes the updates possibly a bit more complicated...

commit 50b2e63e (2020-02-10): Updated logos on splash screen (fixes #258)