brainstorm-tools/brainstorm3 #263: MRI viewer offset

Moo-Marc, 2020-02-19:
Hi all,

I ran into coordinate issues trying to create a rat template.  

First, the image in MRIViewer seems offset.  Compare this same Nifti volume imported in Brainstorm, and opened in my own NiftiViewer, (I made that tool years ago specifically to debug coordinate system issues - i.e. I spent a lot of time making sure it was correct).

[image]
[image]

Both software agree on the voxel coordinates (mine are indexed from 0) that correspond to the origin in "world coordinates" (SForm coordinates in the Nifti file).  So the coordinate transform is applied properly.  But they are obviously displaying different data at that point, with Brainstorm suspiciously off center in the right-left direction.  Note that the voxels here are 0.12 mm, so it's not a half voxel issue.  It might be a half millimeter issue.

Second, I selected the fiducials so the SCS origin would be centered.  Here it is on the MRI.
[image]
However, when displaying the MRI with surfaces, the origin is again offset.
[image]

This all makes sense if the MRIViewer display is bugged and offset by half a millimeter.  Then if I assume that and set the fiducials so the SCS origin matches the world origin (which appears off in the MRIViewer), everything look aligned on the MRI when displaying it with surfaces.

P.S. It would be nice if "align manually with MRI", opened the display centered on the origin.

Moo-Marc, 2020-02-20:
PPS. Actually any figure re-centers at the middle of the axis arrows when displaying the axes.  It would be better to center on the origin for easier zooming and rotating.
[image]

ftadel, 2020-02-23:
The conversion between MRI coordinates and "world coordinates" is handled here:
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/anatomy/cs_convert.m#L76

I'm not sure I remember how I came up with this 0.5 shift... Probably trial and error on some specific example in one of the functions that had it before the last modification of cs_convert: https://github.com/brainstorm-tools/brainstorm3/commit/0ac7e6b6f2cf1dc7dd07d3ea099ef7d2b1459115#diff-6754b9f032722a1eb3cc4cb9e6fdbada

The function cs_convert is probably what needs to be fixed. That would not surprise me that a 0.5 millimeter shift would have gone unnoticed at the human scale, even in SEEG. Maybe simply removing this line would fix the offset you observe. 
But then we need to double-check that all the other functions that were using it are still giving correct results: mri_coregister.m, mri_reslice.m, import_channel.m, process_epileptogenicity.m.
Could you try testing the first two?

Could double-check that your functions give the same results as MRIcron?

Could you share this high-resolution rat brain?

Thanks

Moo-Marc, 2020-02-26:
Hi François,

I exported my Rat Brainstorm subject https://box.bic.mni.mcgill.ca/s/kgw6OopM7TQmuY0. (By the way, export subject includes anat/@defaultsubject in the zip file.  I removed it.)  If you prefer the Nifti file https://box.bic.mni.mcgill.ca/s/tsVpttCrXdyNcpA.

I'll first have a look if removing the 0.5 shift fixes it.

[NiftiViewer](https://gitlab.com/moo.marc/MMM/-/tree/master/View) gives the world coordinates of both the SForm and QForm contained in the Nifti file.  I don't know any other software that does that.  It matches what MRICron shows, but gives more precision on coordinates.

[image]

ftadel, 2020-02-27:
Thanks, this is very helpful to have such a high resolution, it helps debugging issues that were never visible in cases where the voxel size was close to 1mm.

I think that what is bugged is the position of the cross (and hence the capture of the current position from the mouse), not the conversion between coordinate systems.

At slices X=82, Y=208, Z=83, we get the same slices in MRIcron and Brainstorm, the "World" coordinates are (0,0,0), but the position of the cross is incorrect:

[image]

I'll try to fix this today.

ftadel, 2020-02-27:
Fix attempt: https://github.com/brainstorm-tools/brainstorm3/commit/595faccfda87c766d9bd7d2613644f8548b0f52f

It fixes the coordinates of the images in the MRI viewer (interactions and cross display).
It also does two other things you suggested: adjusting the axis lines to the objects that are currently displayed, and centering the camera on (0,0,0) when adding the axis display.

If you think this fixes all the issues related with this rat brain display, please close the issue.

ftadel, 2020-02-27:
[image]

ftadel, 2020-02-27:
[image]

Moo-Marc, 2020-02-27:
Thanks François!  This looks much better.

However, I wanted to double check and fully understand the logic of the remaining half-voxel bits in the code.  I think there's still something odd.

First, I confirmed that Nifti coordinates refer to the center of voxels.  I thought this was also the convention in Brainstorm, though it is not stated in the [coordinate system tutorial](https://neuroimage.usc.edu/brainstorm/CoordinateSystems#MRI_coordinates) (to be added once confirmed). 

If we look at the coordinates of the first voxel in the rat MRI in Brainstorm, I get (I added decimals):
MRI: (0.117, 0.117, 0.117), though the tutorial says it should be (1, 1, 1). Maybe the tutorial should say (VoxelSize_x, VoxelSize_y, VoxelSize_z)?
World: (-9.434  -24.238   -9.590), but I get (-9.492, -24.297, -9.648), so it is half a voxel off towards the origin, as if it's giving the coordinates of the right-anterior-superior voxel corner.  I doubt that's what's intended.  I confirmed the same half-voxel shift throughout.  

To keep the MRI coordinates the same and get the correct world coordinates for the center of voxels, we need to shift by one voxel, not half, on line 76 (Nifti indices start at 0 and we effectively want them to start at 1).

    vox2ras = vox2ras * [1 0 0 -1; 0 1 0 -1; 0 0 1 -1; 0 0 0 1];

(The same fix applies on line 239 in in_mri.m.  I didn't do an exhaustive search of the code.)

For example here's the world origin after the fix.
[image]

ftadel, 2020-02-28:
Alright, let's switch to 1 voxel instead of .5 voxel, this will also make it more similar to what FieldTrip does. 
https://github.com/brainstorm-tools/brainstorm3/commit/85e9b973890cfb273fad038887f55a3989b526e0#diff-6754b9f032722a1eb3cc4cb9e6fdbadaL75

I'm very confused about this, I'm sure it comes from somewhere but I can't recall where... There is probably one use case for which we would need to use this half voxel, maybe for aligning with some MNI template. It may appear some day again, and if it doesn't, good for us. 
All the usages of cs_convert I checked are related with aligning different objects (different volumes registered with SPM, MRI and SEEG electrodes, ...), there is no way we can see a 0.5 voxel error in these cases, we're not even at a millimeter precision.

> First, I confirmed that Nifti coordinates refer to the center of voxels. I thought this was also the convention in Brainstorm, though it is not stated in the coordinate system tutorial (to be added once confirmed).

I added one line to the tutorial: https://neuroimage.usc.edu/brainstorm/CoordinateSystems#MRI_coordinates
Do you think it needs more details?

Moo-Marc, 2020-02-28:
Thanks François! I saw your commit and that you edited the tutorial.

closed 2020-02-28 by Moo-Marc