brainstorm-tools/brainstorm3 #267: MRI display/overlay and registration with DWI (4D images)

tmedani, 2020-02-25:
Hello,

I opened this thread in oder to discuss some parts of the MRI/DTI and later link it to the FEM computation. 

So, I have started to work on some part of the DTI and I'm trying to understand how I can use the brainstorm's builtin functions for this purpose. 

**1- Overlay MRI images on the bst MRI viewer:** Is it possible to overlay MRI images within brainstorm?
I was able to view this when I added a T2 to the T1, and then call the SPM registration, however, when I close this window,  I can not reach it again.

[image]

Is there a way to display both of the images on the same figure?
Also, what does it mean the color on the figure (and the color bar on the right)?

2- **Overlay MRI with surfaces:** it seems possible but I can not do it, there is an error on MatLab when I try to add a surface to the MRI viewer. 
Here is the scenario : 
[image]
[image]

[image]

**3- Register DTI (4D) to T1:** I'm trying to load and register DTI to T1 by following the same process as loading a second MRI for a subject. 
However, since the DTI is 4D data, the process fails. 

I have checked the code, and It seems that the function "  [MRI, vox2ras] = in_mri(MriFile, FileFormat, isInteractive, isNormalize)" is already prepared to read this data. 
however, the calling from bst is only for 3D.

After some modifications, I was able to load one DTI (only one direction on the 4th dimension). 
Here is the view of One DTI registered to the T1 MRI

[image]

@ftadel have you used this before for 4D data (DTI or fMRI) ?
is there any tuto? 

Maybe, we don't need to load all the DTI, however, I think it could be better to check if the DTI is aligned to the T1. 

**_The main purpose of all this is to be able to compute the conductivity tensor, here are the main steps:_ 
a - Register the DTI to the same T1 space (SPM). 
b-  Generate the mesh by one of the implemented methods (SimNibs).
c-  Compute the tensor on the brainstorm T1 space by calling Brainsuite (in progress).
d-  Convert the DTI tensor to conductivity tensors on each voxel (Tuch and al.).
e-  Maps each voxel to its tetra (or hexa) ... need to be fast
6-  For each element (tetra or hexa), associated it to its tensor (only the WM is considered)
7- Compute the anisotropic FEM head model**
 
A+
Takfarinas

ftadel, 2020-02-26:
> Overlay MRI with surfaces: it seems possible but I can not do it, there is an error on MatLab when I try to add a surface to the MRI viewer.

No, it is not possible to add surfaces to the MRI viewer with this button, I disabled it: https://github.com/brainstorm-tools/brainstorm3/commit/3f2f733de8e35e0400b7216d68959c07457f2401
To display the outline of a surface on the MRIViewer, you need to right-click on the surface > MRI registration > Check surface/MRI registration.

> @ftadel have you used this before for 4D data (DTI or fMRI) ?

Most functions cannot work with 4D volumes. Nothing in the interface is ready to handled volumes with multiple layers. 
I think the easiest would be to import 4D images as series of 3D files in the Brainstorm database. For DTI it would be only three files, right? It would make everything much easier to access.

This is the way most MRI viewers (like MRIcron, the SPM viewer...) work: they ask you which volume you want to load. We could either load only one, or load them all at once, depending on what you think is more interesting.

Can you send me an example file?

> is possible to use the function : out_mri_nii.m to save 4D data?

Yes, to export source maps estimated in Brainstorm (but that are not handled internally as "4D volumes") as 4D .nii files for processing in other programs (eg. SPM)

tmedani, 2020-02-28:
> No, it is not possible to add surfaces to the MRI viewer with this button, I disabled it: [3f2f733](https://github.com/brainstorm-tools/brainstorm3/commit/3f2f733de8e35e0400b7216d68959c07457f2401)

it was confusing since the option is displayed. Thank you for fixing it. 

> I think the easiest would be to import 4D images as a series of 3D files in the Brainstorm database.

Indeed, this is the solution that I found, after thinking and discussions, it seems to be not important to load these files to brainstorm for the users. 
It's nice to check that the DTI is aligned with the MRI (T1 as reference).
The BDP from brainsuite will do it during the process. 

From my point of view, the important data to display are the tensors on each element {as a sphere (isotropic case) and ellipse (anisotropic case)}. I have already prepared [some functions](https://github.com/tmedani/duneuro_interface/blob/master/toolbox/visualisation/bst_display_tensor_as_ellipse.m) to do that, but not adapted for brainstorm for now. 

> For DTI it would be only three files, right?

I'm always confused by the names DTI or DWI, 
Assuming the DWI is the correct name, so DWI should have more than 3 (the minimal is 6), 
but in general, there are more.  
Regarding the DTI, I assume it refers to the tensors, which is computed from the DWI, 
and it should have 3 orthogonal vectors (eigen directions) scaled by the associated 3 eigenvalues.

> We could either load only one or load them all at once, depending on what you think is more interesting.

Maybe just one, just for visual checking... and referer the users to Brainsuite for more options.

Can you send me an example file?
Yes, I will send you the link by email.

ftadel, 2020-03-03:
I downloaded the files.
What do you expect me to do with the files:
- _DTI.nii
- .bval
- .bvec

tmedani, 2020-03-05:
> I downloaded the files.
> What do you expect me to do with the files:
> 
> * _DTI.nii
> * .bval
> * .bvec

These files are needed for the next part of the projects. 
They are the DTI data and the information about the magnetic field (bval and bvec)
these data are used by the Brainsuite Diffusion Pipeline to estimate the conductivity tensors.

ftadel, 2020-03-06:
OK. And what do you expect me to do with these files?
What do you need me to develop / change in Brainstorm to help you with this task?

tmedani, 2020-03-06:
> OK. And what do you expect me to do with these files?
> What do you need me to develop / change in Brainstorm to help you with this task?

Thank you, Francois, 

I'm trying to learn about the DTI data, and understand some of the brainsuite processes that use the DTI and also the process of converting the diffusion tensor to the conductivity tensor.

I will translate all into schemes and then to matlab code and share it with you asap. 

From brainstorm side, we will probably need the :
- Display of the tensor overlayed on the mesh. 
- Integrate some scripts that call Brainsuite (I think the installation of brainsuite will be needed for the users).
- Play with the different coordinates systems in order to align all the MRI, DTI, Mesh and, tensors in the same space.   
- Think to an interactive way to call the 'Compute the Conductivity tensor'
It needs
      -  The MRI (T1/T2)
      -  The DTI (*dwi.nii, *.bval, *.bvec)
      -  The brainsuite scripts that generate the diffusion tensors
      -  The mapping code from diffusion tensor to the  conductivity tensor
      -  The final masks (results from the segmentation) <== we need to save these files from the 
                                                                          previous steps (segmentation spm, simnibs ....)
      -  The mesh, we need to map each voxel to a mesh element   
      -  The assignation of the tensor to the mesh.
      -  The display of the tensor (overlay either on the mesh or/and on the MRI)

By the way, later, it will be nice also to check and link this process to the 'Fiber Tracks' that @martcous  develop. 
We should expect that conductivity tensor will follow the fiber's direction.

ftadel, 2020-03-09:
Ok, so nothing needed from me at the moment, if I understand well?
Let me know when you need help with this.

tmedani, 2020-03-22:
@ftadel 

I'm doing some progress for the DTI, 

I think that the best option  to use for the mesh generation(for now and at this level) is the SimNibs. 

So, in order to have the process of the tensor generation working (DTI process), we need the final masks (output of the segmentation). 
(MRI + final masks + mesh)

With simnibs, this maks is saved here : 
maskfilename = bst_fullfile(simnibsDir, ['m2m_' subjid], [subjid  '_final_contr.nii.gz']);

So, here is a scenario and my questions : 

Assuming that the user will call the 'Generate FEM mesh '   then select simnibs (no DTI data). 

The DTI are available, the user has already the mesh, he can call later the DTI process (I added an option for the right-click: generate conductivity tensors) 

[image]

When this function is called, we will need to have the mask somewhere, so we need to save it before cleaning the temporary folder. 

how should we proceed?   

- If the user has the MRI+DTI,  we can embed the DTI process as an option on the 'Generate FEM mesh' and then we don't have to save the masks ? ==> less flexibility

- Save the masks and use them when the DTI process is called?  
==> where is the best place to save the masks (subject db or others ?)

- Where should we add the function "generate conductivity tensors" : 
 on the **anatomy**  or on the **Functional data** 

==> FYI, for the DTI, the users will need to install brainsuite.

Thanks & take care.

Best

ftadel, 2020-03-23:
> If the user has the MRI+DTI, we can embed the DTI process as an option on the 'Generate FEM mesh' and then we don't have to save the masks ? ==> less flexibility

Less flexibility for what?
Will you have lots of processing options for this "DTI process"?
The typical case would be: the user imports all the volumes acquired for the subject at once (so T1,T2,DWI are already in the database), and would like to run the minimum amount of processing steps. It's better to have one step of 4hrs than 2 steps of 2hrs each: you run it at night, and you don't have to connect remotely to your computer to run the next step.
So if there is one goal (anisotropic 3d mesh), I'd say it's better to have only one processing step from the user's point of view.

Other though: the least amount of useless intermediary objects in the database, the better. If we don't use these masks for anything else, then it's better to avoid storing them if we can.

> Save the masks and use them when the DTI process is called?
==> where is the best place to save the masks (subject db or others ?)

In the anatomy folder. I already did something like this for the FieldTrip segmentation process: the masks can be used later for the computation of the forward model.
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/process/functions/process_ft_volumesegment.m#L241

> Where should we add the function "generate conductivity tensors" :  on the anatomy or on the Functional data

It uses anatomical volumes in input and produces volumes of the same size in output, right?
So it can only be stored only on the "anatomy" part of the database.
And it makes sense, it is really anatomy, it's not functional imaging.

tmedani, 2020-03-23:
> Less flexibility for what?

I meant here that if the DTI data comes later than the MRI the user can not process it. 

> Will you have lots of processing options for this "DTI process"?

There are many steps within this process, but not too long (~ one hour for now), or less with code revision and optimization later.

> The typical case would be: the user imports all the volumes acquired for the subject at once (so T1, T2, DWI are already in the database)

for the T1 and T2 is already the case, but as discussed previously the DTI is a 4D and may have more than 20 images on the 4th dimension. I'm not sure that it useful to load them to the GUI or to db.

Also, there are two other files .bval and .bvec  related to the DTI. 
For now, the way that I use is when I call this function, a windows will on and ask the user to select the DTI file, and I assume that the associated files .bval and .bvec are in the same folder and have the same name as the DTI. 

> 1001.nii
> 1001.bval
> 1001.bvec

> It's better to have one step of 4hrs than 2 steps of 2hrs each: you run it at night, and you don't have to connect remotely to your computer to run the next step.
> So if there is one goal (anisotropic 3d mesh), I'd say it's better to have only one processing step from the user's point of view.

yes, we can add an option to call this process during mesh generation.... let see how we can do this when I will finish this function. 

> In the anatomy folder. I already did something like this for the FieldTrip segmentation process: the masks can be used later for the computation of the forward model.
> https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/process/functions/process_ft_volumesegment.m#L241

nice to know, I will have a look how you did it. 

> It uses anatomical volumes in input and produces volumes of the same size in output, right?

yes, it will produce a tensor matrix on each element (hexa or tetra), this could be loaded to the database. 
And could be nice to add a wayto display them (I'm also working on that)

 

> And it makes sense, it is really anatomy, it's not functional imaging.

ok, but this confusing, the DTI is considered as a functional analysis, but tensors 'no' ...

Thank you & stay safe

tmedani, 2020-04-19:
Hey @ftadel, 

I need your help regarding the MRI and the brainstorm coordinate systems. 

Here is the scenario : 
For a subject, we import the T1 and T2 to brainstorm and then set the fiducial points. 
We ran the SimNibs and the get the FEM mesh and the tissues(masks). 
Then I import to matlab the tissues and the mesh.

I want to display/overlay the mesh on the mask like this: 

[image]

but on Matlab script and data :

here is a piece of code : 

```
    case  'simnibs'
        [sMri, vox2ras]  = in_mri(MaskNii); % MaskNii is the 'tissues' from the segmentation
        % Re-write to the fieldtrip format
        mri_segmented = bstMri2ftMri(sMri, vox2ras); % no modification, just data format conversion from best to fieltrip
        % Assuming that the data are comming from SimNibs
        mri_segmented.anatomylabel = {'white' 'gray' 'csf' 'skull' 'scalp'};
        % Replace the eyes by the scalp
        mri_segmented.anatomy(mri_segmented.anatomy == 6) = 5;
```

then, in similar way, I import the mesh from bst

```
FemFiles = file_fullpath(sSubject.Surface(sSubject.iFEM).FileName);
femHead=  load(FemFiles);

% I have imported a low mesh resolution in this example (generated with fieltrip method)
 
% Apply the transoformation from scs to vox
[femHead.Vertices_voxel, Transf] = cs_convert(sMriMask,  'scs',  'voxel', femHead.Vertices);
[femHead.Vertices_ras, Transf] = cs_convert(sMriMask,  'scs',  'world', femHead.Vertices);
[femHead.Vertices_mri, Transf] = cs_convert(sMriMask,  'scs',  'mri', femHead.Vertices);
```

Then using this piece of code, I want to overlay the mesh on the mri/tissues :

```
% 
hFig = figure;
hAx         = axes('Parent',hFig);
fixedVolume = mri_segmented.anatomy;
centerFixed = size(fixedVolume)/2;
slice(hAx,double(fixedVolume),centerFixed(1),centerFixed(2),centerFixed(3));
shading(hAx,'interp');
set(hAx,'Xgrid','on','YGrid','on','ZGrid','on');
set(hFig,'Colormap',colormap('gray'));
hold on;
plotmesh(femHead.Vertices_voxel,  [femHead.Elements femHead.Tissue],'x>0'); hold on;
xlabel('X'); ylabel('Y');  zlabel('Z');
```

here is the result :

[image]

[image]

Do you have any idea why the mesh does not overlay the voxels? 

there is 90* rotation (exchange X by Y) to fit the MRI,

Did I miss any step in my code? 

This is important for the step that assigns the conductivity sensor from the DTI.

ftadel, 2020-04-20:
I'm sorry, this is impossible to guess like this: I don't know what are your files in input, and I don't know what your function bstMri2ftMri.m (I completely removed it from the code I'm editing at the moment).

Why don't you import the files in the Brainstorm database and use the Brainstorm display functions instead of rewriting everything yourself?
All these display functions you are writing are going to be lost after, because when integrating them into Brainstorm we'll have to retranslate them with the Brainstorm display system...

There are plenty of examples in the tutorial scripts (eg. tutorial_introduction.m) and the list of the display functions is available here:
https://neuroimage.usc.edu/brainstorm/Tutorials/Scripting#Reference:_Display_functions

tmedani, 2020-04-20:
Thank you, Francois, 

I think the problem is on the "slice" function. It seems that the orientation X Y is exchanged, I don't know why, but it's not important for now. 

I was using this in order to check the output from Brainsuite software and map it to brainstorm. 

I'm not sure that I can use the brainstorm functions since these data are not in bst format and not in SCS coordinates.  

The main problem was the mapping of the tensor from the brainsuite voxel space to the SCS space. 
With @ajoshiusc we were able to solve this.

The only way to check if the tensors are correct is by displaying them in the right coordinate system.

Here is a quick figure to check the tensors :

[image]

Now, it needs to be integrated into the computation. This will use the knw files and we need to find an optimal way to integrate this option on the duneuro panel.   

I'm working on that ...

Regarding the display of the tensors as an ellipse or main eigenvector,  
I have developed these functions for this purpose on Matlab. 
@ftadel do you think it's relevant to add them to brainstorm?

It needs some works, but I think it could be interesting. 

This may join what @martcous did on the fibers.

ftadel, 2020-04-21:
> Now, it needs to be integrated into the computation. This will use the knw files and we need to find an optimal way to integrate this option on the duneuro panel.

It might be better if we design this together before you start implementing anything. Otherwise you might lose a lot of time coding something that I might change completely after.
Could you describe exactly what the workflow is: 
- what are the files in input (before any processing in Brainstorm)
- what are the functions to read these files
- what type of data does it produce once read
- where were you thinking of saving them in the Brainstorm database
- how were you thinking of passing them to the FEM computation
Please share some example file if I need anything to work on it beyond the subjects I already have: **m46wap** (m46wap_dti.nii, m46wap_t1.nii, m46wap_t2.nii) and **A1001fullMRdata** (1001.bval, 1001.bvec, 1001_DTI.nii, 1001_T1.nii, 1001_T2.nii)

> Regarding the display of the tensors as an ellipse or main eigenvector, I have developed these functions for this purpose on Matlab.
> @ftadel do you think it's relevant to add them to brainstorm?

Yes, it would be good to have this in Brainstorm. 
Please share, together with the example files.

tmedani, 2020-04-21:
> It might be better if we design this together before you start implementing anything. Otherwise, you might lose a lot of time coding something that I might change completely after.

indeed, I will prepare and share with you a clean version of how I did this.
  

> Could you describe exactly what the workflow is:
> 
> * what are the files in the input (before any processing in Brainstorm)
> * what are the functions to read these files
> * what type of data does it produce once read
> * where were you thinking of saving them in the Brainstorm database
> * how were you thinking of passing them to the FEM computation
>   Please share some example file if I need anything to work on it beyond the subjects I already have: **m46wap** (m46wap_dti.nii, m46wap_t1.nii, m46wap_t2.nii) and **A1001fullMRdata** (1001.bval, 1001.bvec, 1001_DTI.nii, 1001_T1.nii, 1001_T2.nii)

this picture may summarize the whole process and data format:

[image]

> Yes, it would be good to have this in Brainstorm.
> Please share, together with the example files.

ok, I will share it as well, but it may need improvement. 

I will work on a clean version of the whole process and push it for discussion.

Thank you

tmedani, 2020-04-22:
Hey, 
@ftadel, I have just created a PR that add an item the menu of the anatomy-mri.
that calls the process to generate the FEM tensors.

I will push therelated  functions to the bst-duneuro for now.

The function "bst_generate_conductivity_tensor" is located on the bst-duneuro repo (need to pull the new release).
There are also the functions that call the Brainsuite process, conversion of the tensor from voxel to scs and the function to display tensors as an ellipse on the mesh.

All these functions are working as they are but still under development/improvement/cleaning.  

In order to call this function from brainstorm, right-click on the MRI and then "Generate FEM tensor"

The installation of the Brainsuite software http://forums.brainsuite.org/download/ is required.

You need to have the DWI data (nifty, bval and bvec). 
As it's implemented now, you need to have all these files in the same folder and have the same name (s01.nii, s01.bvec and s01.bval). 
You need just to select the nii file during the uiopen is called. 

There is no output to brainstorm database, you need to add a breakpoint on the function to see what is going on. 

The main output data needed from this process is either (or both) the :
- DTItensor on each element
- Conductivity tensor en each element 

You can also display the tensors (DTI or Conductivity) on voxel space or scs space.

------- More details ------
There are mainly two processes: 
1- Generate the DTI tensor from the DWI data (15-40 min)
2- Convert the DTI tensors to conductivity tensors (<10 min)

The second process needs the conductivity value of the WM (and GM ... we will discuss this?)
these values of conductivities are not available at this level, we get their values when we call the duneuro.

So maybe, we can do only the first process and integrate the second process when we call the Duneuro computation (if the anisotropy is available). 

Regarding process 1, we can keep it as an independent option and/or integrated it into the FEM mesh generation (if the DWI data are available). 

@ftadel what do you think?
  
Regarding the display process, 
do not ask to display all the tensors, it will take a lot of time and the results will not clear. 
here is an example 
[image]

it a nice picture, but it does not make sense to display million the ellipse. 

The function needs to be adapted, either to display specific tensors or specify axes as it's the case on Brainsuite. ... I'm working on it.

[image]

ftadel, 2020-04-24:
I will integrate it your code in a similar way to what I did with bst-duneuro: linearize everything, rewrite some of it.

As you suggested, I think it would be best to split the process in 2:
1) DWI=>DTI with BrainSuite
[image]  [image] 
[image]

2) DTI => Conductivity tensors: Integrated into the computation of the forward model, because indeed, it requires conductivity values that are set only when calling DUNEuro.

At the moment, I think there is a distribution problem regarding the integration with BrainSuite: The program "bdp.exe" requires the MCR R2015b to be installed. We would ask our users who are already executing Matlab to install an additional old MCR, and execute a second instance of the Matlab engine on the same computer. This seems a bit silly...
Couldn't we use directly a non-compiled Matlab version? (converted to p-files, if the .m files are not public)
@ajoshiusc?

tmedani, 2020-04-24:
> @tmedani I will integrate your code in a similar way to what I did with bst-duneuro: linearize everything, rewrite some of it, making everything smoother.

Yes, all your inputs and modifications are welcome. 

this GUI is awesome, thanks. 

> As you suggested, I think it would be best to split the process in 2:

yes, I think also this is the best option.  
it' still needs some work from my side and I'm trying to understand the math about the conversion from DTI to Conductivity (not easy to understand in some case).

Some basic notes regarding the comment on the interface :
- not only bdp is required but also the 'bse' and 'bfc' which are all part of the package brainsuite.
- selecting bval and bvec are not required if they have the same name as the DWI, however, they need to be in the same folder. 
 
>     I'll check how complicated it would be to add the support for 4D images in the database (scrolling through the volumes as if it were time), it would make the DTI storage much easier...

It would be nice but I don't know how useful for the users?

I have tried to do it, but it needs many changes in the brainstorm. I did it only for one image

also, some points : 
- The DWI images need to be registered to the T1, Brainsuite doest it during the computation. 
- If we load the DWI it needs to be registered to the MRI,

 
> **Problem**:
> At the moment, I think there is a distribution problem regarding the integration with BrainSuite: The program "bdp.exe" requires the MCR R2015b to be installed. We would ask our users who are already executing Matlab to install an additional old MCR and execute a second instance of the Matlab engine on the same computer. This seems a bit silly...
> Couldn't we use directly a non-compiled Matlab version? (converted to p-files, if the .m files are not public)
> @ajoshiusc?

I'm not sure but @ajoshiusc  is the well-positioned person to answer this question.

ftadel, 2020-04-27:
> it' still needs some work from my side and I'm trying to understand the math about the conversion from DTI to Conductivity (not easy to understand in some case).

So maybe I try to finish the process_dwi2dti I'm currently writing based on your code (which would compute the DTI files and save them in the database), and then I wait until you have something stable for the second part (DTI > Conductivity > FEM)

> not only bdp is required but also the 'bse' and 'bfc' which are all part of the package brainsuite.

I've seen that, but these other functions do not have this additional dependency on the MCR 2015b, so less of a problem for the distribution.

> selecting bval and bvec are not required if they have the same name as the DWI, however, they need to be in the same folder.

I made it possible. I already have an example folder where the DWI.nii and .bvec and .bval do not have the same name, and in BIDS we may not be able to rely on this naming logic anymore.
So I made it optional:
- In the interactive menu: Ask for the dwi.nii, then check for the existence of .bval and .bvec with the same name. If they exist, use them automatically, otherwise, ask for them to the user.
- In the process: If only the .nii is specified, then look of the .bval/.bvec (error if not found). If the .bval/.bvec are set in the input options of the process: use them without trying to do any automatic detection.

> >  I'll check how complicated it would be to add the support for 4D images in the database (scrolling through the volumes as if it were time), it would make the DTI storage much easier...
> It would be nice but I don't know how useful for the users?

Not useful to the user, but more compact in the database, less messy, with less risks of mixing things up. I wouldn't like to save 6 dependent volume files in the subject folder without any way to keep track of their dependencies... it would be very prone to manipulation errors. What would happen if you were running multiple times the conversion DWI>DTI? After reloading the database, how would you keep track of which DTI volumes go together?...

> I have tried to do it, but it needs many changes in the brainstorm. I did it only for one image

Indeed, there are many things to change, but I'll try to make it work. It could be useful for other purposes (eg. fMRI) to store and display time-varying volumes.
I'll think about this this week and try to propose an efficient storage solution.

> The DWI images need to be registered to the T1, Brainsuite doest it during the computation.
> If we load the DWI it needs to be registered to the MRI,

Do you mean that we need to register the DWI and T1 if we plan to overlay the DWI and the T1 on the same display? Or do we need to register them in any case before calling the functions that you uploaded on bst-duneuro?
Would SPM registration functions work as well?

tmedani, 2020-04-27:
> So maybe I try to finish the process_dwi2dti I'm currently writing based on your code (which would compute the DTI files and save them in the database), and then I wait until you have something [...]

This is nice indeed, at this level we will have the DTI tensors and it will be good to add a function to display/overlay them on the MRI.

After the generation of the DTI tensors on the mesh we can display can also display the new tenors on the mesh. This is the best way to check the results.

> I made it possible. I already have an example folder where the DWI.nii and .bvec and .bval do not have the same name, and in BIDS we may not be able to rely on this naming logic anymore.
> So I made [...]

Super then,

> Not useful to the user, but more compact in the database, less messy, with less risks of mixing things up. I wouldn't like to save 6 dependent volume files in the subject folder without any way to [...]

yes totally agree, and I would say also that it's useful for the developers' side. 

> Indeed, there are many things to change, but I'll try to make it work. It could be useful for other purposes (eg. fMRI) to store and display time-varying volumes.
> I'll think about this this week and [...]

super ... since it can be used for many purposes 

> > The DWI images need to be registered to the T1, Brainsuite doest it during the computation.
> > If we load the DWI it needs to be registered to the MRI,
> 
> Do you mean that we need to register the DWI and T1 if we plan to overlay the DWI and the T1 on the same display? 

Yes, you can try with 1001 subjects you will see that the two images (T1 and DWI) are not in the same space (I have used brainsuite overlay option).

But after loading one image from the DWI to brainstorm and call SPM, the two data are aligned. 

Or do we need to register them in any case before calling the functions that you uploaded on bst-duneuro?

as it's now, the registration to T1  is done by brainsuite, so I don't perform any registration.  

I don't know if there is any effects on BDP results when we use the raw DWI or the registered dwi with SPM.
Maybe it can affect tensors values?  since bvec is related to the orientation!
@ajoshiusc does it make sense?

> Would SPM registration functions work as well?

Yes, 
also,  it seems that you don't need to run SPM as many time as the number of the DWI images, 
but only one time on the first image and then apply the same transformation for the other images  
(I need to recheck, but I think it's the case ...)

ftadel, 2020-04-29:
I pushed modifications to handle 4D volumes in the database: https://github.com/brainstorm-tools/brainstorm3/commit/99bf0b5d9299269fa6fd1c0c2eb9ea51a9690889
Indeed, it was not trivial :-)

Now I'll resume my work on process_dwi2dti.
It would be good to have some feedback from the BrainSuite team, so I know what direction to take for calling BrainSuite DTI pipeline from Brainstorm.
@ajoshiusc?

ajoshiusc, 2020-04-29:
> **Problem**:
> Couldn't we use directly a non-compiled Matlab version? (converted to p-files, if the .m files are not public)
> @ajoshiusc?

1. The MCR for BrainSuite (2019b for the latest and 2015b for earlier versions) is required during installation. So if users have BrainSuite installed on their computers, they will have the MCR.

2. Even if the source code for BDP.exe is under GNU license, and if needed, can be incorporated into  BrainStorm, BDP assumes as input processed data from BrainSuite which still needs the MCR for labeling and registration steps.

3. If we want to integrate BDP without MCR, we will have to take pieces from the BrainSuite (C++ and Matlab) and BDP (Matlab) codes. My suggestion is that we assume BrainSuite is installed including the correct MCR.

ajoshiusc, 2020-04-29:
> as it's now, the registration to T1 is done by brainsuite, so I don't perform any registration.
>
> I don't know if there is any effects on BDP results when we use the raw DWI or the registered dwi [...]

I think using raw DWI as input to the BDP might be slightly better since each time we apply registration, we might get some interpolation artifacts. If there is no interpolation done during registration, and only the header is modified (storm, qform), in that case, pre-registration in SPM will have no effect. This is assuming the b-vec are appropriately reoriented during registration by SPM.

tmedani, 2020-04-30:
> I pushed modifications to handle 4D volumes in the database: [99bf0b5](https://github.com/brainstorm-tools/brainstorm3/commit/99bf0b5d9299269fa6fd1c0c2eb9ea51a9690889)
> Indeed, it was not trivial :-)

@ftadel 
Super & bravo

@ajoshiusc  thank you for your feedback. 

Therefore, we need to apply the same transformation for the vectors on the bvec file. 
how about the bval? 

It can be affected also in the case where the SPM transformation re-scale the volume! 
does it make sense to change also these values? @ajoshiusc 

 @ftadel I can't reproduce what you showed [here](https://github.com/brainstorm-tools/brainstorm3/issues/267#issuecomment-619069250) 
The process_dwi2dti is not on the master yet?

Regarding the tensor visualization within brainstorm, it's much better to display only the tenor on specific slices/plans
- overlay on the 3d orthogonal MRI viewer.
- cuttings plan for the mesh

It needs some work but I think it worth it ... I will work on it.

ajoshiusc, 2020-04-30:
> @ftadel
> Super & bravo
>
> @ajoshiusc thank you for your feedback.
>
> Therefore, we need to apply the same transformation for the vectors on the bvec file.
> how about the bval?
>
> It can be affected also in [...]

To apply a linear transformation to DWI, we need to do 2 things:
1. Apply the linear transformation to the data matrix.
2. Get only the rotation part (closest orthogonal matrix) from the linear transformation and apply it to bvec. I can share the code from BDP that does this part (reorientDWI_BDP.m).

ftadel, 2020-04-30:
@ajoshiusc Thanks for the detailed answers.

> My suggestion is that we assume BrainSuite is installed including the correct MCR.

Ok, I'll consider that BrainSuite is something completely independent and runs autonomously, not trying to integrate it further with Brainstorm. I just wanted to make sure there was no easier way to approach this before starting the developments.

> I think using raw DWI as input to the BDP might be slightly better since each time we apply registration, we might get some interpolation artifacts.

It's OK this way. I was not planning to allow the importing of DWI/bvec/bval to the brainstorm database anyway, which would be the first step before running the SPM registration with the functions we currently have. 
So the process "DWI2DTI" in Brainstorm will ask for the three files DWI/bvec/bval, call a sequence of BrainSuite executables, and save the DTI results in the Brainstorm database, as an "MRI" entry with multiple volumes in the anatomy folder.

@tmedani 
> The process_dwi2dti is not on the master yet?

Not yet, sorry. I had to go far away from my path to add the support for the multiple-volumes MRI files (necessary for storing the DTI output). I'll work on this in the next few days. I will push this maybe early next week.

> Regarding the tensor visualization within brainstorm, it's much better to display only the tenor on specific slices/plans

Can you give me more details on how works the DTI file (how many volumes, what is the meaning of each of them?) and how you are planning to use these values in your display functions?

tmedani, 2020-04-30:
Thank you, @ajoshiusc, for the codes.

> @ajoshiusc Thanks for the detailed answers.
>
>
> Ok, I'll consider that BrainSuite is something completely independent and runs autonomously, not trying to integrate it further with Brainstorm. I just [...]

All this sounds good, the process will be based on Brainsuite and we will not introduce any bias. 

> So the process "DWI2DTI" in Brainstorm will ask for the three files DWI/bvec/bval, call a sequence of BrainSuite executables, and save the DTI results in the Brainstorm database, as an "MRI" entry with multiple volumes in the anatomy folder.

Good also, 
from my reading, it seems that in some cases instead to have bval and bvec there is a file that combines these two files and it called bmat. 
We can handles that later, but just we have to keep it in mind.

> > The process_dwi2dti is not on the master yet?
> 
> Not yet, sorry. I had to go far away from my path to add the support for the multiple-volumes MRI files (necessary for storing the DTI output). I'll work on this in the next few days. I will push this maybe early next week.

There is no urgency, I have another work to complete to come to this last version.

> Can you give me more details on how works the DTI file (how many volumes, what is the meaning of each of them?) and how you are planning to use these values in your display functions?

YES... I will summarize what I finally understood :)

So, the input for the brainsuite is the T1, DWI, bval and bvec (or bmat). 
The size of the T1 is NxMxK.

Step 1:  the bse or brain surface extractor, where only the brain is kept and the other removed.
Step 2 : bfc or the bias field correction, where the image corrected from the non-uniformity 
Step3 : bdp or brainsuite diffusion pipeline, where either the tenosrs or the OFD are estimated. 
In our case, we use only tensors. 

Each tensor is computed on each voxel, but its value is zeros on the most voxels and not zeros on the brain area (mainly the white matter). 

On each voxel 6 values are computed, V1, V2, V3, L1, L2 and L3. 
These values are called the eigenvectors (V1, V2, V3) and eigenvalues (L1, L2 and L3).
These three vectors are defined on each voxel and they are orthonormal.
Then each vector is scaled by the associated eaigen value.  

In the case of the isotropic model, L1 = L2 = L3,
in the case of an anisotropic model, L1 >> L2 >L3, 
L1 is always the main direction.

the size of the results: 
size(Li) =  size(T1)
size(Vi) =  size(T1)x3 or NxMxKx3

To display we can use an ellipse, its center will be on the center of the voxel (or mesh element) and the three axes will be the three vectors scaled by the values Li. 

In the case of the isotropic model, the result will be a sphere, and in the case of anisotropic, the results will be an ellipse. 

We can also display the tensor as the main vector (V1*L1) and use the standardized color code:  Red  for Left<->Right orientation
Green  for Posterior<->Anterior orientation
Blue  for inferior <->Superior orientation

I have already pieces of codes that can display these tensors and overlay on the mesh (https://github.com/brainstorm-tools/brainstorm3/issues/267#issuecomment-617509448)
  
On the function that I wrote, bst_brainsuite_generate_dti_tensor, 
these values are stored in the DTI variable where
DTI{1:6} = {V1,V2,V3,L1,L2,L3}
I have used the fieldtrip format for the simple reason to adapt it to a previous SimBio code developed by Carsten's team.  

The next steps are to convert these DTI tensors to conductivity tensors, map them to the FEM mesh, and then use them with the FEM computation (via the knw files). 

Thanks

ftadel, 2020-05-05:
@ajoshiusc bdp.exe do not check for the existence of `mclmcrrt9_0.dll` correctly.
With a typical Matlab 2015b installation (not the MCR), this library is available in `C:\Program Files\MATLAB\R2015b\runtime\win64`, and it is sufficient to run the executable.
Users running Matlab 2015b should not have to install the MCR on top of it.

In Brainstorm, I added a .bat to check for the existence of either MCR 2015b or Matlab 2015b:
https://github.com/brainstorm-tools/brainstorm3/blob/master/bin/R2015b/brainstorm3.bat

ftadel, 2020-05-05:
@tmedani @ajoshiusc 
The code for calling BDP uses the function eig2nifti. 
I found it here: https://neuroimage.usc.edu/neuro/Resources/eig2nifti
Is it the version I'm supposed to use?

Should I copy all these .p files to brainstorm3/toolbox/external/eig2nifti?

There is a lot of overlap with the code already available in Braintorm (eg. reading and writing .nii).
Couldn't we get a .m version we could integrate better with Brainstorm?

tmedani, 2020-05-05:
Hey @ftadel 

Yes, I'm using this function to convert the bdp eig files to nii files, this function should be in the bst-duneuro already ?.

> Should I copy all these .p files to brainstorm3/toolbox/external/eig2nifti?

No, only the function eig2nifti.p is required.

> Couldn't we get a .m version we could integrate better with Brainstorm?

I think we can have the Matlab function from @ajoshiusc  ?

@ftadel regarding the next step,
I have finished the process of conversion from DTI to conductivity (I have implemented most of the available methods).

I need to know how/where is the output from the process_dwi2dti in order to link it to the next steps.

I will push the code to the bst-duneuro/inProgress folder.

I have also some pieces of code that can display the tensors on a mesh slice, here are some results.

[image]
[image]

[image]

juangpc, 2020-05-05:
this looks soo good.

ajoshiusc, 2020-05-05:
> > Couldn't we get a .m version we could integrate better with Brainstorm?
> 
> I think we can have the Matlab function from @ajoshiusc ?
> 
Here is the  [Matlab code for eig2nifti](https://drive.google.com/file/d/1s6uAjL9dFRsKaL_aGpjXhawRG6ldnobN/view?usp=sharing). All the dependencies are included.

ajoshiusc, 2020-05-05:
> @ajoshiusc bdp.exe do not check for the existence of `mclmcrrt9_0.dll` correctly.
> With a typical Matlab 2015b installation (not the MCR), this library is available in `C:\Program [...]
Can you tell me more about this? I believe bdp was simply compiled as usual, at least for windows, and should be able to use either MCR or Matlab whichever is first on the PATH.

ftadel, 2020-05-06:
@ajoshiusc 
> Can you please tell me more about this issue? I believe bdp was simply compiled as usual, at least for windows, and should be able to use either MCR or Matlab whichever is first on the PATH.

Oh I see, it is the EXE produced directly by the Matlab compiler... I'm not familiar with this, as we package Brainstorm as a .jar and have to find Matlab ourselves before running it. On Windows, I use the registry for that, I assumed it was what a Matlab compiled .exe would do as well, but no.
Installing Matlab doesn't automatically add the bin folder to the system path, so the .dll doesn't get detected, while the MCR install seems to automatically add the bin folder to the path.
A .bat file detecting the Matlab installation folders and setting the system path accordingly before calling bdp.exe would help, but definitely not a priority, as I guess most people would use the MCR anyway (if the already have Matlab, there is little chance that this is Matlab 2015b).

> Here is the Matlab code for eig2nifti. All the dependencies are included.

Thanks! 
Is it OK if I redistribute these files in brainstorm3/export/eig2nifti/ ? 
Do you want to add a copyright notice?

@tmedani 
> I need to know how/where is the output from the process_dwi2dti in order to link it to the next steps.

Sorry, I'm taking longer than I expected, I'll need a few more days to digest all this. 
I'm working on the integration with BrainSuite at the moment.
I'll try to figure out what happens in the new files you posted and finish all the integration in Brainstorm. 
Is there anything you'd need urgently to keep working?

> this looks soo good.

Beautiful indeed!

ftadel, 2020-05-06:
> > Here is the Matlab code for eig2nifti. All the dependencies are included.
> Is it OK if I redistribute these files in brainstorm3/export/eig2nifti/ ?
> Do you want to add a copyright notice?

Actually, I can see there is no need of saving the .nii files on the hard drive before reading them again in Brainstorm. I'll edit these functions to save directly the volumes into the Brainstorm database. It will probably result in a new function brainstorm3/toolbox/io/in_mri_eig.m

tmedani, 2020-05-06:
> Sorry, I'm taking longer than I expected, I'll need a few more days to digest all this.
> I'm working on the integration with BrainSuite at the moment.
> I'll try to figure out what happens in the new [...]

Super, there is no urgently at this time and I can wait for beginning of next week.

I will try to improve the visualization process (as it 's now it's slow). 

> > this looks soo good.
> 
> Beautiful indeed!

Thanks, @sbaillet & @ftadel  
it will better within brainstorm :) 

> Actually, I can see there is no need of saving the .nii files on the hard drive before reading them again in Brainstorm. I'll edit these functions to save directly the volumes into the Brainstorm database. It will probably result in a new function brainstorm3/toolbox/io/in_mri_eig.m

This will be great since it can be faster and avoid the writing and then loading them as it is now. 

A+

ajoshiusc, 2020-05-06:
> Actually, I can see there is no need of saving the .nii files on the hard drive before reading them again in Brainstorm. I'll edit these functions to save directly the volumes into the Brainstorm [...]

@ftadel sure, you can use it within BrainStorm or modify as necessary. The copyright notice is at the beginning of the code.

ftadel, 2020-05-07:
I'm having trouble going further because BDP crashes with the only subject I have with DWI.nii/bva/bvec (1001).
I don't have any example .eig file to test my code with, I'm not sure what to do.

Here are the input files:
https://www.dropbox.com/s/3xdl8tj8r9mjz9o/bdp_bug.zip?dl=0

And the full console output:

```
BST> System call: bdp "C:\Users\franc\.brainstorm\tmp\brainsuite\output_mri.bfc.nii.gz" --tensor --nii "C:\Work\RawData\Test\fem\A1001\1001_DTI.nii" --t1-mask "C:\Users\franc\.brainstorm\tmp\brainsuite\bse_smooth_brain.mask.nii.gz" -g "C:\Work\RawData\Test\fem\A1001\1001.bvec" -b "C:\Work\RawData\Test\fem\A1001\1001.bval"

BDP Version: 19a (build #0074), released 2019-02-14 
 
================================================================================ 
                         Setting up dataset and inputs 
================================================================================ 
Reading input flags... 
 
Checking input files... 
 
Processing data with fileprefix: 
	C:\Users\franc\.brainstorm\tmp\brainsuite\output_mri 
 
================================================================================ 
                   Co-registration and Distortion Correction 
================================================================================ 
Reading the input parameters for co-registration... 
 
Total usable memory found:   7.88GB 
 
Checking orientation information...Done 
Extracting 0-diffusion (b=0) image from input DWIs... 
 
DWI mask is not defined in input flags. BDP will try to estimate a (pseudo) mask 
from 0-diffusion (b=0) image. Automatic mask estimation may not be accurate in 
some sitations and can affect overall quality of co-registration. In case 
co-registration is not accurate, you can define a DWI mask by using flag 
--dwi-mask <mask_filename>. The mask can be generated and hand edited in 
BrainSuite interface. This mask would be used only for registration purposes 
(and not for statistics computation). 
 
------------------------------------------------------------------------ 
          Access violation detected at Thu May 07 12:55:34 2020 
------------------------------------------------------------------------ 
 
Configuration: 
  Crash Decoding      : Disabled 
  Crash Mode          : continue (default) 
  Current Graphics Driver: Unknown hardware  
  Default Encoding    : windows-1252 
  Graphics card 1     : Intel Corporation ( 0x8086 ) Intel(R) HD Graphics 530 Version 26.20.100.7263 
  Graphics card 2     : NVIDIA ( 0x10de ) NVIDIA GeForce GTX 960M Version 26.21.14.4145 
  Host Name           : DESKTOP-SQA66ET 
  MATLAB Architecture : win64 
  MATLAB Root         : C:\Program Files\MATLAB\MATLAB Runtime\v90 
  MATLAB Version      : 8.6.0.267246 (R2015b) 
  OpenGL              : hardware 
  Operating System    : Microsoft Windowsá10 Professionnel 
  Processor ID        : x86 Family 6 Model 94 Stepping 3, GenuineIntel 
  Virtual Machine     : Java 1.7.0_60-b19 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode 
  Window System       : Version 10.0 (Build 18363) 
 
Fault Count: 1 
 
 
Abnormal termination: 
Access violation 
 
Register State (from fault): 
  RAX = 00000000b31d0d40  RBX = 00000000b31d0fe0 
[... 72 lines cut ...]
[ 58] 0x00007ff8d34ece51                      C:\WINDOWS\SYSTEM32\ntdll.dll+00446033 RtlUserThreadStart+00000033 
 
 
This error was detected while a MEX-file was running. If the MEX-file 
is not an official MathWorks function, please examine its source code 
for errors. Please consult the External Interfaces Guide for information 
on debugging MEX-files. 
 
If this problem is reproducible, please submit a Service Request via: 
    http://www.mathworks.com/support/contact_us/ 
 
A technical support engineer might contact you with further information. 
 
Thank you for your help.** This crash report has been saved to disk as C:\Users\franc\AppData\Local\Temp\matlab_crash_dump.20720-1 ** 
 
 
 
MATLAB is exiting because of fatal error 

status =

  -1.0737e+09
```

Am I doing anything wrong?

tmedani, 2020-05-07:
Hi,

I had a similar error message before due to the confusion between T1 and T2 
but after checking the mask it seems that the error is something else. 

I'm trying to run the process using your data on my computer ... and it works 
here is the input: 
```
OutPutFolder = pwd  %(bdp_bug)
bvalFilename = '1001.bval'
bvecFilename = '1001.bvec'
dtiFilename = '1001_DTI.nii'

[status_bdp,cmdout_bdp]  = system(['bdp ' fullfile(OutPutFolder,'output_mri.bfc.nii.gz') ' --tensor --nii ' dtiFilename...
'  --t1-mask '  fullfile(OutPutFolder,'bse_smooth_brain.mask.nii.gz')...
'  -g ' bvecFilename ' -b ' bvalFilename]);
```
here is the output: 

> cmdout_bdp =
> 
>     '
>      BDP Version: 19b (build #0077), released 2019-12-31
>      
>      ================================================================================
>                               Setting up dataset and inputs
>      ================================================================================
>      Reading input flags...
>      
>      Checking input files...
>      
>      Processing data with fileprefix:
>      	C:\Users\33649\Dropbox\bdp_bug\brainsuite\output_mri
>      
>      ================================================================================
>                         Co-registration and Distortion Correction
>      ================================================================================
>      Reading the input parameters for co-registration...
>      
>      Total usable memory found:  36.57GB
>      
>      Checking orientation information...Done
>      Extracting 0-diffusion (b=0) image from input DWIs...
>      
>      DWI mask is not defined in input flags. BDP will try to estimate a (pseudo) mask
>      from 0-diffusion (b=0) image. Automatic mask estimation may not be accurate in
>      some sitations and can affect overall quality of co-registration. In case
>      co-registration is not accurate, you can define a DWI mask by using flag
>      --dwi-mask <mask_filename>. The mask can be generated and hand edited in
>      BrainSuite interface. This mask would be used only for registration purposes
>      (and not for statistics computation).
>      Saved (pseudo) masks: C:\\Users\\33649\\Dropbox\\bdp_bug\\brainsuite\\output_mri.dwi.RAS.mask.nii.gz
>      
>      
>      Starting Registration based distortion Correction...
>      Loading data...
>      
>      Starting rigid registration of input images...
>      Reading input data...
>      Setting/generating masks...
>      Matching centroids (approx. align)...
>      Matching resolution of the images...
>      Normalizing intensity of images... 
>      Performing affine registration with 6 degrees of freedom...
>      Search-based initialization... 
>      Scale 1 of 6 running...
>      Scale 2 of 6 running...
>      Scale 3 of 6 running...
>      Scale 4 of 6 running...
>      Scale 5 of 6 running...
[... 75 lines cut ...]
>      registration in BrainSuite.
>      
>      BDP saved following BST files:
>      	C:\Users\33649\Dropbox\bdp_bug\brainsuite\output_mri.tensor.T1_coord.bst
>      
>      
>      BDP finished all processing on 07-May-2020 04:42:48 for data with fileprefix:
>      	C:\Users\33649\Dropbox\bdp_bug\brainsuite\output_mri 
>      
>      A summary of the processing is saved in file:
>      	C:\Users\33649\Dropbox\bdp_bug\brainsuite\output_mri.BDPSummary.txt 
>      
>      '

I think your version is not working, could you update it and check?

your version : BDP Version: 19a (build 0074), released 2019-02-14 
my version : BDP Version: 19b (build 0077), released 2019-12-31

in meantime here is the eig file from bdp   : 
https://www.dropbox.com/s/463pbye2yvjvnwc/output_mri.dwi.RAS.correct.T1_coord.eig.nii.gz?dl=0

Let me know how it's going...

A+

tmedani, 2020-05-07:
by the way, 

@ajoshiusc  why bdp display this message?

```
DWI mask is not defined in input flags. BDP will try to estimate a (pseudo) mask
 from 0-diffusion (b=0) image. Automatic mask estimation may not be accurate in
 some sitations and can affect overall quality of co-registration. In case
 co-registration is not accurate, you can define a DWI mask by using flag
 --dwi-mask <mask_filename>. The mask can be generated and hand edited in
 BrainSuite interface. This mask would be used only for registration purposes
 (and not for statistics computation).
```

we have explicitly added the mask on the command 

[status_bdp,cmdout_bdp]  = system(['bdp ' fullfile(OutPutFolder,'output_mri.bfc.nii.gz') ' --tensor --nii ' dtiFilename...
**'  --t1-mask '  fullfile(OutPutFolder,'bse_smooth_brain.mask.nii.gz')**...
'  -g ' bvecFilename ' -b ' bvalFilename]);

Is there something missing in the arguments?

Thank you

ftadel, 2020-05-07:
> your version : BDP Version: 19a (build 0074), released 2019-02-14
> my version : BDP Version: 19b (build 0077), released 2019-12-31

Thanks for pointing this out. I'm installing the new version and I'll check.

I thought my version was up-to-date because the homepage of the BrainSuite website (http://brainsuite.org/) says "BrainSuite19a now available - The latest version of BrainSuite (v.19a) is available for download"
@ajoshiusc Maybe you should fix this?

ajoshiusc, 2020-05-08:
> by the way,
> 
> @ajoshiusc why bdp displays this message?
> 
> ```
> DWI mask is not defined in input flags. BDP will try to estimate a (pseudo) mask
>  from 0-diffusion (b=0) image. Automatic mask estimation may not be accurate in
>  some sitations and can affect overall quality of co-registration. In case
>  co-registration is not accurate, you can define a DWI mask by using flag
>  --dwi-mask <mask_filename>. The mask can be generated and hand edited in
>  BrainSuite interface. This mask would be used only for registration purposes
>  (and not for statistics computation).
> ```
> 
> we have explicitly added the mask on the command
> 
> [status_bdp,cmdout_bdp] = system(['bdp ' fullfile(OutPutFolder,'output_mri.bfc.nii.gz') ' --tensor --nii ' dtiFilename...
> **' --t1-mask ' fullfile(OutPutFolder,'bse_smooth_brain.mask.nii.gz')**...
> ' -g ' bvecFilename ' -b ' bvalFilename]);

The message is about mask for the diffusion images, the mask that is specified on the command line is for T1 images. 
The flag --dwi-mask can be used to define region of interest for diffusion modelling, without coregistration to T1 images. It is optional and can be useful if we only have diffusion data and are interested only in the diffusion modelling in a specific ROI. The mask for diffusion images can be generated manually using the mask tool in BrainSuite, but it is not necessary. BDP usually can generate the diffusion mask.

ftadel, 2020-05-08:
Here it is:
https://github.com/brainstorm-tools/brainstorm3/commit/56b311b9a0d3efd2b7c07b6197e6ca91c6ee11e2

Import the T1, then right-click on the subject folder > Convert DWI to DTI, select DWI+bvec+bval:
[image]

The BrainSuite installation folder must be informed in the Brainstorm preferences:
[image]

It produces in output a file "DTI-EIG" with the 12 volumes in it. If you display it on top of the T1 MRI, you can scroll between the volumes as if it were time (see the time panel in the top-right corner of the Brainstorm window)
[image]

> I think your version is not working, could you update it and check?

It works well with 2019b version indeed!

ftadel, 2020-05-08:
> @ftadel sure, you can use it within BrainStorm or modify as necessary. The copyright notice is at the beginning of the code.

I ended up not using eig2nifti at all, I just added a few lines of code to the Nifti reader in Brainstorm: 
https://github.com/brainstorm-tools/brainstorm3/commit/56b311b9a0d3efd2b7c07b6197e6ca91c6ee11e2#diff-7bf44945605e5164b961ba6b6f74a919R364

ftadel, 2020-05-08:
@tmedani 
I don't manage to orient myself in all the functions bst-duneuro/inProgress. It seems that many functions are old versions, or redundant, and some functions are missing (eg. bst_display_tensor_as_ellipse)

If I understood well, I have these things left to do:
1) Display the "DTI-EIG" volume:
   a) in the volume (similar to sources displayed on the MRI viewer => colored voxels) and 
   b) in 3D (elipses for 3 orthogonal slices, similarly to what is done when displaying the T1 MRI in 3D)
2) Integrate DTI-EIG in the duneuro computation:
   a) Add it to the inputs of the duneuro process
   b) In the duneuro code, integrate your code as 

For each of these tasks, can you give me the pointers to your functions, the input (in terms of files saved in the Brainstorm database), and the expected output.

### 1.a) Display DTI-EIG in the MRI viewer:

> > We can also display the tensor as the main vector (V1*L1) and use the standardized color code: > > Red for Left<->Right orientation
> > Green for Posterior<->Anterior orientation
> > Blue for inferior <->Superior orientation

So we use all the 12 volumes to create one RGB volume that can be displayed?
What's the arithmetic to compute the red/green/blue layers, based on the 12 volumes (V1, L1x, L1y, L1z, V2, L2x, L2y, L2z, V3, L3x, L1y, L3z)
Do you have code example on how to create one DTI slice in RGB?

### 1.b) Display DTI-EIG in 3D 

Which function should I use as a reference for trying to work something out?

### 2.b)  Integrate DTI-EIG in the duneuro computation

Which function should I use as a reference?
Where is it supposed to go in the existing Brainstorm code?

Thanks!

tmedani, 2020-05-08:
> @tmedani
> The message is about mask for the diffusion images, the mask that is specified on the command line is for T1 images.
> The flag --dwi-mask can be used to define region of interest for diffusion modelling, without coregistration to T1 images. It is optional and can be useful if we only have diffusion data and are interested only in the diffusion modelling in a specific ROI. The mask for diffusion images can be generated manually using the mask tool in BrainSuite, but it is not necessary. BDP usually can generate the diffusion mask. If the registration of T1 and DWI fails, we can use manual mask,

Super thank you @ajoshiusc, therefore we will keep it as it is now. 

 
@ftadel 
> Here it is:
> [56b311b](https://github.com/brainstorm-tools/brainstorm3/commit/56b311b9a0d3efd2b7c07b6197e6ca91c6ee11e2)

This is just perfect, I can now test the next steps. 

> I don't manage to orient myself in all the functions bst-duneuro/inProgress. It seems that many functions are old versions, or redundant, and some functions are missing (eg. bst_display_tensor_as_ellipse)

I'm going to clean this folder and keep only the useful function.

I will summarize and answer all your request in my next comment.

Thanks

tmedani, 2020-05-15:
Hi Francois, 

here are my answers : 

> @tmedani
> I don't manage to orient myself in all the functions bst-duneuro/inProgress. It seems that many functions are old versions, or redundant, and some functions are missing (eg. bst_display_tensor_as_ellipse)

I have updated the functions and all you need are on the 'inProgress/cleaned' folder
I'm still keeping all the other since I need to do more testing and checking. 

I have explained in this post how you can call these functions from brainstorm 
https://github.com/brainstorm-tools/brainstorm3/pull/282#issuecomment-627745837

You can try

> 
> If I understood well, I have these things left to do:
> 
> 1. Display the "DTI-EIG" volume:
>    a) in the volume (similar to sources displayed on the MRI viewer => colored voxels) and
>    b) in 3D (elipses for 3 orthogonal slices, similarly to what is done when displaying the T1 MRI in 3D)

You got it, 
there are different ways to display the DTI on the MRI. 
We can measure some values like the FA or MeanDiffusion and then color the voxels according to these values. But I'm not sure if it's relevant, all of these features are integrated with brainsuite. 
We can add this but it's not a high priority. 

The second option is to display the tensors as mainEigen vector or ellipse on the mesh, this is more important I think.
You can check an example of this within https://github.com/brainstorm-tools/brainstorm3/pull/282#issuecomment-627745837

Here we ask the user to display either vector or ellipse 
https://github.com/brainstorm-tools/bst-duneuro/blob/6af9def390e415545d4535b893a9267b422cd926/inProgress/cleaned/bst_view_tensor.m#L92

The code is not optimized as it is on this version. 

> 2. Integrate DTI-EIG in the duneuro computation:
>    a) Add it to the inputs of the duneuro process

Yes, when we have the tensors somewhere, we need to add option to say that we are using anisotropy, then the knw files will be used instead of the msh/cond files. 

>    b) In the duneuro code, integrate your code as
> 

> For each of these tasks, can you give me the pointers to your functions, the input (in terms of files saved in the Brainstorm database), and the expected output.

-for the computation of the conductivity tensors 
input : DTI-EIG, sMRI (or mask) the Mesh and the conductivity values.
output : conductivity tensors (3x3 matrix) or 3x3 matrix of vector and 1x3 eigenvalue (for each mesh element)

-for the display of the conductivity  tensor : 

input : mesh + the computed eigen values and vectors. 
output : tensors as ellipse or vectors on the mesh.

-for the FEM computation 
The Mesh and the tensors in order to write the knw files and then the associated mini file.

> ### 1.a) Display DTI-EIG in the MRI viewer:
>
> So we use all the 12 volumes to create one RGB volume that can be displayed?

somehow yes, these 12 values are the eigenvectors (3x3 components  for 3 vector) and the three eigenvalues (length of these three vectors) 

In the mesh side, each ellipse is displayed with a color corresponding to its main direction
 
here is the way how I did it : 
we know the main directions are: RAS  with the corresponding color RGB
I use the main vector components to specify the amount of the RGB to use

 h.FaceColor = abs([v(1,1) v(2,1)  v(3,1)])
where v(1,i) is the vector of the main direction, then the color of the ellipse will follow the component of the main vector.  
check here : 
https://github.com/brainstorm-tools/bst-duneuro/blob/6af9def390e415545d4535b893a9267b422cd926/inProgress/cleaned/bst_display_fem_tensors.m#L62

> Do you have code example on how to create one DTI slice in RGB?

Yes, I worked on this example, check here :  
https://github.com/brainstorm-tools/bst-duneuro/blob/6af9def390e415545d4535b893a9267b422cd926/inProgress/cleaned/bst_view_tensor.m#L52

you can run it from brainstorm and stop by to check. 

> ### 2.b) Integrate DTI-EIG in the duneuro computation
> Which function should I use as a reference?
> Where is it supposed to go in the existing Brainstorm code?

I will work on it and check where is the best place.

tmedani, 2020-05-15:
Hi, 

just if we came back later to this thread, or if someone is following it,  
there is part of the discussion [here ](https://github.com/brainstorm-tools/brainstorm3/pull/282)

ftadel, 2020-06-09:
I finished a first of the FEM tensor display:
https://github.com/brainstorm-tools/brainstorm3/commit/41e087faec7feb566fa38b3af13566d4ff714851

When right-clicking on the FEM, if the Tensors field is available, it shows menus to display them. Two options: "ellipse" or "arrows".
[image]

Then you get asked which tissues you want to show (possibility to select multiple):
[image]

It displays a slice of tensors on top of the T1 MRI slices in 3D (the thickness of the slice is computed based on the average elements size, and could be adjusted if needed):
[image]

[image]

You can move the displayed slice with the Resect sliders in the Surface tab, or with a right-click+move on the 3D figure.

[image]

Please let me know what needs to be changed.
Close the issue when you're happy with the result.

tmedani, 2020-06-09:
Hi @ftadel 

That's great and much faster than the previous version. Thanks for that. 

Is it possible to add the FEM mesh?

I got this error when I want to add the FEM mesh 

```

Index exceeds the number of array elements (1).

Error in panel_surface>UpdateSurfaceProperties (line 1024)
    isAnatomy = strcmpi(TessInfo(iSurface).Name, 'Anatomy');

Error in panel_surface>UpdatePanel (line 871)
            UpdateSurfaceProperties();

Error in panel_surface>AddSurface (line 1282)
    UpdatePanel();

Error in panel_surface>ButtonAddSurfaceCallback (line 786)
    iTess = AddSurface(hFig, SurfaceFile);

Error in panel_surface>@(h,ev)ButtonAddSurfaceCallback (line 73)
        jButtonSurfAdd = gui_component('ToolbarButton', jToolbar, [], [],
        {IconLoader.ICON_SURFACE_ADD, TB_DIM},    'Add a surface',
        @(h,ev)ButtonAddSurfaceCallback);

```

I have tried to overlay the mesh on the 3D MRI display without the tensors and the same errors appear. 

However, we were able to do it before
https://github.com/brainstorm-tools/brainstorm3/issues/267#issuecomment-616227844

Something is changed between these two versions. 

If we can add this option we will have a full package and then close this issue 👍 

Thanks
Takfarinas

tmedani, 2020-06-11:
> > Is it possible to add the FEM mesh?
> 
> I made it possible to display the tensors either on the MRI or on the FEM mesh, which is not trivial at all, as they are not resected in the same coordinate system.

This is awesome!
 
Having all these options within brainstorm is unique ... I'm not sure if there is other tools that can offer all of this easily with a super interactive GUI.

tmedani, 2020-06-11:
> > is it possible to add an option to control the size of the ellipse?
> 
> No, I think they should be normalized from the display code itself.
> 
> > Maybe we can set a maximum volume of the ellipse where the display is good and then normalize the users' values by this normalized value?
> 
> Let me know which measure you'd like to use for the normalization.
> If you want it to be relative to the average size of the elements in the mesh, I used the heuristic below to estimate what is the average distance between the element center and the vertices of the mesh. I use this measure as the thickness of the slice of tensors to display (the larger the elements, the thicker the slice)
> 
> https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/gui/view_fem_tensors.m#L166
> 
> `TensorDisplay.tol = 0.5 .* sqrt(mean(sum(bst_bsxfun(@minus, FemMat.Vertices(FemMat.Elements(iElemTissue(1:10:end),1),:), TensorDisplay.ElemCenter(1:10:end,:)) .^ 2, 2)));`
> 
> Does it make sense for the ellipses size?

hum... this is not trivial indeed. 

the size of the ellipse is related to its position (to the tissue type) and not to the size of the element.
The size of the element is related to the density of the mesh. 

In order to keep a meaningful display of the ellipse, the same factor should be applied to all of them. 
The volume of the ellipse reflects the value of the conductivity at this position, and the shape the orientation of the conductivity. 

The problem is when the user wants to display the tensors of more than one tissue. 
When there is a big ratio between the conductivity values ( for example CSF/WM > 10), 
a problem on the display will appear. 

The idea is to have something like this 

[image]

from this paper https://pubmed.ncbi.nlm.nih.gov/16364662/

However, I don't have an idea how to do it right now :(

ftadel, 2020-06-11:
Let me know when you want to change anything to the display.
For the moment, I consider that I don't have anything urgent to code for this topic.

tmedani, 2020-06-11:
Yes, 
most of the expected functions are now available. 

Further discussion is here: https://github.com/brainstorm-tools/brainstorm3/pull/282

Merci :)

closed 2020-06-11 by tmedani