brainstorm-tools/brainstorm3 #284: Support reading and writing JNIfTI files - a JSON wrapper of nifti-1/nifti-2

fangq, 2020-04-24:
First, some background:

Last year, I started a new project called OpenJData (http://openjdata.org, https://openjdata.org/wiki/), with a goal to create a simple and extensible data-annotation standard for general scientific data storage. See some of our [rationales](http://openjdata.org/wiki/index.cgi?JData/Basics) and [examples](https://openjdata.org/wiki/index.cgi?JData/Examples/Basic) from the wiki site. More data samples can be found in [this folder](https://github.com/fangq/jnifti/tree/master/samples).

The goal of this standard is to use the widely supported JSON/UBJSON as the underlying format, and create a semantic layer to encode basic data structures ([JData spec](https://github.com/fangq/jdata)), neuroimaging scans ([JNIfTI spec](https://github.com/fangq/jnifti), [JSNIRF spec](https://github.com/fangq/jsnirf) for fNIRS), and other domain-specific data, such as [JMesh](https://github.com/fangq/jmesh) for shapes/geometries, JDICOM/JGifTI etc (to be developed).

Here, I want to bring to your attention **the [JNIfTI spec](https://github.com/fangq/jnifti)** because NIfTI-1/2 data have a large presence in neuroimaging software. As you may know, NIfTI files have several limitations, see my post in this thread https://github.com/bids-standard/bids-specification/issues/197#issuecomment-548883677

JNIfTI is basically a JSON/UBJSON wrapper over the NIfTI-1/2 data, with a header that is 100% compatible with nifti (thus two-way loss-less conversion - `nii2jnii/jnii2nii`). It directly addresses most of the limitations of NIfTI, making it easy for further extension, supporting internal compression (instead of `.nii.gz`), making it human readable and easy to integrate with metadata.

To read/write JNIfTI files (.jnii - JSON based JNIfTI files, or .bnii - UBJSON based JNIfTI files), you may use a lightweight toolbox, the JNIfTI toolbox:

https://github.com/fangq/jnifti/tree/master/lib/matlab

which by itself is also a nifti-1/2/Analyze7.5 file reader/writer. 

JNIfTI toolbox is **already included** in the Iso2mesh toolbox. Since BrainStorm has already supported Iso2mesh, you really **don't need to do anything else, but to associate the '.jnii' and '.bnii'** file to the `loadjnifti/savejnifti` functions. 

Again, similar to [my comment here](https://github.com/fangq/easyh5/issues/7#issuecomment-619050662), ZMat is optional if your user mostly uses MATLAB in the GUI mode.

Let me know what you think.

ftadel, 2020-04-27:
Thank you for this presentation of OpenJData / JNIfTI. 
I added it to our (infinite) to-do list: https://neuroimage.usc.edu/brainstorm/Next#Input_.2F_output

Since we have much more implementation requests than available development resources, we tend to give more time to requests that we've already have multiple times. Before working on this implementation, we'll probably wait until OpenJData becomes part of the BIDS specs, or until some users have a specific need for it. Wrapping up our FEM pipeline (including iso2mesh and brain2mesh) is more of a priority for the next few months.

But at least, I am aware of this project and will keep its possibilities in mind.
Thanks!

fangq, 2020-04-27:
JNIfTI is nothing by a JSON file (and its binary counter part). All it needs is JSONLab (and optionally zmat).

JSONLab is very compact (200KB uncompressed including documentations), highly portable (work with matlab and octave) with a permissive license.

If you can add JSONLab to BrainStorm as an optional (or required) dependency, supporting JNIfTI is only a matter of a few lines of codes. I will be happy to create a pull request for that. 

The only other decision left to be made is whether you want to support binary data compression/decompression. This is done via zmat (https://github.com/fangq/zmat, a mex file about 200k of size). But if you already have something in place that supports stream-level compression/decompression, then zmat can be replaced.

rcassani, 2025-09-17:
@fangq, text and binary NIfTI files are now supported in Brainstorm [144f6cc](https://github.com/brainstorm-tools/brainstorm3/commit/144f6cc1063b83d4a27efcf48fc9ef663f64dfb8)