brainstorm-tools/brainstorm3 #287: Cannot extract SSP time series

gsdouk, 2020-04-30:
Hello,

I am trying, without succeeding, to extract the SSP time series and save them in a new file. Specifically, I follow step by step the relative tutorial: [Tutorial 13: Artifact cleaning with SSP](https://neuroimage.usc.edu/brainstorm/Tutorials/ArtifactsSsp)  and its section: **Extract the time series**. First of all, I imported a segment of the recordings without applying the SSP projectors (as suggested) and then, after opening the recordings I just imported, I select from the menu **Artifacts > Load projector as montages** (as the second bullet suggests). The first issue is that no projectors are loaded as montages, no window pop-up or something. Secondly, on the pipeline editor when I select the **Apply Montage** process there is nowhere on the montage name drop down menu the name of the SSP projectors montage. I tried to save the active projectors by selecting the menu **Artifacts > Select active projectors** but I could not find the time series of the projectors on the saved mat file. 

I am sorry if I missed or misunderstood something but, is there any other way to save the SSP time series as a struct with field containing the _SSP_projectors x time points_ in order to apply further processing on Matlab?    

Thank you in advance for your time.

ftadel, 2020-05-01:
I've just checked with the data from two online tutorials (Introduction auditory dataset with SSP, and epilepsy dataset with ICA), and both still work as expected. You must have at least one set of SSP decomposition available for the file you want to process. 

> The first issue is that no projectors are loaded as montages, no window pop-up or something

You're not supposed to see anything occurring at this stage. It's only when looking into the montages menu that you should see SSP/ICA montages (in the Record tab of the main Brainstorm window, or in the popup menu of the time series figure).

> Secondly, on the pipeline editor when I select the Apply Montage process there is nowhere on the montage name drop down menu the name of the SSP projectors montage

They should be added at the very end of the list.

If you don't understand where is your issue coming from, please make sure of the following first:
- Update Brainstorm to the latest version 
- Check that whether you don't have silent error messages (red text) in the Matlab command window. If you see any red text, copy-paste it here.
- Check with the dataset from the introduction tutorials, which you should still have in your database, in order to be able to reproduce strictly what is illustrated in the tutorial
- Please post a screen capture of the window you obtain when selecting the menu "Artifacts > Select active projectors" with the continuous file opened
- In this window, select one SSP decomposition, unselect all the components checkboxes, select all the components lines (highlighted in blue - press CTRL+A), and click on the button "Display components time series" => Post a screen capture of the entire screen, showing the drop-down menu of the montages opened in the main Brainstorm window (in the toolbar of the Record tab)

gsdouk, 2020-05-03:
>Update Brainstorm to the latest version

I am using the latest version

>Check that whether you don't have silent error messages (red text) in the Matlab command window. If you see any red text, copy-paste it here.

There are no silent error messages

>Check with the dataset from the introduction tutorials, which you should still have in your database, in order to be able to reproduce strictly what is illustrated in the tutorial

I used it and followed step by step the tutorials from the beginning and still cannot save the projectors as time series

>Please post a screen capture of the window you obtain when selecting the menu "Artifacts > Select active projectors" with the continuous file opened

[image]

> In this window, select one SSP decomposition, unselect all the components checkboxes, select all the components lines (highlighted in blue - press CTRL+A), and click on the button "Display [...]

[image]

Some more print screens after following the tutorial. The first one is after I selected the menu **Artifacts > Load projectors as montages**

[image]

And the second one is after importing a segment of the recordings into the database showing the very end of the drop-down menu of the **Apply montage** process.

[image]

ftadel, 2020-05-04:
All you screen captures show the times series of the continuous file, and everything looks good.
The last figure shows the options of the process "Apply montage", but for the imported segment 60-90s. This relates to a different channel file, and therefore it could contain a different list of projectors. If you imported this data block before computing the SSP, it is normal that there is no "SSP montage" available.

- Delete the imported folder, 
- import again a short block of data from the first run,
- double-click on this new **imported file** to open it
- menu Artifacts > Load projectors as montages, then make sure you can see them in the montage menu
- select the new imported file in the Process1 and check the list of montages available in the options of the process "apply montage"

gsdouk, 2020-05-20:
If I import a block of data **without applying the SSP/ICA projectors** [as the tutorial designates](https://neuroimage.usc.edu/brainstorm/Tutorials/ArtifactsSsp#Extract_the_time_series) the projectors are not available neither in the menu Artifacts > Load projectors as montages nor  in the options of the process "apply montage". However, if I select the option **to apply the SSP/ICA projectors** while importing the block the projectors are available in the menu Artifacts and in the options of the process "apply montage". 

About the components: they are saved as Nsensors x Ncomponents, is there any way for them to be saved as Ncomponents x Time or are they saved somewhere else on the hard drive in that format?

ftadel, 2020-05-25:
> If I import a block of data without applying the SSP/ICA projectors as the tutorial designates the projectors are not available neither in the menu Artifacts > Load projectors as montages nor in the options of the process "apply montage".

You are right, the tutorials were giving the wrong instructions. If you do not select the option "Apply SSP/ICA projectors", all the projectors get discarded from the channel file created in the new folder. 
I fixed this section of the tutorial to instruct to keep this option selected:
https://neuroimage.usc.edu/brainstorm/Tutorials/ArtifactsSsp#Extract_the_time_series
Please let me know if this seems correct to you now.

> About the components: they are saved as Nsensors x Ncomponents, is there any way for them to be saved as Ncomponents x Time or are they saved somewhere else on the hard drive in that format?

This is exactly what you get in the final files, after running the process "Apply montage". The files are saved in .mat files, you can access them in a variety of ways. 
https://neuroimage.usc.edu/brainstorm/Tutorials/Epoching#On_the_hard_drive
https://neuroimage.usc.edu/brainstorm/Tutorials/Scripting#File_structures

I'm closing this issue now. Please reopen it if you find additional problems in the documentation or the software.

closed 2020-05-25 by ftadel

gsdouk, 2020-05-25:
> Please let me know if this seems correct to you now.

Yes, I think it is quite understandable.

 
> This is exactly what you get in the final files, after running the process "Apply montage". 

By exploring the relative .mat file I understand that components are saved as Nsensors x Ncomponents, as the [corresponding tutorial](https://neuroimage.usc.edu/brainstorm/Tutorials/ArtifactsSsp#Extract_the_time_series) denotes. Specifically, I am working with EMEG data (total # of sensors = 410) and I applied ICA only on a EEG data (# of sensors = 74).  The components are saved on 410x9 (= Nsensors x Ncomponents) double matrix with non - zero values only on the rows that represent the EEG sensors. According to the [tutorial](https://neuroimage.usc.edu/brainstorm/Tutorials/ArtifactsSsp#Extract_the_time_series) **each column is one spatial component**, so each IC has a length (of non zero values, which is the information I want) = 74 (equal # of EEG sensors). What I was expecting is for the IC's length to be equal with the time vector (length = 600.000).

ftadel, 2020-05-26:
> By exploring the relative .mat file I understand that components are saved as Nsensors x Ncomponents, as the corresponding tutorial denotes. 

Maybe you are looking in the channel file.
In the newly created data files, you have access to the IC or SSP time series (Ncomponents x Ntime).
https://neuroimage.usc.edu/brainstorm/Tutorials/Epoching#On_the_hard_drive
https://neuroimage.usc.edu/brainstorm/Tutorials/Scripting#File_structures