brainstorm-tools/brainstorm3 #289: Nirstorm - installation from brainstorm gui

Edouard2laire, 2020-05-05:
Hi Francois, 

Do you think it could be a good idea to have a button in brainstorm (like for OpenMeeg) to be able to directly download and install nirstorm in Brainstorm ? 

if yes, I can make a PR for that based on brain entropy installation script (https://github.com/brainstorm-tools/brainstorm3/blob/master/external/brainentropy/be_install.m). 
Just to be sure, to make that work, I just need to put all the nirstorm functions inside the nirstorm folder in external*, right ? 

*https://github.com/brainstorm-tools/brainstorm3/tree/master/external/nirstorm

ftadel, 2020-05-06:
Why not, if this is easier for the distribution or nirstorm.

You could add a new menu "`Install Nirstorm`" that calls a function (eg. `brainstorm3/external/nirstorm/nst_install.m`), at the end of the menu "Update".
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/gui/gui_brainstorm.m#L158

For the downloader/installer, you could copy-paste the latest installers I wrote for the new FEM tools:
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/process/functions/process_generate_fem.m#L1056
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/process/functions/process_generate_fem.m#L1162

thomas-vincent, 2020-05-09:
Even if that would be a nice feature, I recommend not going that way for
now.
The main reason is that it adds a non-neglible part to maintain, with
little gain. For now, the installation system works well and is designed to
handle several scenarios to distinguish between end-user installation and
in-dev prototype features related to current research from students.
My advice is that PhD and master students should focus on research rather
than software infrastructure development. What make a good software is that
it robustly solves relevant problems, and even better: problems that no
other software address. Ergonomics is of lower importance in research, as
long as everything is well documented.

ftadel, 2020-05-12:
Ok, then I'm closing this issue.

closed 2020-05-12 by ftadel