brainstorm-tools/brainstorm3 #293: FEM: Remove bottom part of head model

ftadel, 2020-05-14:
Hi, 

Just a low priority task to add to the todo list. 

In order to reduce the FEM computation time, it's highly recommended to remove the bottom part of the head model. 

The recommendation (Lanfer et al), is to cut the head at  40mm below the lowest part of the skull. 

Since cutting the mesh if not recommended (or not possible), we need to cut the MRI or the MASK before calling any mesh method. 

We can set it as the default option or add it as an option before calling the mesh generator.

_Originally posted by @tmedani in https://github.com/brainstorm-tools/brainstorm3/issues/185#issuecomment-626079804_

ftadel, 2020-05-14:
Do you mean: before calling SimNIBS only?

We could add something that cuts the volume before saving the T1.nii in the temporary folder:
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/process/functions/process_generate_fem.m#L585

It could be something similar to the defacing using the SPM technique, cutting below a given plane in MNI space:
https://github.com/brainstorm-tools/brainstorm3/blob/master/toolbox/process/functions/process_mri_deface.m#L202

ftadel, 2020-05-14:
PS: we need to be careful about not cutting above the lowest of the electrodes.
With EGI 256 electrode hydrocel nets, the lowest electrodes are really low in the neck.

ajoshiusc, 2020-05-14:
> Hi,
> 
> Just a low priority task to add to the todo list.
> 
> In order to reduce the FEM computation time, it's highly recommended to remove the bottom part of the head model.
> 
> The recommendation (Lanfer et al), is to cut the head at 40mm below the lowest part of the skull.

@tmedani 
We can cut the mesh, that way we are not affecting other processing. To cut the mesh, you can first find the centroids of tetrahedrons and then delete the tetrahedrons that have centroid below the specified plane. After that, we need to delete the vertices that have not been used and reindex. I have code for that (deleting unused vertices). Would that make sense?

tmedani, 2020-05-14:
Hello,
Thanks  @ftade  for opening this issue.

> Do you mean: before calling SimNIBS only?

Not only simnibs but most of the methods.

> PS: we need to be careful about not cutting above the lowest of the electrodes.
> With EGI 256 electrode hydrocel nets, the lowest electrodes are really low in the neck.

Indeed, we can add it as an option that asks the user to keep the whole volume or just the top part. 

> > In order to reduce the FEM computation time, it's highly recommended to remove the bottom part of the head model.
> 
> @tmedani
> We can cut the mesh, that way we are not affecting other processing. To cut the mesh, you can first find the centroids of tetrahedrons and then delete the tetrahedrons that have centroid below the specified plane. After that, we need to delete the vertices that have not been used and reindex. I have code for that (deleting unused vertices). Would that make sense?

Thanks, @ajoshiusc, I know in some situations we need to re-index all the nodes for the 'new' mesh, and sometimes it fails.
If you have already tested version it would be great to test it on our head models.

Both options (cut mri or cut mesh) may have their advantages. 

let test and discuss...

tmedani, 2020-06-01:
Hi, 

the first version of the code is in this [pull request](https://github.com/brainstorm-tools/brainstorm3/pull/301#issue-426182419)

@ajoshiusc I tested and adapted your code to tetra mesh and it works fine, thanks :).

tmedani, 2020-06-11:
This message is from this [pullrequest ](https://github.com/brainstorm-tools/brainstorm3/pull/301#issue-426182419 )

We need just to change Z =  85, to Z = 115

explanation below.

then we can close it :)

Hi,
Again this is great, 

I think with Z = -85, 
there is only 10 mm from the lower part of the skull to the cutting plan

[image]
[image]

So as you said this is slightly high. 

I have checked with  Z = -115, and  there is ~ 40mm 

[image]
[image]

Let's use this as the default value since it fits what is said in this paper 
https://onlinelibrary.wiley.com/doi/full/10.1002/hbm.24754

[image]

And their figures fit better the cut at **Z = 115**

[image]

We should explain this on the tutorial since this is also related to the position of the sensor. 

btw, a small bug, the 'Get Coordinates' panel is not working on the FEM mesh display. 
[image]

Thanks & A+

ftadel, 2020-06-15:
> And their figures fit better the cut at Z = 115

Done: https://github.com/brainstorm-tools/brainstorm3/commit/d4582fd0411b0dcb06d9068dbcb61de877505a3b

> a small bug, the 'Get Coordinates' panel is not working on the FEM mesh display.

Fixed: https://github.com/brainstorm-tools/brainstorm3/commit/3de21ca7d2e5dabadaae1fb92c4f09ca50107015

closed 2020-06-15 by ftadel