brainstorm-tools/brainstorm3 #299: Errors after latest update

gsdouk, 2020-05-31:
Hello,

I have combined EEG/MEG data and there is a problem with the process Apply SSP & CTF compensation. Until the day before 30/May/2020, there was absolutely no problem applying this process. Now I am getting this error:    
[image]

I have run the same process many times before on the exact same data files without a problem (I was using the previous version of 29/May/2020).  On the latest version, (29/May/2020) I have the aforementioned error.

Secondly, when using the latest version of Brainstorm (29/May/2020) ICA yields different results from when using the previous version. Exact same input and different results before and after the 29th May. The first image is taken the first week of May and the second one is taken 31/May and is wrong (exact same data, exact same processing steps before ICA was applied). Is the update related with any of these processes? How can I get the same behavior as before?
[image]
     
[image]

Finally, the [Download](https://neuroimage.usc.edu/bst/download.php) page is not available since 30/May (or that day i noticed it). So I cannot download the tutorial data and run the tutorials again.  

Also, I have reset/reinstall the latest version (I downloaded the code from github) of Brainstorm several times.

Thank you in advance.

ftadel, 2020-06-01:
> I have run the same process many times before on the exact same data files without a problem (I was using the previous version of 29/May/2020). On the latest version, (29/May/2020) I have the aforementioned error.

There were changes in bst_process.m in the past month, so it is plausible that some new bugs appeared. But I could not reproduce your bug you reported.
If you want us to help you debug this, could you please share a short example data file ?

- Right-click on the subject > File > Duplicate subject
- Delete all the files that are not needed to reproduce the behavior:
  - in the anatomy, keep only the MRI and the head surface
  - in the functional data, keep only the channel file and one short block of imported data (1s long or less) - delete forward and inverse models
- Make sure you can still reproduce the error with this minimal dataset
- Right-click on the subject > File > Export subject
- Upload the zip file somewhere (dropbox, google drive...) and share the download link here
- Describe exactly how you generate the error (exactly where you click and in which order, screen capture of the options of the two processes: ICA and Apply ICA)

> Secondly, when using the latest version of Brainstorm (29/May/2020) ICA yields different results from when using the previous version. Exact same input and different results before and after the [...]

I tested with the EEG/Epilepsy tutorial dataset (https://neuroimage.usc.edu/brainstorm/Tutorials/Epilepsy#Artifact_cleaning_with_ICA), I still get equivalent ICA decomposition with an older version (February 26th, 2020) and with today's version. Example of the IC topographies I get in output:

[image]
[image]

They differ, but ICA algorithm Infomax from EEGLAB starts is **not deterministic**. It starts with a random initializer and most of the time would produce different decompositions if executed twice. However, for the IC that are clearly distinct, the algorithm should identify them at every execution. It is a good way to understand if an ICA decomposition is meaningful: run the algorithm a few times, and the components that are identified at every iteration are probably more meaningful than the others. 
This is why I would consider the decompositions above "equivalent": they are different, but the components that we are interested in (blinks, sacades and other eye-related components) are always identified, and with the same order (ie. same level of correlation with the EOG).

Yours are too different though... Are you sure the list of bad channels is the same?

> Finally, the Download page is not available since 30/May (or that day i noticed it). So I cannot download the tutorial data and run the tutorials again.

This is already fixed. Thanks for reporting.

gsdouk, 2020-06-03:
I downloaded the latest (02/06) Brainstorm version and I run everything again, many times, and the bug did not appear again. Also, ICA yields the corrects results. 

>Yours are too different though... Are you sure the list of bad channels is the same?

Yes, they were the same. I tested it many times. 

After extracting the ICs, I process them. I wish to reconstruct data (inverse ICA) using the new ICs. Is there any way to achieve it? Loading saved montages does not work.

closed 2020-06-08 by ftadel