brainstorm-tools/brainstorm3 #302: FEM source space construction

tmedani, 2020-06-02:
Hi, 

In this issue, we will discuss and solve the FEM source space generation for the realistic head model. 

This will be important in the case where we separate the brain into WM and GM. 

For more accurate results, all the dipoles (sources) should be within the GM layer, and ideally at least one element far from the neighboring tissues.  

The cortex generated with CAT is good, however, with the SimNibs FEM mesh, some points can be either no the CSF or the WM.

We need to solve this for more realistic head models. 

A+

tmedani, 2020-07-24:
Hello, 

A first solution for this issue is proposed in this PR 
https://github.com/brainstorm-tools/brainstorm3/pull/316#pullrequestreview-455193388

This is faster than the standard approach which checks all the dipoles and all the GM elements and takes between 1-3 hours to build the source space. 
In this pull request, we use the SPM-CAT (0rrany other cortex) cortex and then check the position of the dipoles, then all the dipoles located outside the GM are moved slightly to the nearest centroid of the GM element and it takes only few minutes.

A+

ftadel, 2020-08-03:
Modified solution proposed in this commit: https://github.com/brainstorm-tools/brainstorm3/commit/29176ee27f3f1f71237e66dbb28eac2f64f0ee86

The new option is available in the DUNEuro option panel (adaptive to the layers available):
[image]
[image]

This solution keeps your logic for identifying the closest GM element for dipoles that are outside the GM.
But I think your solution of replacing the position of the dipole directly with the centroid of the element doesn't work, as in my tests it seemed to project a lot of source points into exactly the same location... I'm proposing here a first draft for something more adaptive, that moves the dipoles progressively towards the center of the centroid until it is inside the element, but not necessarily reaching its center. In the Matlab command window, you can see by how many millimeters each dipole was moved.
What do you think?

To evaluate the impact on the shape of the cortex, you can uncomment the two lines with calls to view_surface_matrix. This lets you use the Surface tab in the Brainstorm window to manipulate the figure (transparency, smoothing, edges...) => much more convenient than most surface viewers in Matlab.
Please check this carefully. The results I got when using a full resolution SimNIBS FEM mesh together with a FreeSurfer cortex surface were a bit weird, it was moving a lot of vertices towards the same attractors, distorting the crests of the gyri...

I'm not sure I understand what you tried to do with the dipoles inside the WM... For now I simply replaced it with the same logic as for the dipoles outside the GM: moving progressively from the WM to the center of the closest GM element.
But maybe we'll have to restore some logic previously implemented in this new function.

Please test it and we'll iterate from there.

tmedani, 2020-08-03:
Hi @ftadel 

> This solution keeps your logic for identifying the closest GM element for dipoles that are outside the GM.
> But I think your solution of replacing the position of the dipole directly with the [...]

it's weird, so first I'm checking and identifying all the dipoles outside the GM, and then apply the translation only on these dipoles. 
It was working in my test case, but if you find this case, then it's not good!

 > I'm proposing here the first draft for something more adaptive, that moves the dipoles progressively towards the center of the element until it is inside the element, but not necessarily reaching its center. In the Matlab command window, you can see by how many millimeters each dipole was moved.

I will check it, 

in my proposition, I tried to do something faster by choosing the centroids of the GM elements.

So ideally, the nearest node of the dipole should be within the GM and at least one element far from the interface(s).  
The FEM source model uses the neighboring nodes to model the dipole, and in the case where the neighboring nodes are outside the GM, it can generate numerical errors. This is the St-Venant's condition

> What do you think?

I will check it asap.

> To evaluate the impact on the shape of the cortex, you can uncomment the two lines with calls to view_surface_matrix. This lets you use the Surface tab in the Brainstorm window to manipulate the figure (transparency, smoothing, edges...) => much more convenient than most surface viewers in Matlab.

wow .. great to know .. thanks ... I will try it.

> Please check this carefully. The results I got when using a full resolution SimNIBS FEM mesh together with a FreeSurfer cortex surface were a bit weird, it was moving a lot of vertices towards the same attractors, distorting the crests of the gyri...

I did my testing with the spm/cat cortex, and in SimNibs they introduce artificial CSF in the occipital area, then most of the problems will appear in this area.

I will check your implementation, it's really 'casse tete' to find a good and fast algorithm to do this. 

> I'm not sure I understand what you tried to do with the dipoles inside the WM... For now, I simply replaced it with the same logic as for the dipoles outside the GM: moving progressively from the WM to the center of the closest GM element.

I have followed the same logic since I'm using the outer surfaces, so the dipoles within the WM means that they are outside the GM, then we need to move them inside the GM. 

This case will happen when we have the full head model with WM, GM and CSF, then all the dipole should be within the thin GM layer. 

btw, I did some testing and it seems that the function inpolyhedron is faster than the previous one used in 'force inside the skull'. 

A+

ftadel, 2020-08-04:
> it's weird, so first I'm checking and identifying all the dipoles outside the GM, and then apply the translation only on these dipoles. It was working in my test case, but if you find this case, then it's not good!

If you use the high-resolution cortex surface from FreeSurfer, you'll have many vertices that correspond to the same GM element. And therefore, if you simply replace their original location with the center of this element, then they would end up exactly at the same location, or am I missing something? But maybe this is not a problem for the FEM computation anyway?

> So ideally, the nearest node of the dipole should be within the GM and at least one element far from the interface(s).

Both your approach and mine are using the first available node, so it will never be one element far from the interface.
Any in practice we will NEVER have more than one layer of elements in the GM, will we?

> btw, I did some testing and it seems that the function inpolyhedron is faster than the previous one used in 'force inside the skull'.

That's what I discovered yesterday as well. The function inpolyhedron.m seems to be ~10x faster than inpolyhd.m in the few tests I made. Maybe we should think about replacing it everywhere at some point, but we'd need to test its full portability first (works on all Matlab versions, all OS, and in the compiled version).
For now, I think it is not a priority, unless there are calls to inpolyhd.m that are really making some processes too slow to be used. Let's keep optimization of other processes for later. This function "Force inside skull" is called only once and does not take hours, so it's not an emergency to improve it right now.

tmedani, 2020-08-05:
> If you use the high-resolution cortex surface from FreeSurfer, you'll have many vertices that correspond to the same GM element. And therefore, if you simply replace their original location with the [...]

Now it's much clear, I totally agree with you since you are using the high-resolution mesh.
In this case, I would assume that the vertices are too close (perhaps less than 0.5 mm distance between nodes)
This explains why many vertices will end up in the same FEM centroid, you are right. 

>But maybe this is not a problem for the FEM computation anyway?

Yes, for the FEM computation does not care, however, it will have more than one computation for the same dipole, but I think this is not the time-consuming part of the FEM. 
Also, I suppose (not sure) that all the dipoles far by less than 0.5mm from a centroid may have an equivalent activation as they are placed in this centroid (maybe we need to check to be more sure), then less/no effects on the results.
furthermore, since we are moving these nodes internally and we are not saving the 'real' source space, these effects will be invisible from the user side.   
 

> Both your approach and mine are using the first available node, so it will never be one element far from the interface.
> Any in practice we will NEVER have more than one layer of elements in the GM, [...]

Yes, as I said before, our approaches are faster but will not sure to have all the nearest nodes to the dipoles within the GM. 

> Any in practice we will NEVER have more than one layer of elements in the GM, will we?

it depends on the GM thickness and the mesh resolution,
from the literature the thickness of the grey matter varies from 1mm to 5mm with a mean of ~3mm 
in the thick layers, it's easy to fill the condition, in the thin layers no. The only solution for the thin layer is to refine more the GM mesh. 
I will work on this at some time.

The "last chance" solution is to use another source model, such as the PI source model (less accurate than the Venant but stable at the interface).  

> That's what I discovered yesterday as well. The function inpolyhedron.m seems to be ~10x faster than inpolyhd.m in the few tests I made. Maybe we should think about replacing it everywhere at some [...]

Yes, in my case, if I remember well, it goes up to 30/40 times faster. 

also, btw, I think the function "Force inside skull" is proposed after the full openmeeg computation, wich is not optimal and not useful after 20min to1 hours waiting for the results 
I have reported this issue here 
https://github.com/brainstorm-tools/brainstorm3/issues/185#issuecomment-612980597
I did not manage to create a PR for this... I will try to do it asap.

tmedani, 2020-08-06:
Hi @ftadel 

I want through your modification at option 2 which seems to be more stable for the high-density cortex. 

So I think both solutions are complementary, 

Option 1: is faster since it moves directly to the nearest centroid, however, in high mesh resolution cortex  more than one dipoles can end up in the same centroid, this problem has less chance to be produced in the 'normal' cortex

Option 2: can be slower if there are too many dipoles outside, but can avoid the problem of multiples dipole in the same centroid.

For the question which one to use, I would say it depends on the size of the source space (or the resolution of the cortex),
in the case of the lower maybe option 1 is better and higher option 2?

Also, regarding option 2, I have these ideas that can improve it, 

1 - instead of to loop nFix times up to the find the first position inside the targetFaces
with this equation

          tmpVert = (nFix - iFix)/nFix * cfg.GridLoc(iVertOut(i),:) + iFix/nFix * ElemCenter(iTarget,:);
 (smart relation, I still don't get it ... ) 

I propose to use directly (without the loop )

                    tmpVert = 1/10 * cfg.GridLoc(iVertOut(i),:) + 9/10 * ElemCenter(iTarget,:);

or sharper 

                    tmpVert = 1/20 * cfg.GridLoc(iVertOut(i),:) + 19/20 * ElemCenter(iTarget,:);

in order to be sure that the new dipole is within the targetFaces and close to the centroid ==> deeper within the GM,

2- It can be much better if we can push it slightly deeper from the centroid, 
let's say 10 to 20 % of the original distance dipole-centroid as in the cross in this figure

[image]

 in that case, the FEM code will have more chances to find the nearest node within the GM.  

Does this make sense for you?

tmedani, 2020-11-25:
Hello all, 

 
A possible solution is proposed in this PR https://github.com/brainstorm-tools/brainstorm3/pull/364#issue-527308191

@ftadel, if you agree with it we can close this discussion,

Thanks,

closed 2020-11-26 by ftadel