brainstorm-tools/brainstorm3 #306: Anisotropy in the Duneuro panel

tmedani, 2020-06-10:
Hi all, 

@ftadel this is a quick issue that we can solve but we need to discuss it before. 

In the case of the FEM Duneuro computation with the anisotropy tensors, we need to add an input to the duneuro panel. 

In this case, the iso conductivity values are not needed since the tensors are already in the femmat structure. Furthermore, in this case, the cauchy files will be used (knw and geo). 
So we need something that tells the user if he wants to use the tensor or not?
and use this as a boolean in the code in order to use the catchy files. 

also in this case the scalar values of the conductivity are not required. 

As previously discussed, I will not code and push something wrong or not be useful, so let's discuss  

Here is the current panel  
[image]

I added some code and I change the panel to this

[image]
 

Where the case 'UseTensor' will load the tensor from the head model (femmat) and uses the Cauchy files (*.knw and *.geo)

when we select the UseTensor, the values of the conductivity should be disabled.

Does it work for you or do you have a better option?

The case 'isotropic' is not useful for the final user, it's just in my local version for more testing. 
FYI, I'm using this to create N times (length(Element)) the isotropic tensors to fille the knw file and then call the duneuro.  

 

Also, I have updated the tutorial with some results that validate the anisotropy FEM computation, check  [here](https://neuroimage.usc.edu/brainstorm/Tutorials/ReviewForward?highlight=%28duneuro%29#Ansiotropy_model)

A+

ftadel, 2020-06-15:
> when we select the UseTensor, the values of the conductivity should be disabled.

I would go for skipping completely this extra parameter "Use tensors":
- if the tensors are available in the FEM file, we use them and hide the panel "FEM layers and conductivities" (or is just the edit boxes you want to disable, and keep the tissue selection checkboxes?)
- if the tensors are not available in the FEM file, then we keep the panel as it is now.

In most cases, people who have the DTI images would use them, and the ones without would not. So it is not really an option that most users would need to see in this panel. And the fewer options in this panel the better.

For the very few people who would like to test and compare the two versions: you would just need to duplicate the FEM mesh file before computing the FEM tensors, in order to have one with tensors and one without. Then you just change the selection of the default FEM file (in green) for the subject to decide whether you want to use tensors or not.
This can be illustrated in the tutorial.

> Where the case 'UseTensor' will load the tensor from the head model (femmat) and uses the Cauchy files (*.knw and *.geo)

Can you please clarify the workflow related with these two files?
What do they contain and would they change between two different executions of duneuro for a given subject?
My understanding is that these should be "anatomical files" that we could import into the database once for the subject, instead of asking them at each computation.
Could we store what they contain in the FEM mesh file as well? If they should always be used together with a set of "Tensors" (as in the FEM mesh), they should be stored next to the Tensors field.

tmedani, 2020-06-15:
> I would go for skipping completely this extra parameter "Use tensors":
>
> * if the tensors are available in the FEM file, we use them and hide the panel "FEM layers and conductivities" (or is just the [...]

I completely agree this is the best solution. 

> For the very few people who would like to test and compare the two versions: you would just need to duplicate the FEM mesh file before computing the FEM tensors, in order to have one with tensors [...]

Yes, this make sense also. 

> Can you please clarify the workflow related with these two files?
> What do they contain and would they change between two different executions of duneuro for a given subject?
> My understanding is that [...]

No, this is an internal routine. 
there is a test in the bst_duneuro code where we have used the flag 'UseTensor' as a boolean. 

So if the case is isotropic, then Duneuro will use the .msh format for the head mesh and the .con format for the isotropic conductivity.

in the case where the anisotropy is used, the flag UseTensor is set to 1 and then Duneuro will expect the .geo file for the mesh and the .knw file for the conductivity tensors. 

With the new modification, instead to test the value of 'UseTensor' we will check is the field femmat.Tensors is empty or not. 

No more files will be loaded to brainstorm the database. 

Does this make sense for you?

I will change the bst-duneuro code and create a PR. 

However, I can't do it for the panel, 

I'm still having issues to understand how the code of the panels and I can't debug them.
is there any documentation to recommend?

ftadel, 2020-06-18:
Done:
https://github.com/brainstorm-tools/brainstorm3/commit/070bbb8a7d4011b8d95bc8302a31ba4861dceca6

Now when there are Tensors in the FEM file, it shows (using tensors) instead:
[image]

ftadel, 2020-06-18:
(please close this issue if this is satisfying)

tmedani, 2020-06-18:
> Done:
> [070bbb8](https://github.com/brainstorm-tools/brainstorm3/commit/070bbb8a7d4011b8d95bc8302a31ba4861dceca6)
> 
> Now when there are Tensors in the FEM file, it shows (using tensors) instead:

This is super. 

Just one more thing to check, 

we need to check if the right files(knw and geo) are written when we de-select some tissues from this panel. 

I will check it

Thanks again

tmedani, 2020-07-23:
Hi, 

when this [PR  ](https://github.com/brainstorm-tools/brainstorm3/pull/316) will be accepted we can close this issue :)

tmedani, 2020-07-23:
https://github.com/brainstorm-tools/brainstorm3/pull/316#issuecomment-663043097

closed 2020-07-23 by tmedani