brainstorm-tools/brainstorm3 #307: new method for defacing MRI (3D and 4D volume)

tmedani, 2020-06-11:
Hi, 

I just want to ask if it's possible to add a new method to deface the MRI and even the 4D volumes ((DW).

With @ajoshiusc we want to anonymize the MRI without touching the skull as it's the case now.

This is important since the skull on the frontal area can have important effects on the head models.

here is an example of the results :

[image]
 
On the right, the current method of brainstorm, on the left the method that @ajoshiusc implements.

It needs some more work, but I think it worth it.

What do you think? 

Best,

ftadel, 2020-06-11:
Looks good.
If this is Matlab-based, send me the code and I will add it to the process.

ajoshiusc, 2020-06-11:
Here is the Matlab code and instructions for using it:
https://github.com/ajoshiusc/deface_mri

ftadel, 2020-06-15:
Is this a linear registration to a template space, followed by a masking with a predefined mask?

If so, we could simply use Anand's defacing mask with the current SPM affine MNI registration we already have in Brainstorm, right? No added code, no need for external calls to BrainSuite, no computation time beyond the disk I/O.

If this seems reasonable to you, we could start with this super fast option, and move on with BrainSuite-based version only if the outcome is not satisfying.

@ajoshiusc Could you provide us with a version of your brain.deface.mask.nii.gz in MNI space?

tmedani, 2020-06-15:
> If so, we could simply use Anand's defacing mask with the current SPM affine MNI registration we already have in Brainstorm, right?

You mean instead to use the default MNI plan as mask we can use the 'defaced face' as a mask? 
I think this is a good idea indeed?

let's wait for @ajoshiusc comments.

ajoshiusc, 2020-06-15:
I initially was trying to use the MNI atlas for the mask, since that way we don't need additional registration. But the issue is that MNI atlas does not have a full face and so defacing mask cannot be generated for MNI atlas. Its field of view is much smaller. 
Therefore I have generated it for BCI-DNI atlas in BrainSuite and coregister subject to BCI-DNI atlas.
What we can do in theory is that coregister BCI-DNI atlas to MNI atlas and save the transformation. And we have transformation from MNI to BCI. We can concatenate those two transformations.
But in these transformations, I am not sure if field of view transfers correctly.

The direct rigid registration of subject to BCI, only takes a minute or two so I have left it that way currently and will allow this tool to be used as a standalone outside BrainStorm.

tmedani, 2020-06-16:
Hello, 

I followed the following process, 
I load the MRI to  Brainstorm and set the fiducial points. 
then I load the mask 'BCI-DNI_brain.deface.mask.nii.gz' also to brainstorm,  with the option of coregistration and reslice to the previous MRI.

And finally, I multiply the MRI by the mask. 

This process is slightly faster, however,  the final results is not good as expected, it cuts part of the ears, some areas of the head and the removed part is not smooth (Sawtooth shape)

[image]

[image]

ajoshiusc, 2020-06-16:
Instead of loading BCI-DNI_brain.deface.mask.nii.gz, can you try loading BCI-DNI_brain.deface.nii.gz? The first one is binary mask and the second one is with anatomy, so the second one might register better.

tmedani, 2020-06-16:
@ajoshiusc 
but how can I use it as mask?

ajoshiusc, 2020-06-16:
@tmedani Thresholding the image by 0 will give you the mask.

tmedani, 2020-06-16:
ok got it, thank you. 

Here is the results, no much better 
[image]

ftadel, 2020-06-16:
Drawing a defacing mask on the MNI brain directly could be easier.
Or maybe a surface which includes the areas to remove in the MNI brain (ICBM152).

Everything else would be easy: converting the coordinates of the surface "to remove" to subject space, converting the surface to a mask of voxels, masking the subject volumes with this mask.

@ajoshiusc Did you draw manually this mask slice by slice?

ajoshiusc, 2020-06-16:
@ftadel Yes, I drew it on BCI-DNI atlas manually.We can't have it on MNI space because MNI space doesn't have face in the FOV (screenshot below).
[image]

ftadel, 2020-06-17:
I'm not sure what volume this is, but when we talk about "MNI template" in the context of Brainstorm, this refers to the ICBM152 non-linear atlas (main MNI reference for the past 10 years):
https://www.mcgill.ca/bic/software/tools-data-analysis/anatomical-mri/atlases/icbm152-non-linear-2009

However, it still doesn't include a face...
Here are volumes I prepared using Brainstorm, padding the ICBM152 MRI with zeros, and interpolating the head surface we're using our template:
[ICBM_300z.zip](https://github.com/brainstorm-tools/brainstorm3/files/4792653/ICBM_300z.zip)
[ICBM_headmask.zip](https://github.com/brainstorm-tools/brainstorm3/files/4792655/ICBM_headmask.zip)

Could you prepare a "face mask" for these volumes, in .nii format with the same transformations?

ajoshiusc, 2020-06-18:
@ftadel 
Here is the mask in the same space as the ICBM300z.
[ICBM_300z.deface.mask.nii.gz](https://github.com/brainstorm-tools/brainstorm3/files/4796371/ICBM_300z.deface.mask.nii.gz)

Here is a screenshot
[image]

Please take a look and let me know if any modification is needed,

ajoshiusc, 2020-06-19:
Here is a version of the mask that is inflated a bit.
[ICBM_300z.dilated.deface.mask.nii.gz](https://github.com/brainstorm-tools/brainstorm3/files/4806014/ICBM_300z.dilated.deface.mask.nii.gz)

[image]

ftadel, 2020-06-20:
What about this implementation?
https://github.com/brainstorm-tools/brainstorm3/commit/d715e27ff2995853425c617769b4622734791dfe

[image]

[image]

closed 2020-11-09 by ftadel